robotology/robotology-superbuild #991: Homebrew Failure Early January 2021

traversaro, 2022-01-10:
Homebrew jobs are failing, but in this case it seem a legitimate failure due to the latest fmt:
~~~
2022-01-10T03:06:48.2128850Z [4/127] Building CXX object src/TextLogging/CMakeFiles/TextLogging.dir/src/Logger.cpp.o
2022-01-10T03:06:48.2229960Z FAILED: src/TextLogging/CMakeFiles/TextLogging.dir/src/Logger.cpp.o 
2022-01-10T03:06:48.2336590Z /Applications/Xcode_12.4.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DFMT_LOCALE -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -DTextLogging_EXPORTS -D_USE_MATH_DEFINES -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include -isystem /usr/local/include -O3 -DNDEBUG -isysroot /Applications/Xcode_12.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -fPIC -std=gnu++17 -MD -MT src/TextLogging/CMakeFiles/TextLogging.dir/src/Logger.cpp.o -MF src/TextLogging/CMakeFiles/TextLogging.dir/src/Logger.cpp.o.d -o src/TextLogging/CMakeFiles/TextLogging.dir/src/Logger.cpp.o -c /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/src/Logger.cpp
2022-01-10T03:06:48.2343380Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/src/Logger.cpp:8:
2022-01-10T03:06:48.2444820Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include/BipedalLocomotion/TextLogging/Logger.h:12:
2022-01-10T03:06:48.2545890Z In file included from /usr/local/include/spdlog/fmt/ostr.h:19:
2022-01-10T03:06:48.2648460Z In file included from /usr/local/include/fmt/ostream.h:13:
2022-01-10T03:06:48.2749090Z In file included from /usr/local/include/fmt/format.h:48:
2022-01-10T03:06:48.2851980Z /usr/local/include/fmt/core.h:1727:3: error: static_assert failed due to requirement 'formattable' "Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt"
2022-01-10T03:06:48.2950910Z   static_assert(
2022-01-10T03:06:48.3052200Z   ^
2022-01-10T03:06:48.3155890Z /usr/local/include/fmt/core.h:1853:23: note: in instantiation of function template specialization 'fmt::v8::detail::make_arg<true, fmt::v8::basic_format_context<fmt::v8::appender, char>, fmt::v8::detail::type::custom_type, const BipedalLocomotion::TextLogging::Verbosity &, 0>' requested here
2022-01-10T03:06:48.3254900Z         data_{detail::make_arg<
2022-01-10T03:06:48.3356320Z                       ^
2022-01-10T03:06:48.3460300Z /usr/local/include/fmt/core.h:1872:10: note: in instantiation of function template specialization 'fmt::v8::format_arg_store<fmt::v8::basic_format_context<fmt::v8::appender, char>, BipedalLocomotion::TextLogging::Verbosity>::format_arg_store<const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.3562950Z   return {std::forward<Args>(args)...};
2022-01-10T03:06:48.3664200Z          ^
2022-01-10T03:06:48.3768780Z /usr/local/include/spdlog/logger.h:332:52: note: in instantiation of function template specialization 'fmt::v8::make_format_args<fmt::v8::basic_format_context<fmt::v8::appender, char>, const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.3872010Z             fmt::detail::vformat_to(buf, fmt, fmt::make_format_args(args...));
2022-01-10T03:06:48.3973520Z                                                    ^
2022-01-10T03:06:48.4076830Z /usr/local/include/spdlog/logger.h:83:9: note: in instantiation of function template specialization 'spdlog::logger::log_<const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.4179180Z         log_(loc, lvl, fmt, std::forward<Args>(args)...);
2022-01-10T03:06:48.4280560Z         ^
2022-01-10T03:06:48.4384560Z /usr/local/include/spdlog/logger.h:89:9: note: in instantiation of function template specialization 'spdlog::logger::log<const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.4487040Z         log(source_loc{}, lvl, fmt, std::forward<Args>(args)...);
2022-01-10T03:06:48.4588590Z         ^
2022-01-10T03:06:48.4692470Z /usr/local/include/spdlog/logger.h:170:9: note: in instantiation of function template specialization 'spdlog::logger::log<const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.4795350Z         log(level::err, fmt, std::forward<Args>(args)...);
2022-01-10T03:06:48.4897230Z         ^
2022-01-10T03:06:48.5001870Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/src/Logger.cpp:35:16: note: in instantiation of function template specialization 'spdlog::logger::error<const BipedalLocomotion::TextLogging::Verbosity &>' requested here
2022-01-10T03:06:48.5106130Z         log()->error("Failed to change verbosity to level {}", verbosity);
2022-01-10T03:06:48.5209610Z                ^
2022-01-10T03:06:48.5311080Z 1 error generated.
2022-01-10T03:06:48.5413460Z [5/127] Building CXX object src/TextLogging/CMakeFiles/TextLogging.dir/src/LoggerBuilder.cpp.o
~~~

fyi @GiulioRomualdi

GiulioRomualdi, 2022-01-10:
I am planning to release blf today/tomorrow with the fix

traversaro, 2022-01-11:
The issue was fixed by https://github.com/ami-iit/bipedal-locomotion-framework/pull/495 in stable and unstable branches, the 0.6.0 blf release and https://github.com/robotology/robotology-superbuild/pull/990 should fix it also for latest releases.

closed 2022-01-21 by traversaro