robotology/robotology-superbuild #1002: Unable to find `i2c/smbus.h` (from conda-forge dependencies installation on Manjaro)

HosameldinMohamed, 2022-01-19:
Similar to #923 but when enabling `ROBOTOLOGY_ENABLE_ICUB_HEAD` ON, I was trying to install the superbuild on Manjaro. But I get an error:

Following [the documentation](https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation). Here's what I have tried.

```
mamba create -n robTest
conda activate robTest
mamba install -c conda-forge cmake compilers make ninja pkg-config
mamba install -c conda-forge ace asio assimp boost eigen gazebo glew glfw gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml spdlog lua
mamba install -c conda-forge bash-completion expat-cos6-x86_64 freeglut libdc1394 libselinux-cos6-x86_64 libxau-cos6-x86_64 libxcb-cos6-x86_64 libxdamage-cos6-x86_64 libxext-cos6-x86_64 libxfixes-cos6-x86_64 libxxf86vm-cos6-x86_64 mesalib mesa-libgl-cos6-x86_64
```

```
/home/hosameldin/Github/robotology-superbuild/src/YARP/src/devices/imuBosch_BNO055/imuBosch_BNO055.cpp:19:11: fatal error: i2c/smbus.h: No such file or directory
```

Even though `i2c/smbus.h` is available in the system files, in `/usr/include`. It's not clear to me when the compiler of Conda ignores system files and when it doesn't.

This is the package that provides the header file for instance https://archlinux.org/packages/community/x86_64/i2c-tools/

HosameldinMohamed, 2022-01-19:
@traversaro you mentioned something about not being able to find `I2C` package by Ubuntu/Debian.

```
ENABLE_yarpmod_imuBosch_BNO055 OFF - Dependencies unsatisfied: 'YARP_HAS_I2C' - imuBosch_BNO055-NOTFOUND
```

traversaro, 2022-01-19:
As you mentioned, the issue is exactly the one that is described in https://github.com/robotology/robotology-superbuild/issues/923#issuecomment-976366528 , including the reason why on Debian/Ubuntu the problem does not emerge. The only problem is that `libi2c`/`i2c-tools` is not packaged on conda-forge, so the workaround is not so easy. Let's fix this.

traversaro, 2022-01-19:
> Let's fix this.

See https://github.com/conda-forge/staged-recipes/pull/17583 .

HosameldinMohamed, 2022-01-19:
> > Let's fix this.
> 
> See [conda-forge/staged-recipes#17583](https://github.com/conda-forge/staged-recipes/pull/17583) .

Wow 🆒

traversaro, 2022-01-21:
> > > Let's fix this.
> > 
> > 
> > See [conda-forge/staged-recipes#17583](https://github.com/conda-forge/staged-recipes/pull/17583) .
> 
> Wow 🆒

This was fixed. If you install `libi2c`, YARP should find that one and you should not have any compilation problem.

HosameldinMohamed, 2022-01-21:
> That PR was merged. If you install `libi2c`, YARP should find that one and you should not have any compilation problem.

Just installed the package and re-compiled, no errors! Thanks a lot @traversaro!

closed 2022-01-21 by traversaro