robotology/robotology-superbuild #1006: CI Failure End of January 2021

traversaro, 2022-01-26:
This night almost all the CI jobs (except LatestReleases) failed ( https://github.com/robotology/robotology-superbuild/actions/runs/1748726517 ) with error: 
~~~
2022-01-26T02:43:08.8673105Z [12/23] Building CXX object CMakeFiles/UnicyclePlanner.dir/src/DCMTrajectoryGenerator.cpp.o
2022-01-26T02:43:08.8673524Z [13/23] Building CXX object CMakeFiles/UnicyclePlanner.dir/src/FeetMinimumJerkGenerator.cpp.o
2022-01-26T02:43:08.8674206Z [14/23] Building CXX object tests/CMakeFiles/DcmInterpolationTest.dir/DcmInterpolationTest.cpp.o
2022-01-26T02:43:08.8674627Z FAILED: tests/CMakeFiles/DcmInterpolationTest.dir/DcmInterpolationTest.cpp.o 
2022-01-26T02:43:08.8677268Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++  -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/UnicyclePlanner/tests -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/UnicyclePlanner/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/share/miniconda3/envs/test/include/eigen3 -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -std=c++14 -MD -MT tests/CMakeFiles/DcmInterpolationTest.dir/DcmInterpolationTest.cpp.o -MF tests/CMakeFiles/DcmInterpolationTest.dir/DcmInterpolationTest.cpp.o.d -o tests/CMakeFiles/DcmInterpolationTest.dir/DcmInterpolationTest.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/UnicyclePlanner/tests/DcmInterpolationTest.cpp
2022-01-26T02:43:08.8679183Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/UnicyclePlanner/tests/DcmInterpolationTest.cpp:20:
2022-01-26T02:43:08.8680069Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/UnicyclePlanner/tests/utils.h:15:10: fatal error: UnicycleTrajectoryGenerator.h: No such file or directory
2022-01-26T02:43:08.8680553Z    15 | #include "UnicycleTrajectoryGenerator.h"
2022-01-26T02:43:08.8680803Z       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2022-01-26T02:43:08.8681023Z compilation terminated.
2022-01-26T02:43:08.8681378Z [15/23] Building CXX object CMakeFiles/UnicyclePlanner.dir/src/DCMTrajectoryGeneratorHelper.cpp.o
2022-01-26T02:43:08.8681804Z [16/23] Building CXX object CMakeFiles/UnicyclePlanner.dir/src/FreeSpaceEllipse.cpp.o
~~~

traversaro, 2022-01-26:
@S-Dafarra do you think it could be related to https://github.com/robotology/unicycle-footstep-planner/pull/47 ?

S-Dafarra, 2022-01-26:
Sorry about that. Fixing it

S-Dafarra, 2022-01-26:
I should have fixed it. I released a new tag

traversaro, 2022-01-26:
Great, thanks!

closed 2022-01-26 by traversaro