robotology/robotology-superbuild #1008: Unable to find `soxr.h` (from conda-forge dependencies installation on Manjaro)

HosameldinMohamed, 2022-01-31:
Very similar to #1002. And with the same setup procedure.

Probably `conda-forge` dependencies are missing `soxr`.

```
/home/hosameldin/dev/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/SoundFilters.cpp:21:10: fatal error: soxr.h: No such file or directory
   21 | #include <soxr.h>
```

But in this case, the header `soxr.h` is available Conda in the channel [`HCC`](https://anaconda.org/hcc/soxr). If this package is fine maybe we add it to the [documentation](https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation).

HosameldinMohamed, 2022-01-31:
CC @traversaro

traversaro, 2022-02-01:
Thanks @HosameldinMohamed for reporting the issue! Not that probably this is due to https://github.com/robotology/yarp/pull/2665, and soxr is not installed even in apt/homebrew/vcpkg .  For what regards the conda package, I would avoid to depend on packages that are not part of `conda-forge` (or stricly related channels like `robostack` or `robotology`). The reason is that packages in other channels may not be compatible with the one from conda-forge, for example because they duplicate some dependency, or because they do not support some platforms supported by us and conda-forge. For example, the HCC channel does not support Windows and macOS. As soxr is quite a simple CMake-based project, it should not be difficult to add it on conda-forge.

HosameldinMohamed, 2022-02-01:
> As soxr is quite a simple CMake-based project, it should not be difficult to add it on conda-forge, taking the example from https://github.com/conda-forge/staged-recipes/pull/15408/files or similar simple CMake-package recipes.

Cool! I'll give it a try!

traversaro, 2022-02-01:
> > As soxr is quite a simple CMake-based project, it should not be difficult to add it on conda-forge, taking the example from https://github.com/conda-forge/staged-recipes/pull/15408/files or similar simple CMake-package recipes.
> 
> Cool! I'll give it a try!

Great, let me know if you need an hand.

traversaro, 2022-02-03:
Cool!

> The one thing I don't understand is how to add a command to extract the package?

That is already done automatically by the `conda build` command before calling the `build.sh` script. In my experience, the easiest way to test that everything works is to open the PR and inspect the CI results. Do not worry, until you mention a reviewers group nobody will check a PR on `staged-recipes`, so there is no risk of losing someone's time.

HosameldinMohamed, 2022-03-23:
PR https://github.com/conda-forge/staged-recipes/pull/17935

It is passing the linter checks, it has issues with the LICENSE file location and also the test files.

HosameldinMohamed, 2022-04-08:
PR https://github.com/conda-forge/staged-recipes/pull/17935 is merged 🎉 

[`soxr`](https://sourceforge.net/projects/soxr/) is packaged by Conda via `conda-forge` channel https://anaconda.org/conda-forge/soxr

traversaro, 2022-04-08:
Great, feel free to open a PR adding the dependency, thanks!

HosameldinMohamed, 2022-04-08:
I installed it and compiled YARP. OK!

HosameldinMohamed, 2022-04-08:
> Great, feel free to open a PR adding the dependency, thanks!

Opened https://github.com/robotology/robotology-superbuild/pull/1087

closed 2022-04-08 by traversaro