robotology/robotology-superbuild #1012: python packages installed by ROBOTOLOGY_USES_PYTHON are not listed by pip list --verbose

traversaro, 2022-02-02:
If one installs python packages such as idyntree, yarp or casadi, loads the `setup.sh` script tha sets the `PYTHONPATH` env variable, and then runs `pip list --verbose`, the packages are not listed. This is problematic because it means that `pip` is not aware of the packages, so for example if someone runs `pip install .` for a package that requires `casadi` (see for example https://github.com/ami-iit/ADAM), the `casadi` package frop pypi is **silently** installed, instead of using the one from the superbuild. This is particularly problematic as the casadi from `pypi` has statically linked copy of ipopt that may conflict with the one used in the superbuild. 

For example, if the casadi from the superbuild uses an ipopt recompiled to support ma27, if the casadi from pypi is installed, ma27 may be not supported anymore.

traversaro, 2022-02-02:
For example, currently installing ADAM in a system with the robotology-superbuild enabled results in:
~~~
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting jaxlib
  Using cached jaxlib-0.1.76-cp38-none-manylinux2010_x86_64.whl (65.2 MB)
Collecting numpy>=1.20
  Using cached numpy-1.22.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.8 MB)
Processing /home/traversaro/.cache/pip/wheels/f5/29/21/51db27425657863937ea10cd93fcf1e6c7d7b59902c2e3259f/urdf_parser_py-0.0.4-py3-none-any.whl
Collecting matplotlib
  Using cached matplotlib-3.5.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (11.3 MB)
Requirement already satisfied: scipy in /usr/lib/python3/dist-packages (from ADAM==0.0.1) (1.3.3)
Collecting casadi
  Using cached casadi-3.5.5-cp38-none-manylinux1_x86_64.whl (34.2 MB)
Collecting torch
  Using cached torch-1.10.2-cp38-cp38-manylinux1_x86_64.whl (881.9 MB)
Processing /home/traversaro/.cache/pip/wheels/d1/97/06/063433f9bb61c4877aede35bb3dfc95026a547996e4181b3db/jax-0.2.28-py3-none-any.whl
Collecting prettytable
  Using cached prettytable-3.0.0-py3-none-any.whl (24 kB)
Collecting absl-py
  Using cached absl_py-1.0.0-py3-none-any.whl (126 kB)
Collecting flatbuffers<3.0,>=1.12
  Using cached flatbuffers-2.0-py2.py3-none-any.whl (26 kB)
~~~

traversaro, 2022-02-02:
Everything seems related to the directory `lib/python3.8/site-packages/casadi-3.5.5.dist-info/`, if I rename it the package listed by pip list changes:
~~~
traversaro@IITICUBLAP257:~/.local/lib/python3.8/site-packages$ mv casadi-3.5.5.dist-info/ this_is_not_casadi-3.5.5.dist-info/
traversaro@IITICUBLAP257:~/.local/lib/python3.8/site-packages$ pip list --verbose | grep casadi
this-is-not-casadi     3.5.5                /home/traversaro/.local/lib/python3.8/site-packages pip
~~~

traversaro, 2022-02-02:
Quoting from https://github.com/conda-forge/idyntree-feedstock/pull/8#issuecomment-946461042 : 
> even if perhaps not ideal as it just installs an empty METADATA file, not fully compliant to https://packaging.python.org/specifications/recording-installed-packages/ and https://packaging.python.org/specifications/core-metadata/#core-metadata where at least a non-empty `METADATA` with the `Metadata-Version`, `Name` and  `Version` fiels are required

traversaro, 2022-02-02:
Indeed, adding a directory named `casadi-3.5.5.dist-info` in `install/lib/python3/dist-packages` with an empty `METADATA` file (just the empty directory was not enough) is sufficient to make pip aware of the installed casadi:
~~~
traversaro@IITICUBLAP257:~/ADAM$ pip install .
Processing /home/traversaro/ADAM
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting matplotlib
  Using cached matplotlib-3.5.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (11.3 MB)
Collecting torch
  Using cached torch-1.10.2-cp38-cp38-manylinux1_x86_64.whl (881.9 MB)
Requirement already satisfied: casadi in /home/traversaro/robotology-superbuild/build_ipopt/install/lib/python3/dist-packages (from ADAM==0.0.1) (3.5.5)
~~~

Probably we can just modify CMake of the involved projects to install that file in that directory. Even if an empty `METADATA` may work for now, this may be an implementation detail, so to be compliant with https://packaging.python.org/specifications/recording-installed-packages/ and https://packaging.python.org/specifications/core-metadata/#core-metadata  we may add `Metadata-Version` `Name` and `Version` fields to it.

traversaro, 2022-02-02:
@Giulero @CarlottaSartore did you ever experienced this problem of the PyPI casadi being installed when you wanted to use the robotology-superbuild/local casadi instead?

traversaro, 2022-02-02:
Related PRs:
* https://github.com/robotology/idyntree/pull/963
* https://github.com/ami-iit/casadi/pull/1
*

traversaro, 2022-02-03:
> @Giulero @CarlottaSartore did you ever experienced this problem of the PyPI casadi being installed when you wanted to use the robotology-superbuild/local casadi instead?

Not with `casadi`, but @fabiodinatale and @CarlottaSartore experienced **exactly** this problem. They had a superbuild with iDynTree with irrlicht enabled in the superbuild, but then they installed https://github.com/icub-tech-iit/urdf-modifiers with `pip install "urdf-modifiers @ git+https://github.com/icub-tech-iit/urdf-modifiers"`, and that installed idyntree via PyPI because that library depends on `idyntree`, see https://github.com/icub-tech-iit/urdf-modifiers/blob/23d5517413e1ce765ab5857130f0b30f588b0430/setup.cfg#L14 . However, the idyntree installed via PyPI does not have Irrlicht enabled (see https://github.com/robotology/idyntree/blob/208a28561d8cba1e0d06cc8b9df962c15a15b35b/setup.py#L58). https://github.com/robotology/idyntree/pull/963 should solve this problem as long as idyntree is installed via the superbuild, but it is quite easy to do this problem by following the default instructions of the README fyi @Nicogene @GrmanRodriguez  .

Giulero, 2022-02-03:
Yes exactly! I did not experience the problem with `casadi`. I might be wrong (probably I am), but it seems that the `casadi` interface with the solvers "searches" in the whole system, and hence it's independent of the version and env you install it (does it mean that we can't use different versions of solvers?).

On the other hand, I had problems when calling the `irrlicht` visualizer when `idyntree` was installed via `pip`.

traversaro, 2022-02-03:
> Yes exactly! I did not experience the problem with `casadi`. I might be wrong (probably I am), but it seems that the `casadi` interface with the solvers "searches" in the whole system, and hence [...]

Mhhh so you mean that even if you installed `casadi` from pip, you were able to use the `casadi`/`ipopt` used and compiled with the superbuild? That sounds strange, and bay be just a lucky combination of `ABI`/`LD_LIBRARY_PATH` settings. 

> On the other hand, I had problems when calling the `irrlicht` visualizer when `idyntree` was installed via `pip`.

Ack! So how do you tipically manage this before the fixes in https://github.com/robotology/idyntree/pull/963? You did installed idyntree via robotology-superbuild and then installed `ADAM` with `pip install --no-deps .` instead of `pip install .`?

traversaro, 2022-02-03:
@Giulero we quickly debugged this with @CarlottaSartore and I think we found the answer of why you did not encounter problem. @CarlottaSartore had indeed PyPI's idyntree installed in her virtualenv:
~~~
pip list --verbose | grep idyntree
idyntree          4.3.1   /home/carlotta/iit_ws/element_hardware-intelligence/elementEnv/lib/python3.8/site-packages pip
~~~

However, it seems that the idyntree from the robotology-superbuild takes the precedence, i.e. executing in python:
~~~
import idyntree
idyntree.__file__
/home/carlotta/iit_ws/robotology-superbuild/build/install/lib/python3/dist-packages/idyntree/__init__.py
~~~

@Giulero my guess that also in your case if you run:
~~~
import casadi
casadi.__file__
~~~

For some reason the superbuild's casadi will be found.

traversaro, 2022-02-08:
I checked in a recently clone and built apt-based superbuild, and indeed it seems to be working fine:
~~~
traversaro@IITICUBLAP257:~/test/robotology-superbuild/build$ pip list --verbose
Package                Version              Location                                                                            Installer
---------------------- -------------------- ----------------------------------------------------------------------------------- ---------
appdirs                1.4.3                /usr/lib/python3/dist-packages
atomicwrites           1.1.5                /usr/lib/python3/dist-packages
attrs                  19.3.0               /usr/lib/python3/dist-packages
Automat                0.8.0                /usr/lib/python3/dist-packages
blinker                1.4                  /usr/lib/python3/dist-packages
casadi                 3.5.5                /home/traversaro/test/robotology-superbuild/build/install/lib/python3/dist-packages cmake
certifi                2019.11.28           /usr/lib/python3/dist-packages
chardet                3.0.4                /usr/lib/python3/dist-packages
Click                  7.0                  /usr/lib/python3/dist-packages
cloud-init             21.4                 /usr/lib/python3/dist-packages
colorama               0.4.3                /usr/lib/python3/dist-packages
command-not-found      0.3                  /usr/lib/python3/dist-packages
configobj              5.0.6                /usr/lib/python3/dist-packages
constantly             15.1.0               /usr/lib/python3/dist-packages
cryptography           2.8                  /usr/lib/python3/dist-packages
cupshelpers            1.0                  /usr/lib/python3/dist-packages
dbus-python            1.2.16               /usr/lib/python3/dist-packages
decorator              4.4.2                /usr/lib/python3/dist-packages
defer                  1.0.6                /usr/lib/python3/dist-packages
distlib                0.3.0                /usr/lib/python3/dist-packages
distro                 1.4.0                /usr/lib/python3/dist-packages
distro-info            0.23ubuntu1          /usr/lib/python3/dist-packages
docutils               0.16                 /usr/lib/python3/dist-packages
entrypoints            0.3                  /usr/lib/python3/dist-packages
filelock               3.0.12               /usr/lib/python3/dist-packages
freetype-py            2.1.0.post1          /usr/lib/python3/dist-packages
future                 0.18.2               /usr/lib/python3/dist-packages
httplib2               0.14.0               /usr/lib/python3/dist-packages
hyperlink              19.0.0               /usr/lib/python3/dist-packages
idna                   2.8                  /usr/lib/python3/dist-packages
idyntree               4.3.1                /home/traversaro/test/robotology-superbuild/build/install/lib/python3/dist-packages cmake
imageio                2.4.1                /usr/lib/python3/dist-packages
importlib-metadata     1.5.0                /usr/lib/python3/dist-packages
incremental            16.10.1              /usr/lib/python3/dist-packages
~~~

traversaro, 2022-02-08:
We just need to:
* [ ] Send PR upstream to casadi
* [ ] Send PR to fork and upstream to manif
* [ ] Send PR to blf

And then all the Python projects handled by the superbuild should be ok.

traversaro, 2022-02-08:
>  Send PR upstream to casadi

Done: https://github.com/casadi/casadi/pull/2878 .

>  Send PR to blf

Done: https://github.com/ami-iit/bipedal-locomotion-framework/pull/508

traversaro, 2022-02-08:
>  Send PR to fork and upstream to manif

Done: https://github.com/artivis/manif/pull/243 (just upstream, as there is no manifpy package on PyPI it is not problematic to wait).

traversaro, 2022-02-08:
As we already submitted all the PR that we could submit upstream, and given that the problem is fixed at least for the packages for which the problem emerged in the first place (`casadi` and `idyntree`) I think we can close this issue.

closed 2022-02-08 by traversaro

traversaro, 2022-12-14:
> As we already submitted all the PR that we could submit upstream, and given that the problem is fixed at least for the packages for which the problem emerged in the first place (`casadi` and `idyntree`) I think we can close this issue.

Actually, we forgot to bump the casadi version to actually fix this. This is finally fixed in https://github.com/robotology/robotology-superbuild/pull/1318 . 

fyi @diegoferigo