robotology/robotology-superbuild #1020: Conda CI Failure Mid February 2022

traversaro, 2022-02-11:
Error:
~~~
2022-02-11T02:12:39.2514067Z [command]/usr/share/miniconda3/condabin/mamba create --name test
2022-02-11T02:12:39.8237286Z ##[warning]
# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/usr/share/miniconda3/lib/python3.9/site-packages/conda/exceptions.py", line 1080, in __call__
        return func(*args, **kwargs)
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 934, in exception_converter
        raise e
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 928, in exception_converter
        exit_code = _wrapped_main(*args, **kwargs)
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 882, in _wrapped_main
        result = do_call(args, p)
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 748, in do_call
        exit_code = create(args, parser)
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 624, in create
        return install(args, parser, "create")
      File "/usr/share/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 577, in install
        if not transaction.prompt():
    UnboundLocalError: local variable 'transaction' referenced before assignment

`$ /usr/share/miniconda3/condabin/mamba create --name test`

  environment variables:
                 CIO_TEST=<not set>
                    CONDA=/usr/share/miniconda3
           CONDA_PKGS_DIR=/home/runner/conda_pkgs_dir
               CONDA_ROOT=/usr/share/miniconda3
           CURL_CA_BUNDLE=<not set>
      DEPLOYMENT_BASEPATH=/opt/runner
        GITHUB_EVENT_PATH=/home/runner/work/_temp/_github_workflow/event.json
              GITHUB_PATH=/home/runner/work/_temp/_runner_file_commands/add_path_f573de62-4c7d-4
                          ce0-8c28-fa0c51a3d380
                     PATH=/usr/share/miniconda3/condabin:/home/linuxbrew/.linuxbrew/bin:/home/li
                          nuxbrew/.linuxbrew/sbin:/home/runner/.local/bin:/opt/pipx_bin:/home/ru
                          nner/.cargo/bin:/home/runner/.config/composer/vendor/bin:/usr/local/.g
                          hcup/bin:/home/runner/.dotnet/tools:/snap/bin:/usr/local/sbin:/usr/loc
                          al/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap
                          /bin
       REQUESTS_CA_BUNDLE=<not set>
        SELENIUM_JAR_PATH=/usr/share/java/selenium-server.jar
            SSL_CERT_FILE=<not set>
               SWIFT_PATH=/usr/share/swift/usr/bin

     active environment : None
       user config file : /home/runner/.condarc
 populated config files : /usr/share/miniconda3/.condarc
                          /home/runner/.condarc
          conda version : 4.11.0
    conda-build version : not installed
         python version : 3.9.10.final.0
       virtual packages : __linux=5.11.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /usr/share/miniconda3  (writable)
      conda av data dir : /usr/share/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/runner/conda_pkgs_dir
       envs directories : /usr/share/miniconda3/envs
                          /home/runner/.conda/envs
               platform : linux-64
             user-agent : conda/4.11.0 requests/2.27.1 CPython/3.9.10 Linux/5.11.0-1028-azure ubuntu/20.04.3 glibc/2.31
                UID:GID : 1001:121
             netrc file : None
           offline mode : False
[... 92 lines cut ...]
2022-02-11T02:12:39.8309675Z         mamba (0.21.0) supported by @QuantStack
2022-02-11T02:12:39.8309831Z 
2022-02-11T02:12:39.8310044Z         GitHub:  https://github.com/mamba-org/mamba
2022-02-11T02:12:39.8310331Z         Twitter: https://twitter.com/QuantStack
2022-02-11T02:12:39.8310496Z 
2022-02-11T02:12:39.8310753Z █████████████████████████████████████████████████████████████
2022-02-11T02:12:39.8310908Z 
2022-02-11T02:12:39.8310913Z 
2022-02-11T02:12:39.8310996Z Looking for: []
2022-02-11T02:12:39.8311105Z 
2022-02-11T02:12:39.8807320Z ##[endgroup]
2022-02-11T02:12:39.8807736Z ##[endgroup]
2022-02-11T02:12:39.8810575Z ##[error]The process '/usr/share/miniconda3/condabin/mamba' failed with exit code 1
2022-02-11T02:12:39.9661635Z Post job cleanup.
2022-02-11T02:12:40.0840388Z [command]/usr/bin/git version
2022-02-11T02:12:40.0888697Z git version 2.35.1
2022-02-11T02:12:40.0920449Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2022-02-11T02:12:40.0957842Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2022-02-11T02:12:40.1183875Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2022-02-11T02:12:40.1209541Z http.https://github.com/.extraheader
2022-02-11T02:12:40.1218040Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2022-02-11T02:12:40.1254563Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
~~~

Seems a mamba regressions.

traversaro, 2022-02-11:
Probably it was https://github.com/mamba-org/mamba/releases/tag/2022.02.07 that took a few days to propagate to conda-forge due to https://github.com/conda-forge/status/issues/125 .

traversaro, 2022-02-11:
It was indeed tracked by https://github.com/mamba-org/mamba/pull/1460, a release should appear soon to fix this.

Huitongtong, 2022-02-14:
> Probably it was https://github.com/mamba-org/mamba/releases/tag/2022.02.07 that took a few days to propagate to conda-forge due to [conda-forge/status#125](https://github.com/conda-forge/status/issues/125) .

Any updates on this issue? I still have the same error for creating 'mamba create -n robotologyenv'  environment.

traversaro, 2022-02-14:
A new mamba release is arriving soon (see https://github.com/mamba-org/mamba/commit/b3aef48677b08e540e7ec4b08c03d6099ff48969). 

However, the workaround is quite simple, just create the environment with some packages instead of completly empty, such as:
~~~
mamba create -n robotologyenv cmake
~~~

traversaro, 2022-02-16:
Workaround added in https://github.com/robotology/robotology-superbuild/pull/1026 and removed in https://github.com/robotology/robotology-superbuild/pull/1034 .

closed 2022-02-16 by traversaro