robotology/robotology-superbuild #1057: Compilation issues when installing the superbuild on windows from sources with conda

gabrielenava, 2022-03-06:
I followed the installation instructions in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation to install the superbuild on WIndows 10 with options `ROBOTOLOGY_USES_MATLAB` and `ROBOTOLOGY_USES_DYNAMICS` both set to `ON`. I am using "Visual Studio 17 2022" as compiler.

I noted that when after I compile the superbuild, several repos are created inside the `src` folder, but they are actually empty (no git clone performed).   The compilation apparently terminates with no final error messages.

The empty repos are: `whole-body-controllers`, `WBToolbox`, `idyntree-yarp-tools`, `matlab-whole-body-simulator`, `whole-body-estimators`.

 I have then checked the full output of the terminal from the moment I run `cmake`, and I spotted two error messages when compiling YARP:

```cmake

 yarpcar.vcxproj -> C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\YARP\bin\Debug\yarpcard.dll
    main.cpp
main.obj : error LNK2019: unresolved external symbol "void __cdecl cv::cvtColor(class cv::debug_build_guard::_InputArra
y const &,class cv::debug_build_guard::_OutputArray const &,int,int)" (?cvtColor@cv@@YAXAEBV_InputArray@debug_build_gua
rd@1@AEBV_OutputArray@31@HH@Z) referenced in function "class cv::Mat __cdecl yarp::cv::toCvMat<unsigned char>(class yar
p::sig::ImageOf<unsigned char> &)" (??$toCvMat@E@cv@yarp@@YA?AVMat@0@AEAV?$ImageOf@E@sig@1@@Z) [C:\Users\gnava\Document
s\GitHub\robotology-superbuild\build\src\YARP\src\yarpdatadumper\yarpdatadumper.vcxproj] [C:\Users\gnava\Documents\GitH
ub\robotology-superbuild\build\YARP.vcxproj]
C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\YARP\bin\Debug\yarpdatadumper.exe : fatal error LNK1120
: 1 unresolved externals [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\YARP\src\yarpdatadumper\yarpd
atadumper.vcxproj] [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\YARP.vcxproj]
```

```cmake

    yarpviz.vcxproj -> C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\YARP\bin\Debug\yarpviz.exe
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(245,5): e
rror MSB8066: Custom build for 'C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\4e857eb5f4144b08
06934a8d20c23dd4\YARP-skip-update.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\4e857eb5f
4144b0806934a8d20c23dd4\YARP-configure.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\4e85
7eb5f4144b0806934a8d20c23dd4\YARP-build.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\4e8
57eb5f4144b0806934a8d20c23dd4\YARP-install.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\
52ef495c0705661a9f3052d9a7d91d6b\YARP-complete.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFi
les\c858d14d3983228d6c6013fc82657248\YARP.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\CMakeLists.txt' ex
ited with code 1. [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\YARP.vcxproj]

```

I am not sure these errors are somehow related with the missing repos or they are unrelated errors. What I can say is that for a project, I also tried to compile the superbuild with a custom project tag. Then, if I re-install the superbuild from scratch, with this custom tag, iDynTree also does not compile correctly and the source folder remains empty. Again I see errors similar to the above when compiling YARP while I don't see other info related to iDynTree.

In this second case I also have all terminal messages dumped into a file: [log_cmake_superbuild](https://github.com/robotology/robotology-superbuild/files/8193340/log.txt)

Here the tags of iDynTree for my custom project:

```yaml

  iDynTree:
    type: git
    url: https://github.com/ami-iit/idyntree-hde-fork.git
    version: 6fec756ec9dca59b1ef266bd807d2be3759d7d3c
```

traversaro, 2022-03-06:
Are you sure you are using the compilation command provided in the docs? It seems that you are compiling in Debug mode, while with conda-forge only Release or RelWithDebInfo are supposerted.

traversaro, 2022-03-06:
> I am not sure these errors are somehow related with the missing repos or they are unrelated errors.

The missing repos are a result of this errors: as the compilation fails with YARP, the rest of the repos that depend on YARP are not cloned nor compiled.

gabrielenava, 2022-03-07:
>Are you sure you are using the compilation command provided in the docs, i.e. cmake --build . --config Release? It seems that you are compiling in Debug mode, while with conda-forge only Release or RelWithDebInfo are supported.

indeed, when I have recorded the log files I was experimenting variations and I just run `cmake --build .` without `--config Release`. I will repeat the test to verify if the errors disappear.

gabrielenava, 2022-03-08:
I have deleted by build folder and compiled again my custom project with `cmake --build . --config Release`, but this did not solve the issue: src folders are still empty.

I didn't see big differences looking at the terminal messages: at this point, I am wondering if for some reasons I am still compiling the superbuild on `debug`. However this time I followed the instructions.

Here the log (from a second run of cmake after creating the new build folder): 
[compilation_superbuild.txt](https://github.com/robotology/robotology-superbuild/files/8208617/compilation_superbuild.txt)

traversaro, 2022-03-09:
There is an error when compiling `icub-main`:
~~~
LINK : fatal error LNK1181: cannot open input file 'glut.lib' [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\ICUB\src\tools\iCubGui\src\iCubGui.vcxproj] [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\ICUB.vcxproj]
    Automatic MOC for target iCubSkinGui
    Generating qrc_qml.cpp
    Building Custom Rule C:/Users/gnava/Documents/GitHub/robotology-superbuild/src/ICUB/src/tools/iCubSkinGui/src/CMakeLists.txt
    main.cpp
    qrc_qml.cpp
    mocs_compilation_Release.cpp
    Generating Code...
    iCubSkinGui.vcxproj -> C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\ICUB\bin\Release\iCubSkinGui.exe
    Building Custom Rule C:/Users/gnava/Documents/GitHub/robotology-superbuild/src/ICUB/src/modules/iKinCartesianSolver/CMakeLists.txt
    main.cpp
~~~

My guess is that for some reason you did installed the faulty CMake version 3.22.2 (see https://github.com/robotology/robotology-superbuild/issues/1032). This is quite strange, as we had `cmake=3.21` in the installation instructions until CMake 3.22.3 came out (see https://github.com/robotology/robotology-superbuild/pull/1054/files). You should be able to update CMake easily in the environment with `mamba update cmake` and this should solve the problem.

traversaro, 2022-03-09:
If instead you have an external CMake that you are using, make sure that you have either CMake 3.21 or 3.22.3, **not** 3.22.0, 3.22.1 or 3.22.2.

gabrielenava, 2022-03-09:
I confirm that the issue was due to my call to `cmake-gui`, which was opening `cmake 3.22.2` for compilation of the superbuild. After uninstalling the external cmake (maybe a bit aggressive but effective) I was able to compile the suberbuild with no errors, and all repos are installed correctly. 

However I still have failures when I try to compile the superbuild with the custom tags of my external project, for some reasons I don't find IPOPT. With the default installation I don't have this error. But maybe it is due to issues with the project tags.

```cmake

 CMake Error at C:/Users/gnava/AppData/Local/mambaforge/envs/robsub/Library/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
    Could NOT find IPOPT (missing: IPOPT_LIBRARIES)
  Call Stack (most recent call first):
    C:/Users/gnava/AppData/Local/mambaforge/envs/robsub/Library/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
    cmake/FindIPOPT.cmake:235 (find_package_handle_standard_args)
    cmake/iDynTreeDependencies.cmake:22 (find_package)
    cmake/iDynTreeDependencies.cmake:54 (idyntree_handle_dependency)
    CMakeLists.txt:43 (include)
  
  
  -- Configuring incomplete, errors occurred!
  See also "C:/Users/gnava/Documents/GitHub/robotology-superbuild/build/src/iDynTree/CMakeFiles/CMakeOutput.log".
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(245,5): error MSB8066: Custom build for 'C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\683be23d4af06ff1632512d848f53f1f\iDynTree-skip-update.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\683be23d4af06ff1632512d848f53f1f\iDynTree-configure.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\683be23d4af06ff1632512d848f53f1f\iDynTree-build.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\683be23d4af06ff1632512d848f53f1f\iDynTree-install.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\52ef495c0705661a9f3052d9a7d91d6b\iDynTree-complete.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\c858d14d3983228d6c6013fc82657248\iDynTree.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\CMakeLists.txt' exited with code 1. [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\iDynTree.vcxproj]

```

Here the log file:

[superbuild_cmake_mytag.txt](https://github.com/robotology/robotology-superbuild/files/8214668/superbuild_cmake_mytag.txt)

traversaro, 2022-03-09:
I guess that the iDynTree fork is missing the fix from https://github.com/robotology/idyntree/pull/793 . (fyi @RiccardoGrieco)

gabrielenava, 2022-03-09:
I have corrected the iDynTree branch in my custom project to 

Now there are errors while compiling ICUB:

```cmake

 Building Custom Rule C:/Users/gnava/Documents/GitHub/robotology-superbuild/src/ICUB/src/tools/iCubGui/src/CMakeLists.txt
LINK : fatal error LNK1181: cannot open input file 'glut.lib' [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\ICUB\src\tools\iCubGui\src\iCubGui.
vcxproj] [C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\ICUB.vcxproj]

    wholeBodyPlayer.vcxproj -> C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\src\ICUB\bin\Release\wholeBodyPlayer.exe
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(245,5): error MSB8066: Custom build for 'C:\Users\gn
ava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\d55ad0ff2ef80710cc9169c3428084b0\ICUB-skip-update.rule;C:\Users\gnava\Documents\GitHub\robotology-supe
rbuild\build\CMakeFiles\d55ad0ff2ef80710cc9169c3428084b0\ICUB-configure.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\d55ad0ff2ef807
10cc9169c3428084b0\ICUB-build.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\d55ad0ff2ef80710cc9169c3428084b0\ICUB-install.rule;C:\Us
ers\gnava\Documents\GitHub\robotology-superbuild\build\CMakeFiles\52ef495c0705661a9f3052d9a7d91d6b\ICUB-complete.rule;C:\Users\gnava\Documents\GitHub\robotology-s
uperbuild\build\CMakeFiles\c858d14d3983228d6c6013fc82657248\ICUB.rule;C:\Users\gnava\Documents\GitHub\robotology-superbuild\CMakeLists.txt' exited with code 1. [C
:\Users\gnava\Documents\GitHub\robotology-superbuild\build\ICUB.vcxproj]

```

they seem more similar to the previous compilation issue when I was using a wrong compiler, but now I have only the compiler provided by mamba.

gabrielenava, 2022-03-09:
meanwhile I have opened a PR to update the documentation: https://github.com/robotology/robotology-superbuild/pull/1059

I think that the problem I am having with the custom project tags could be debugged also offline: the default compilation works to the original problem of this issue is solved. Other issues seem related to the choice of tags in the custom `yaml` file.

gabrielenava, 2022-03-09:
cc @traversaro

traversaro, 2022-03-09:
> they seem more similar to the previous compilation issue when I was using a wrong compiler, but now I have only the compiler provided by mamba.

Probably it is just a caching problem. Can you try to delete `robotology-superbuild/build/src/ICUB` and try again?

gabrielenava, 2022-03-10:
I have tried but still the error persists, I link here my custom `yaml`file:

[aeps.zip](https://github.com/robotology/robotology-superbuild/files/8222968/aeps.zip)

traversaro, 2022-03-10:
Just for history, the commit that is missing is https://github.com/robotology/icub-main/pull/756 . The v1.20.0 version of icub-main does not have it (see https://github.com/robotology/icub-main/blob/v1.20.0/src/tools/iCubGui/src/CMakeLists.txt), while v1.21.0 has it: https://github.com/robotology/icub-main/blob/v1.21.0/src/tools/iCubGui/src/CMakeLists.txt . Just for fun, we could try if v1.21.0 works fine (even if we could get other incompatibilities).

traversaro, 2022-03-18:
@gabrielenava can we close the issue as it was related to the use of old releases?

gabrielenava, 2022-03-22:
sure, closing!

closed 2022-03-22 by gabrielenava