robotology/robotology-superbuild #1066: pass CMake options by -DYCM_EP_ADDITIONAL_CMAKE_ARGS

Nicolasr3, 2022-03-21:
Hi all,

I'm trying to build a docker image of robotology-superbuild with some additional options, like **yarpmod_grabber** and **yarpmod_opencv_grabber**.

For that, I passed to the variable **ROBOTOLOGY_PARAMS** the option **-DYCM_EP_ADDITIONAL_CMAKE_ARGS**.
But I'm having some trouble because at the end of the building the opencv_grabber  and grabber are not installed.

This is the first part of docker images:

[image]

When I'm trying to pass more than one options by the **-DYCM_EP_ADDITIONAL_CMAKE_ARGS**, it seems that something get lost  during the parsing. Maybe because I'm trying to add a string inside another string.

Above I did some  echo of the variables **ROBOTOLOGY_PARAMS** and **YCM_EP_ADDITIONAL_ARGS** during the building of the image, and seems that are setted correctly:
 
[image]

However, after the installation this is the result of the CMakeCache.txt:

[image]

The **YCM_EP_ADDITIONAL_CMAKE_ARGS** is not completed, in fact is missing the second part of the options.

Do you know if I'm doing wrong some parsing during the configuration of those ENV variables inside the dockerfile?

The robotology-superbuild version used is: *2022.02.0*.
The docker bash version is: *GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)*.

At the moment, by passing the option **-DYCM_EP_ADDITIONAL_CMAKE_ARGS** directly inside the cmake command seems works.

[image]

Thanks in advance

Nicola

traversaro, 2022-03-21:
Hi @Nicolasr3 ! I tried to rename the issue to be more descriptive of the problem.

Did you tried to test without docker the case in which all the CMake options are passed via a single env variable `ROBOTOLOGY_PARAMS` to understand if the same issue appears in that case and if the issue is actually unrelated to docker?

Pinging also @Nicogene @AlexAntn @vtikha  that may know more on this.

Nicogene, 2022-03-21:
> I'm trying to build a docker image of robotology-superbuild with some additional options, like yarpmod_grabber and yarpmod_opencv_grabber

I am not 100% sure but `grabber` should be deprecated, the new name of that device is `frameGrabber_nws_yarp`
For `opencv_grabber` instead it has recently added in the superbuild, see:

https://github.com/robotology/robotology-superbuild/commit/1c36b787b2fcd88120c5c58bdec46f35cd84a124

You can use the iCub Tech docker images directly
the image `icubteamcode/superbuild:master-stable_sources` (or `binaries`) is 2 days old, then it should include `opencv_grabber`