robotology/robotology-superbuild #1082: Yarp icub arm hand  gripping  object through  python

ArfnZl, 2022-04-02:
[image]

I have griped ball with full joints but when i change the position of shoulder the ball does not move it remains at same position and position of arm is also changed

ArfnZl, 2022-04-02:
@traversaro please reply why object is not moving through  gripping  from one position  to another

ArfnZl, 2022-04-03:
@traversaro please  reply

traversaro, 2022-04-04:
Hello @ArfnZl, 
  * As already answered in https://github.com/robotology/robotology-superbuild/issues/1056#issuecomment-1060677969, in general the robotology-superbuild issue tracker is just to report issues on robotology-superbuild itself, for general Q&A on iCub/YARP you can answer questions on https://github.com/robotology/community/discussions/categories/q-a . 
  * The image that you tried to upload in the original issue was not uploaded. In general, all your questions seems to lack a lot of context that is necessary to help you. In general, whenever you report an unexpected behaviour it is important to report it with a so-called "Minimal, Reproducible Example", see https://stackoverflow.com/help/minimal-reproducible-example .
  * There is any reason why you need an answer so quickly? If you are in any kind of emergency feel free to drop a mail at the address you can find at https://traversaro.github.io/ . If there is no emergency, it is tipically considered not highly respectful of random strangers on the internet that may be willing to help  (even if there is nothing that forces them to do so!) to continuously ping them every ~12h. Beside being not respectful, you just risk decrease the probability that they take their time to answer you. : ]

traversaro, 2022-04-13:
@ArfnZl feel free to open new issues if you think it make sense. I think I can close this issue.

closed 2022-04-13 by traversaro