robotology/robotology-superbuild #1098: Icub object gripping  problem

ArfnZl, 2022-04-17:
Dear i am trying to grip the ball but when hand touches it slipped kindly guide how to resolve this slipping problem... I am using pyhon anaconda with i cub...i have tried many DoF

ArfnZl, 2022-04-17:
https://user-images.githubusercontent.com/83446699/163709888-a5a16a5d-162c-4909-aac8-04c9f9b58e4c.mp4

ArfnZl, 2022-04-17:
@traversaro please  sir

traversaro, 2022-04-19:
Hi @ArfnZl, as I already explained in https://github.com/robotology/robotology-superbuild/issues/1082#issuecomment-1087230288: 
* This repo is not meant for general question on iCub, for that post on https://github.com/robotology/community/discussions 
* If you want to help on some case, it is much more probable that you will receive a reply if you provide the code you are working on.

closed 2022-04-19 by traversaro