robotology/robotology-superbuild #1107: Troubles with Unstable tag (devel)

MatteoPerotto, 2022-05-02:
Hi all, I'm about to perform some tests on the mk4 hand and I need the ``devel`` version of the superbuild. I simply pulled, created a new build folder (the older is renamed for back-up purposes) and configured the ``ROBOTOLOGY_PROJECT_TAGS`` to Unstable. 

However, in building, the following error occurs:

<pre>[ 18%] <font color="#3465A4"><b>Performing build step for &apos;YARP&apos;</b></font>
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [CMakeFiles/YARP.dir/build.make:108: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:459: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:95: all] Error 2
</pre>

Do you have any suggestion on which mistakes I'm making?

traversaro, 2022-05-02:
I would advise to use a completly new robotology-superbuild repo, as the content of all the `src` directory will be changed as well, not just creating a different build folder. Furthermore, remember to remove from the `.bashrc` or any similar fail the sourcing of the `setup.sh` of the non-unstable build.

traversaro, 2022-05-02:
@MatteoPerotto if you are at IIT-CRIS I can also pass by and see if there are any remaining problems.

MatteoPerotto, 2022-05-02:
@traversaro thank you, I now try to clone a new robotology-superbuild repo and then, if I have problems, I write you. Thanks for the help

traversaro, 2022-05-09:
@MatteoPerotto did you solved your problems? If yes, feel free to close the issue, thanks!