robotology/robotology-superbuild #1108: Homebrew/Xcode/Apple Clang CI Failure Begin May 2022

traversaro, 2022-05-02:
There is a CI failure related to Homebrew/Xcode/Apple Clang (see https://github.com/robotology/robotology-superbuild/pull/1106):

~~~
2022-05-02T13:35:32.9065980Z [2/7] Building CXX object src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o
2022-05-02T13:35:32.9167660Z FAILED: src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o 
2022-05-02T13:35:32.9272560Z /Applications/Xcode_12.4.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -Dyarp_telemetryDeviceDumper_EXPORTS -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/YARP_telemetry/src/libYARP_telemetry/src -isystem /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/local/include -isystem /usr/local/include/eigen3 -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wunused -Wvla -Wmissing-include-dirs -Wreorder -Wsizeof-pointer-memaccess -Woverloaded-virtual -Wtautological-undefined-compare -Wmismatched-new-delete -Wparentheses-equality -Wundef -Wredundant-decls -Wunknown-pragmas -Wunused-result -Wc++2a-compat -Wheader-guard -Wignored-attributes -Wnewline-eof -Wdangling-else -Wgcc-compat -Wmicrosoft-exists -Wstatic-inline-explicit-instantiation -Wmisleading-indentation -Wtautological-compare -Winconsistent-missing-override -Wnull-conversion  -Wno-unused-parameter  -Wdeprecated-declarations  -O3 -DNDEBUG -isysroot /Applications/Xcode_12.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -fPIC -std=c++17 -MD -MT src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o -MF src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o.d -o src/telemetryDeviceDumper/CMakeFiles/yarp_telemetryDeviceDumper.dir/yarp_plugin_telemetryDeviceDumper.cpp.o -c /Users/runner/work/robotology-superbuild/robotology-superbuild/build/src/YARP_telemetry/src/telemetryDeviceDumper/yarp_plugin_telemetryDeviceDumper.cpp
2022-05-02T13:35:32.9377080Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/build/src/YARP_telemetry/src/telemetryDeviceDumper/yarp_plugin_telemetryDeviceDumper.cpp:10:
2022-05-02T13:35:32.9478970Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/YARP_telemetry/src/telemetryDeviceDumper/TelemetryDeviceDumper.h:27:
2022-05-02T13:35:32.9580710Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/YARP_telemetry/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h:271:14: warning: unused variable 'ok' [-Wunused-variable]
2022-05-02T13:35:32.9682030Z         bool ok = configure(_bufferConfig);
2022-05-02T13:35:32.9783130Z              ^
2022-05-02T13:35:32.9886090Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/YARP_telemetry/src/libYARP_telemetry/src/yarp/telemetry/experimental/BufferManager.h:559:21: error: use 'template' keyword to treat 'createMatioCppConvertFunction' as a dependent template name
2022-05-02T13:35:32.9988070Z         bufferInfo->createMatioCppConvertFunction<T>();
2022-05-02T13:35:33.0089190Z                     ^
2022-05-02T13:35:33.0190320Z                     template 
2022-05-02T13:35:33.0291420Z 1 warning and 1 error generated.
~~~

traversaro, 2022-05-02:
I guess it is related to https://github.com/robotology/yarp-telemetry/pull/168 . @S-Dafarra @Nicogene 

Personally I am quite inclined to progressively support less and less Homebrew (see https://github.com/robotology/robotology-superbuild/issues/842), but anyhow you may be interested to this.

S-Dafarra, 2022-05-02:
I can try to fix that

traversaro, 2022-05-03:
It seems that https://github.com/robotology/yarp-telemetry/pull/170 fixed the issue, see https://github.com/robotology/robotology-superbuild/actions/runs/2262650254 .

closed 2022-05-03 by traversaro