robotology/robotology-superbuild #1113: Add OpenVR and OpenXR related repositories to `ROBOTOLOGY_ENABLE_TELEOPERATION` component

traversaro, 2022-05-09:
It probably make sense to add to the ROBOTOLOGY_ENABLE_TELEOPERATION component the repos:
* https://github.com/ami-iit/yarp-device-openxrheadset
* https://github.com/ami-iit/yarp-openvr-trackers

The core depenendencies of this repo are:
* https://github.com/ValveSoftware/openvr (but we actually use https://github.com/ami-iit/openvr for the cmake fixes that upstream is not interested in merging)
* https://github.com/KhronosGroup/OpenXR-SDK

traversaro, 2022-05-09:
For both OpenXR and OpenVR, some packages are available in Ubuntu, but either they are not available in 20.04, or the version in 20.04 is too old. For this reason, we probably just include them in the superbuild.

traversaro, 2022-05-10:
We also need recent tags on these repos:
* [ ] https://github.com/ami-iit/openvr 
* [ ] https://github.com/ami-iit/yarp-device-openxrheadset
* [ ] https://github.com/ami-iit/yarp-openvr-trackers

traversaro, 2022-05-10:
I also asked to @S-Dafarra why we can't just use OpenXR and drop altogether OpenVR. He explained me that at the momement the limitation is at the SteamVR level, as using the trackers without visor requires https://www.khronos.org/registry/OpenXR/specs/1.0/html/xrspec.html#XR_MND_headless and SteamVR still does not implement it (see https://steamcommunity.com/app/250820/discussions/8/3121550424355682585/).
Even if this extension was implemented, we would need to migrate https://github.com/ami-iit/yarp-openvr-trackers to OpenXR.

S-Dafarra, 2022-05-10:
> We also need recent tags on these repos:
> 
> * [x]  https://github.com/ami-iit/openvr --> https://github.com/ami-iit/openvr/releases/tag/v1.16.8.1
> * [ ]  https://github.com/ami-iit/yarp-device-openxrheadset
> * [ ]  https://github.com/ami-iit/yarp-openvr-trackers

Regarding ``yarp-openvr-trackers`` we can do it also now. For ``yarp-device-openxrheadset`` I wanted to land at least https://github.com/ami-iit/yarp-device-openxrheadset/pull/18 first.

closed 2022-05-15 by traversaro