robotology/robotology-superbuild #1114: Deprecate ROBOTOLOGY_USES_OCULUS_SDK  and yarp-device-ovrheadset

traversaro, 2022-05-10:
Since we are working on switching the VR support to OpenXR (https://github.com/robotology/robotology-superbuild/issues/1113) I was wondering if we can deprecate and eventually remove the `ROBOTOLOGY_USES_OCULUS_SDK` option and the [`yarp-device-ovrheadset` repo](https://github.com/robotology/yarp-device-ovrheadset). 
Is anyone still using this option @S-Dafarra @randaz81

S-Dafarra, 2022-05-10:
From the AMI teleoperation side, it was used last time for a demo in September. There is no drop-in replacement for ``yarp-device-ovrheadset``, but I agree in removing the option from the superbuild.

traversaro, 2022-09-28:
> I think it is fair to announce that it is deprecated in 2022.05 and remove it in 2022.08 .

We never announced that, but I think there is no problem in removing it.

closed 2023-07-06 by traversaro