robotology/robotology-superbuild #1118: Unable to reinstall robot-log-visualizer after the first installation

GiulioRomualdi, 2022-05-14:
I installed the superbuild, from sources, in a clean **ubuntu 22.04** installation 
I tried to run the `robot-log-visualizer` and I got the following error
```
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
```
So I strated to debug the code. I added some prints here and there and I tried to reinstall it by typing `make robot-log-visualizer` in `robotology-superbuild/build` However I noticed that the new version of the code is not installed
```
── robot_log_visualizer
│   ├── file_reader
│   │   ├── __init__.py
│   │   ├── __pycache__
│   │   │   ├── __init__.cpython-310.pyc
│   │   │   └── signal_provider.cpython-310.pyc
│   │   └── signal_provider.py
│   ├── __init__.py
│   ├── __main__.py
│   ├── plotter
│   │   ├── __init__.py
│   │   ├── matplotlib_viewer_canvas.py
│   │   └── __pycache__
│   │       ├── __init__.cpython-310.pyc
│   │       └── matplotlib_viewer_canvas.cpython-310.pyc
│   ├── __pycache__
│   │   ├── __init__.cpython-310.pyc
│   │   └── __main__.cpython-310.pyc
│   ├── robot_visualizer
│   │   ├── __init__.py
│   │   ├── meshcat_provider.py
│   │   ├── meshcat_visualizer.py
│   │   └── __pycache__
│   │       ├── __init__.cpython-310.pyc
│   │       ├── meshcat_provider.cpython-310.pyc
│   │       └── meshcat_visualizer.cpython-310.pyc
│   ├── ui
│   │   ├── autogenerated
│   │   │   ├── about.py
│   │   │   ├── __init__.py
│   │   │   ├── plot_tab.py
│   │   │   ├── __pycache__
│   │   │   │   ├── about.cpython-310.pyc
│   │   │   │   ├── __init__.cpython-310.pyc
│   │   │   │   ├── plot_tab.cpython-310.pyc
│   │   │   │   └── visualizer.cpython-310.pyc
│   │   │   └── visualizer.py
│   │   ├── gui.py
│   │   ├── __init__.py
│   │   ├── misc
│   │   │   ├── about.ui
│   │   │   ├── close-circle-outline.svg
│   │   │   ├── document-text-outline.svg
│   │   │   ├── folder-open-outline.svg
│   │   │   ├── game-controller-outline.svg
│   │   │   ├── icon.png
│   │   │   ├── __init__.py
│   │   │   ├── pause-outline.svg
│   │   │   ├── play-outline.svg
│   │   │   ├── plot_tab.ui
│   │   │   ├── __pycache__
│   │   │   │   └── __init__.cpython-310.pyc
│   │   │   ├── terminal-outline.svg
│   │   │   ├── videocam-outline.svg
│   │   │   └── visualizer.ui
│   │   ├── plot_item.py
│   │   └── __pycache__
│   │       ├── gui.cpython-310.pyc
│   │       ├── __init__.cpython-310.pyc
│   │       └── plot_item.cpython-310.pyc
│   └── utils
│       ├── __init__.py
│       ├── __pycache__
│       │   ├── __init__.cpython-310.pyc
│       │   └── utils.cpython-310.pyc
│       └── utils.py
├── robot_log_visualizer-0.1.5.dist-info
│   ├── direct_url.json
│   ├── entry_points.txt
│   ├── INSTALLER
│   ├── LICENSE
│   ├── METADATA
│   ├── RECORD
│   ├── REQUESTED
│   ├── top_level.txt
│   └── WHEEL
├── robot_log_visualizer-0.1.6.dev0+dirty.dist-info
│   ├── direct_url.json
│   ├── entry_points.txt
│   ├── INSTALLER
│   ├── LICENSE
│   ├── METADATA
│   ├── RECORD
│   ├── REQUESTED
│   ├── top_level.txt
│   └── WHEEL
```

If I remove `robot_log_visualizer-0.1.6.dev0+dirty.dist-info` `robot_log_visualizer-0.1.5.dist-info` and `robot_log_visualizer` and I tried to reinstall it again, `robot_log_visualizer` still contain the version of previously installed by the superbuild

GiulioRomualdi, 2022-05-14:
This is the output of `git status` in `~/robot-code/robotology-superbuild/src/robot-log-visualizer`
```
 git status 
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   robot_log_visualizer/ui/gui.py
```

traversaro, 2022-05-15:
Interesting, the command that is executed to install robot-log-visualizer is the one in https://github.com/robotology/robotology-superbuild/blob/d99fabafb240f19ff318683e2d49d1882e03f15b/cmake/RobSupPurePythonYCMEPHelper.cmake#L53 : 
~~~
${Python3_EXECUTABLE} -m pip install --upgrade --no-deps --target=${YCM_EP_INSTALL_DIR}/${ROBSUB_PYTHON_INSTALL_DIR} -VV <SOURCE_DIR>
~~~

What is the output of `make robot-log-visualizer` ? Perhaps the `INSTALL` command should uninstall the installed version before  installing the new one?

traversaro, 2022-05-15:
Related PR: https://github.com/robotology/robotology-superbuild/pull/1069 .

traversaro, 2022-05-15:
By the way, I did most of my tests on Ubuntu 20.04, so this could be a pip version-related issue.

traversaro, 2022-05-15:
@GiulioRomualdi can you check if https://github.com/robotology/robotology-superbuild/pull/1120 solves your problems?

traversaro, 2023-04-26:
I think https://github.com/robotology/robotology-superbuild/pull/1120 solved the problem, if not feel free to re-open or open a new issue, thanks!