robotology/robotology-superbuild #1119: Pressure tactile sensors is not reading valued

ArfnZl, 2022-05-15:
Hey.. I am working  on icub yarp python.. I want to add tactile  pressure  sensors for measure gripping  force...i also added tactile sensors which is giving 0,255 valued only... My project is gripping object but object is not slipping 
Problem............. 
Please  provide me python code of pressure tactile  sensors for measure  gripping  force... 
[image]
[image]

ArfnZl, 2022-05-15:
[image]

ArfnZl, 2022-05-15:
[image]

traversaro, 2022-05-15:
Hi @ArfnZl, as I already explained in https://github.com/robotology/robotology-superbuild/issues/1098#issuecomment-1102618534 and https://github.com/robotology/robotology-superbuild/issues/1082#issuecomment-1087230288 : 

> * This repo is not meant for general question on iCub, for that post on https://github.com/robotology/community/discussions 
> * If you want to help on some case, it is much more probable that you will receive a reply if you provide the code you are working on.

So, please open an issue in https://github.com/robotology/community/discussions and provide the code you are running to simplifying the life of people that may want to help you, thanks.

closed 2022-05-15 by traversaro