robotology/robotology-superbuild #1125: Great CI Failure of Devel Branch of end of May 2022

traversaro, 2022-05-19:
As predicted in several issues (https://github.com/robotology/whole-body-estimators/issues/150, https://github.com/robotology/walking-teleoperation/issues/95, https://github.com/robotology/walking-controllers/issues/110) the devel branch is now broken due to the widespread use of deprecated YARP methods after the merge of https://github.com/robotology/yarp/pull/2833 .

traversaro, 2022-05-19:
Components currently affected:

| Component | ✔️ if it compiles fine, ❌ if it does not compile |
|:----------:|:----------------:|
|  ROBOTOLOGY_ENABLE_CORE | ✔️ |
| ROBOTOLOGY_ENABLE_ROBOT_TESTING | ✔️ |
| ROBOTOLOGY_ENABLE_DYNAMICS |❌ | 
| ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS | ❌ |
| ROBOTOLOGY_ENABLE_ICUB_HEAD | ✔️ | 
| ROBOTOLOGY_ENABLE_ICUB_BASIC_DEMOS | ✔️ |
| ROBOTOLOGY_ENABLE_TELEOPERATION |❌ | 
| ROBOTOLOGY_ENABLE_HUMAN_DYNAMICS | ✔️ | 
| ROBOTOLOGY_ENABLE_EVENT_DRIVEN | ✔️ |

prashanthr05, 2022-05-19:
> As predicted in several issues ([robotology/whole-body-estimators#150](https://github.com/robotology/whole-body-estimators/issues/150), [robotology/walking-teleoperation#95](https://github.com/robotology/walking-teleoperation/issues/95), [robotology/walking-controllers#110](https://github.com/robotology/walking-controllers/issues/110)) the Unstable PROJECT_TAGS is now broken due to the widespread use of deprecated YARP methods after the merge of [robotology/yarp#2833](https://github.com/robotology/yarp/pull/2833) .
> 
> See for example https://github.com/robotology/robotology-superbuild/actions/runs/2349354178 .

Opened a PR in https://github.com/robotology/whole-body-estimators/pull/151.

traversaro, 2022-05-19:
From the latest manually triggered job (https://github.com/robotology/robotology-superbuild/actions/runs/2350411064), apparently also icub-main is failing due to https://github.com/robotology/yarp/pull/2837 : 
~~~
2022-05-19T08:19:07.9169677Z [75/466] Building CXX object src/libraries/icubmod/dragonfly2/CMakeFiles/dragonfly2.dir/common/DragonflyDeviceDriver2.cpp.o
2022-05-19T08:19:07.9170338Z FAILED: src/libraries/icubmod/dragonfly2/CMakeFiles/dragonfly2.dir/common/DragonflyDeviceDriver2.cpp.o 
2022-05-19T08:19:07.9173638Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -D_REENTRANT -Ddragonfly2_EXPORTS -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2 -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2/common -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2/linux -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/icubmod/dragonfly2/CMakeFiles/dragonfly2.dir/common/DragonflyDeviceDriver2.cpp.o -MF src/libraries/icubmod/dragonfly2/CMakeFiles/dragonfly2.dir/common/DragonflyDeviceDriver2.cpp.o.d -o src/libraries/icubmod/dragonfly2/CMakeFiles/dragonfly2.dir/common/DragonflyDeviceDriver2.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2/common/DragonflyDeviceDriver2.cpp
2022-05-19T08:19:07.9176071Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2/common/DragonflyDeviceDriver2.cpp:10:
2022-05-19T08:19:07.9176926Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/dragonfly2/common/DragonflyDeviceDriver2.h:290:34: error: expected class-name before ',' token
2022-05-19T08:19:07.9177616Z   290 |     public IFrameGrabberControls2,
2022-05-19T08:19:07.9177882Z       |                                  ^
~~~

That means that we can put all ❌ in the table in https://github.com/robotology/robotology-superbuild/issues/1125#issuecomment-1131316130 . :D

traversaro, 2022-05-20:
New nightly build (https://github.com/robotology/robotology-superbuild/actions/runs/2355753933), new failure:
~~~
2022-05-20T03:32:37.2037017Z [309/424] Performing build step for 'blocktest-yarp-plugins'
2022-05-20T03:32:37.2038322Z FAILED: src/blocktest-yarp-plugins/CMakeFiles/YCMStamp/blocktest-yarp-plugins-build /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktest-yarp-plugins/CMakeFiles/YCMStamp/blocktest-yarp-plugins-build 
2022-05-20T03:32:37.2040030Z cd /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktest-yarp-plugins && /usr/share/miniconda3/envs/test/bin/cmake --build . && /usr/share/miniconda3/envs/test/bin/cmake -E touch /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktest-yarp-plugins/CMakeFiles/YCMStamp/blocktest-yarp-plugins-build
2022-05-20T03:32:37.2040943Z [1/31] Generating code from thrift/clock_rpc.thrift
2022-05-20T03:32:37.2041458Z [2/31] Building CXX object CMakeFiles/ClockRpc.dir/thrift/ClockServer.cpp.o
2022-05-20T03:32:37.2041932Z [3/31] Linking CXX static library libClockRpc.a
2022-05-20T03:32:37.2042457Z [4/31] Building CXX object CMakeFiles/blocktestyarpplugins.dir/src/actionApplyForce.cpp.o
2022-05-20T03:32:37.2043034Z FAILED: CMakeFiles/blocktestyarpplugins.dir/src/actionApplyForce.cpp.o 
2022-05-20T03:32:37.2047411Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -Dblocktestyarpplugins_EXPORTS -I/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktest-yarp-plugins -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src -I/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktest-yarp-plugins/thrift -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/blocktestcore -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -Wall -Wextra -O3 -DNDEBUG -fPIC -std=gnu++17 -MD -MT CMakeFiles/blocktestyarpplugins.dir/src/actionApplyForce.cpp.o -MF CMakeFiles/blocktestyarpplugins.dir/src/actionApplyForce.cpp.o.d -o CMakeFiles/blocktestyarpplugins.dir/src/actionApplyForce.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp
2022-05-20T03:32:37.2051006Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp: In member function 'virtual BlockTestCore::execution YarpActions::ActionApplyForce::execute(const BlockTestCore::TestRepetitions&)':
2022-05-20T03:32:37.2052357Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:63:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2053029Z    63 |     cmd.addDouble(std::atof(tokenized[1].c_str()));
2022-05-20T03:32:37.2053416Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2054246Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:64:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2054909Z    64 |     cmd.addDouble(std::atof(tokenized[2].c_str()));
2022-05-20T03:32:37.2055282Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2056174Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:65:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2056729Z    65 |     cmd.addDouble(std::atof(tokenized[3].c_str()));
2022-05-20T03:32:37.2057146Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2057936Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:66:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2058608Z    66 |     cmd.addDouble(std::atof(tokenized[4].c_str()));
2022-05-20T03:32:37.2058981Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2059805Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:67:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2060466Z    67 |     cmd.addDouble(std::atof(tokenized[5].c_str()));
2022-05-20T03:32:37.2060829Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2061725Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:68:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2062384Z    68 |     cmd.addDouble(std::atof(tokenized[6].c_str()));
2022-05-20T03:32:37.2062680Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2063566Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionApplyForce.cpp:69:9: error: 'class yarp::os::Bottle' has no member named 'addDouble'
2022-05-20T03:32:37.2064225Z    69 |     cmd.addDouble(std::atof(tokenized[7].c_str()));
2022-05-20T03:32:37.2064598Z       |         ^~~~~~~~~
2022-05-20T03:32:37.2065091Z [5/31] Building CXX object CMakeFiles/blocktestyarpplugins.dir/src/actionCanRead.cpp.o
[... cut ...]
2022-05-20T03:32:37.2067084Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionCanWrite.cpp: In member function 'virtual BlockTestCore::execution YarpActions::ActionCanWrite::execute(const BlockTestCore::TestRepetitions&)':
2022-05-20T03:32:37.2068401Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/blocktest-yarp-plugins/src/actionCanWrite.cpp:41:10: warning: unused variable 'exists' [-Wunused-variable]
2022-05-20T03:32:37.2069060Z    41 |     auto exists {true};
2022-05-20T03:32:37.2069320Z       |          ^~~~~~
2022-05-20T03:32:37.2069675Z ninja: build stopped: subcommand failed.
~~~

traversaro, 2022-05-20:
> New nightly build (https://github.com/robotology/robotology-superbuild/actions/runs/2355753933), new failure:

Upstream issue: https://github.com/robotology/blocktest-yarp-plugins/issues/11 .

traversaro, 2022-05-20:
> > New nightly build (https://github.com/robotology/robotology-superbuild/actions/runs/2355753933), new failure:
> 
> Upstream issue: [robotology/blocktest-yarp-plugins#11](https://github.com/robotology/blocktest-yarp-plugins/issues/11) .

Fixed by @triccyx in , thanks a lot! 

New CI run (https://github.com/robotology/robotology-superbuild/runs/6521602062), new failure:
~~~
2022-05-20T09:28:06.9673044Z [2/31] Building CXX object src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/stateExtendedReader.cpp.o
2022-05-20T09:28:06.9673728Z FAILED: src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/stateExtendedReader.cpp.o 
2022-05-20T09:28:06.9676751Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DYARP_STATIC_PLUGIN -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -MD -MT src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/stateExtendedReader.cpp.o -MF src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/stateExtendedReader.cpp.o.d -o src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/stateExtendedReader.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp
2022-05-20T09:28:06.9679278Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp: In member function 'bool StateExtendedInputPort::getLastSingle(int, int, double*, yarp::os::Stamp&, double&)':
2022-05-20T09:28:06.9680659Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp:157:132: error: 'yarp::os::Vocab' has not been declared
2022-05-20T09:28:06.9681616Z   157 |                 yError() << "RemoteControlBoard internal error while reading data. Cannot get 'single' data of type " << yarp::os::Vocab::decode(field);
2022-05-20T09:28:06.9682345Z       |                                                                                                                                    ^~~~~
2022-05-20T09:28:06.9683341Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp: In member function 'bool StateExtendedInputPort::getLastSingle(int, int, int*, yarp::os::Stamp&, double&)':
2022-05-20T09:28:06.9684651Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp:190:132: error: 'yarp::os::Vocab' has not been declared
2022-05-20T09:28:06.9685587Z   190 |                 yError() << "RemoteControlBoard internal error while reading data. Cannot get 'single' data of type " << yarp::os::Vocab::decode(field);
2022-05-20T09:28:06.9686309Z       |                                                                                                                                    ^~~~~
2022-05-20T09:28:06.9687291Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp: In member function 'bool StateExtendedInputPort::getLastVector(int, double*, yarp::os::Stamp&, double&)':
2022-05-20T09:28:06.9688638Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp:257:132: error: 'yarp::os::Vocab' has not been declared
2022-05-20T09:28:06.9689675Z   257 |                 yError() << "RemoteControlBoard internal error while reading data. Cannot get 'vector' data of type " << yarp::os::Vocab::decode(field);
2022-05-20T09:28:06.9690705Z       |                                                                                                                                    ^~~~~
2022-05-20T09:28:06.9691526Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp: In member function 'bool StateExtendedInputPort::getLastVector(int, int*, yarp::os::Stamp&, double&)':
2022-05-20T09:28:06.9692513Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/ReadOnlyRemoteControlBoard/stateExtendedReader.cpp:290:132: error: 'yarp::os::Vocab' has not been declared
2022-05-20T09:28:06.9693306Z   290 |                 yError() << "RemoteControlBoard internal error while reading data. Cannot get 'vector' data of type " << yarp::os::Vocab::decode(field);
2022-05-20T09:28:06.9693783Z       |                                                                                                                                    ^~~~~
2022-05-20T09:28:06.9694296Z [3/31] Building CXX object src/lib/ReadOnlyRemoteControlBoard/CMakeFiles/readonlyremotecontrolboard.dir/yarp_plugin_ReadOnlyRemoteControlBoardLib_readonlyremotecontrolboard.cpp.o
[... cut ...] id you mean 'asVocab32'?
2022-05-20T09:28:06.9702636Z   181 |             JointTypeEnum axisVocab = static_cast<JointTypeEnum>(axis.asList()->get(1).asVocab());
2022-05-20T09:28:06.9702986Z       |                                                                                        ^~~~~~~
2022-05-20T09:28:06.9703278Z       |                                                                                        asVocab32
2022-05-20T09:28:06.9703539Z ninja: build stopped: subcommand failed.
~~~

traversaro, 2022-05-20:
New failure, new fix: https://github.com/robotology/forcetorque-yarp-devices/pull/30 .

traversaro, 2022-05-23:
Other error:
~~~
2022-05-23T03:55:05.8809863Z [8/56] Building CXX object devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/HumanStateProvider.cpp.o
2022-05-23T03:55:05.8810879Z FAILED: devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/HumanStateProvider.cpp.o 
2022-05-23T03:55:05.8816692Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DHumanStateProvider_EXPORTS -DUSING_DEPRECATED_UPPERCASE_YARP_OS_TARGET -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider/SYSTEM -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/interfaces/IHumanState -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/HumanDynamicsEstimationLibrary/utils/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/HumanDynamicsEstimationLibrary/algorithms/include -isystem /usr/share/miniconda3/envs/test/include/eigen3 -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/osqp -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -std=gnu++17 -MD -MT devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/HumanStateProvider.cpp.o -MF devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/HumanStateProvider.cpp.o.d -o devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/HumanStateProvider.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider/HumanStateProvider.cpp
2022-05-23T03:55:05.8822577Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider/HumanStateProvider.cpp: In member function 'virtual bool hde::devices::HumanStateProvider::impl::CmdParser::read(yarp::os::ConnectionReader&)':
2022-05-23T03:55:05.8824228Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider/HumanStateProvider.cpp:320:26: error: 'class yarp::os::Bottle' has no member named 'addVocab'; did you mean 'addVocab32'?
2022-05-23T03:55:05.8824848Z   320 |                 response.addVocab(yarp::os::Vocab::encode("many"));
2022-05-23T03:55:05.8825145Z       |                          ^~~~~~~~
2022-05-23T03:55:05.8825388Z       |                          addVocab32
2022-05-23T03:55:05.8826224Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanStateProvider/HumanStateProvider.cpp:320:45: error: 'yarp::os::Vocab' has not been declared
2022-05-23T03:55:05.8826776Z   320 |                 response.addVocab(yarp::os::Vocab::encode("many"));
2022-05-23T03:55:05.8827081Z       |                                             ^~~~~
2022-05-23T03:55:05.8827506Z [9/56] Building CXX object devices/HumanWrenchProvider/CMakeFiles/HumanWrenchProvider.dir/HumanWrenchProvider.cpp.o
2022-05-23T03:55:05.8828863Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanWrenchProvider/HumanWrenchProvider.cpp: In member function 'virtual int hde::devices::HumanWrenchProvider::getState(int)':
2022-05-23T03:55:05.8830236Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/HumanDynamicsEstimation/devices/HumanWrenchProvider/HumanWrenchProvider.cpp:884:1: warning: control reaches end of non-void function [-Wreturn-type]
2022-05-23T03:55:05.8830791Z   884 | }
2022-05-23T03:55:05.8830999Z       | ^
2022-05-23T03:55:05.8831384Z [10/56] Building CXX object devices/HumanStateProvider/CMakeFiles/HumanStateProvider.dir/IKWorkerPool.cpp.o
2022-05-23T03:55:05.8831824Z ninja: build stopped: subcommand failed.
~~~

traversaro, 2022-05-25:
With the last fix, the devel failure are finally fixed, see https://github.com/robotology/robotology-superbuild/runs/6592944460 . I think we can close the issue.