robotology/robotology-superbuild #1127: YARP_telemetry --> robometry migrations

traversaro, 2022-05-23:
If we want to avoid installing both YARP_telemetry and robometry as a transition period, we may need a bit of coordination among the different actions for the transition (see https://github.com/robotology/robotology-superbuild/pull/1111). For each action I also indicated

In particular, the actions that I think are required are:
* A1: Release first version of robometry (even 0.9.0 or  0.9.9, no need for 1.0.0) @Nicogene and update the `LatestReleases.yaml` file in https://github.com/robotology/robotology-superbuild/pull/1111 
* A2: Merge https://github.com/robotology/robotology-superbuild/pull/1111 @traversaro 
* A3: Merge https://github.com/robotology/wearables/pull/155 @lrapetti
* A4: Release new version of wearables containing https://github.com/robotology/wearables/pull/155 @lrapetti
* A5: Merge https://github.com/ami-iit/bipedal-locomotion-framework/pull/533 @S-Dafarra (as @GiulioRomualdi is at ICRA) 
* A6: Release new version of blf containing https://github.com/ami-iit/bipedal-locomotion-framework/pull/533 @S-Dafarra (as @GiulioRomualdi is at ICRA)
* A7: Update robotology-superbuild with releases done in A4 and A5

We are not particularly in hurry for A3/A4 and A5/A6, but notice that until A3 and A5 are completed, wearables and blf in the superbuild (stable project tags) will compile without telemetry support.

traversaro, 2022-05-24:
> > [ ] A5: Merge [YarpLoggerDevice: ported to robometry ami-iit/bipedal-locomotion-framework#533](https://github.com/ami-iit/bipedal-locomotion-framework/pull/533) @S-Dafarra (as @GiulioRomualdi is at ICRA)
> 
> The problem is that this could cause conflicts with [ami-iit/bipedal-locomotion-framework#516](https://github.com/ami-iit/bipedal-locomotion-framework/pull/516). On this I would need the @GiulioRomualdi input before proceeding

What is the time frame for that? As mentioned in https://github.com/robotology/robotology-superbuild/pull/1111#issuecomment-1120858103, we can always install libYARP_telemetry and robometry side by side until all downstream libraries have been updated.

S-Dafarra, 2022-05-24:
Related comment: https://github.com/ami-iit/bipedal-locomotion-framework/pull/516#issuecomment-1134930056 

At this point maybe having the both is probably preferable :thinking:

Nicogene, 2022-05-24:
Since we decided to keep both YARP_telemetry and robometry side-by-side(https://github.com/robotology/robotology-superbuild/issues/1127#issuecomment-1135724099) I removed the points A5 and A6 since are not required anymore

traversaro, 2022-05-24:
> Since we decided to keep both YARP_telemetry and robometry side-by-side([#1127 (comment)](https://github.com/robotology/robotology-superbuild/issues/1127#issuecomment-1135724099)) I removed the points A5 and A6 since are not required anymore

A5 and A6 are not required for 2022.05, but I think we can leave them and also add A8 for when blf will be migrated.

traversaro, 2022-08-01:
blf 0.8.0 with robometry support has been released, probably we can add the dependency of blf on robometry and remove libYARP_telemetry ?

Nicogene, 2022-08-03:
Yes! I can do it today

traversaro, 2022-08-03:
Great!

PR #1229 (2022-08-04): Remove YARP_telemetry
It has been deprecated in favour of robometry

It fixes #1127
closed 2022-08-04 by traversaro