robotology/robotology-superbuild #1132: Configuring error with CMake 3.23

diegoferigo, 2022-05-26:
I have update CMake in a clean docker image from the kitware's PPA:

```
matlab@4d251a1ccceb:~$ dpkg -l | grep cmake
ii  cmake                                    3.23.2-0kitware1ubuntu20.04.1     amd64        cross-platform, open-source make system
ii  cmake-curses-gui                         3.23.2-0kitware1ubuntu20.04.1     amd64        curses based user interface for CMake (ccmake)
ii  cmake-data                               3.23.2-0kitware1ubuntu20.04.1     all          CMake data files (modules, templates and documentation)
```

The superbuild fails to configure with the following error:

```
[...]
# -- YCM not found. Bootstrapping it.
# -- Downloading YCMEPHelper.cmake - Attempt 1 of 3
# -- Downloading YCMEPHelper.cmake - SUCCESS
# -- Downloading CMakeParseArguments.cmake - Attempt 1 of 3
# -- Downloading CMakeParseArguments.cmake - SUCCESS
# -- Downloading ExternalProject.cmake - Attempt 1 of 3
# -- Downloading ExternalProject.cmake - SUCCESS
# -- Downloading StyleGITHUB.cmake - Attempt 1 of 3
# -- Downloading StyleGITHUB.cmake - SUCCESS
# CMake Error: File /usr/share/cmake-3.23/Modules/RepositoryInfo.txt.in does not exist.
# CMake Error at build/CMakeFiles/ExternalProject.cmake:2517 (configure_file):
#   configure_file Problem configuring file
# Call Stack (most recent call first):
#   build/CMakeFiles/ExternalProject.cmake:3264 (_ep_add_download_command)
#   build/CMakeFiles/YCMEPHelper.cmake:1038 (ExternalProject_Add)
#   build/CMakeFiles/YCMEPHelper.cmake:1346 (ycm_ep_helper)
#   cmake/YCMBootstrap.cmake:102 (ycm_bootstrap)
#   CMakeLists.txt:56 (include)
# 
# 
# -- Performing download step (git clone) for 'YCM'
# CMake Error at build/CMakeFiles/YCMEPHelper.cmake:1358 (message):
#   Cannot clone YCM repository (1)
# Call Stack (most recent call first):
#   cmake/YCMBootstrap.cmake:102 (ycm_bootstrap)
#   CMakeLists.txt:56 (include)
```

I didn't investigate the origin of the problem, but I noticed the following:

```
matlab@4d251a1ccceb:~$ find /usr/share/ | grep RepositoryInfo
/usr/share/cmake-3.23/Modules/ExternalProject/RepositoryInfo.txt.in

matlab@4d251a1ccceb:~$ dpkg -S /usr/share/cmake-3.23/Modules/ExternalProject/RepositoryInfo.txt.in
cmake-data: /usr/share/cmake-3.23/Modules/ExternalProject/RepositoryInfo.txt.in
```

So the file exists, but it's in another directory. A temporary workaround is the next command:

```
sudo ln -s /usr/share/cmake-3.23/Modules/ExternalProject/RepositoryInfo.txt.in /usr/share/cmake-3.23/Modules/
```

It's probably not directly related to the superbuild, but it's worth having this ticket here in case this problem occurs on other systems and people try to find for existing issues.

traversaro, 2022-05-26:
For reference, this is a duplicate of https://github.com/robotology/robotology-superbuild/issues/1074 and it was solved by https://github.com/robotology/ycm/pull/393 . If you are using a custom project tags, I think it probably make sense to at least bump the YCM version to 0.13.1 to avoid this problem.

traversaro, 2022-05-26:
Another "workaround" is to use the apt-provided CMake version for Ubuntu 20.04 , that is by the far the one that is more tested in that configuration.

diegoferigo, 2022-05-27:
Thanks for the references @traversaro! I didn't search extensively enough to find the previous issue. For reasons, I'm using an old tag of the superbuild from summer 2021, and maybe it bootstraps an older version of YCM.

Closing.

closed 2022-05-27 by diegoferigo