robotology/robotology-superbuild #1134: LatestReleases failure end of May 2022

traversaro, 2022-05-26:
Sincehttps://github.com/robotology/robotology-superbuild/pull/1129, `LatestReleases` is not compiling, due to several repos missing a latest release compatible with YARP 3.7 .

traversaro, 2022-05-26:
We are missing releases:
* walking-teleoperation (see https://github.com/robotology/robotology-superbuild/issues/1133#issuecomment-1139086851)
* https://github.com/ami-iit/bipedal-locomotion-framework/ https://github.com/robotology/walking-controllers @S-Dafarra @GiulioRomualdi 
* https://github.com/robotology/human-dynamics-estimation @lrapetti @RiccardoGrieco 
* https://github.com/robotology/whole-body-estimators @prashanthr05 @HosameldinMohamed 
* https://github.com/robotology/forcetorque-yarp-devices @RiccardoGrieco

I know this is a bit tedious task and everybody is busy with demos/conferences/etc/etc. For this reason, feel free to ask me support, I would be happy to backport the YARP 3.7 fixes to the master (or on the top of the latest releases, if you do not like to release master for whatever reason) and do a release with a patch version bump.

S-Dafarra, 2022-05-27:
For ``walking-teleoperation`` and ``walking-controllers`` I will make the new release after https://github.com/robotology/walking-controllers/pull/113 and https://github.com/robotology/walking-teleoperation/pull/99 are merged.

traversaro, 2022-05-27:
Great, for idyntree-yarp-tools I did https://github.com/robotology/idyntree-yarp-tools/releases/tag/v0.0.5 .

S-Dafarra, 2022-05-27:
``walking-controllers``: https://github.com/robotology/walking-controllers/releases/tag/v0.5.1

lrapetti, 2022-05-27:
> - [ ] https://github.com/robotology/human-dynamics-estimation @lrapetti @RiccardoGrieco

PR: https://github.com/robotology/human-dynamics-estimation/pull/303

S-Dafarra, 2022-05-27:
``walking-teleoperation``: https://github.com/robotology/walking-teleoperation/releases/tag/v1.3.0

traversaro, 2022-05-27:
https://github.com/robotology/whole-body-estimators/pull/152

traversaro, 2022-05-30:
Fixed by https://github.com/robotology/robotology-superbuild/pull/1136 .

closed 2022-05-30 by traversaro