robotology/robotology-superbuild #1152: Compilation failure with graphviz 4 on Windows

traversaro, 2022-06-09:
@S-Dafarra experienced a compilation error on Windows due to the fact that graphviz 4 was installed. Apparently the problem was solved by downgrading graphviz to 3. I wonder why we did not catch this error in the CI, but then I noticed that on CI we are compiling against graphviz 2.50 (see for example https://github.com/robotology/robotology-superbuild/runs/6766540385):
~~~
ace                       7.0.7                h0e60522_0    conda-forge
asio                      1.14.1               h0e60522_1    conda-forge
assimp                    5.2.4                hc2aa0de_0    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blosc                     1.21.1               h74325e0_3    conda-forge
boost                     1.74.0          py310hc781a3c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.5.18.1          h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
certifi                   2022.5.18.1     py310h5588dad_0    conda-forge
cfitsio                   4.0.0                hd67004f_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.2               h39d44d4_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
esdcan                    6.3.0                h57928b3_0    robotology
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.33.3          py310he2412df_0    conda-forge
fortran-compiler          1.4.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               h6676e37_8    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.11.0              hc81694e_0    conda-forge
geos                      3.10.1               h39d44d4_1    conda-forge
geotiff                   1.7.0                hc8731e1_5    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glfw                      3.3.7                h8ffe710_0    conda-forge
glm                       0.9.9.4              h1ad3211_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  2.50.0               hefbd956_1    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
h5py                      3.6.0           nompi_py310h00cbb18_100    conda-forge
harfbuzz                  3.1.1                hc601d6f_0    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
[... 156 lines cut ...]
vs2017_win-64             19.16.27033          hb90652a_6    conda-forge
vswhere                   3.0.3                h57928b3_0    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xerces-c                  3.2.3                h0e60522_5    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.12               h8ffe710_0    conda-forge
zstd                      1.5.2                h6255e5f_1    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

@S-Dafarra can you provide the environment in that machine to understand what is the difference?

S-Dafarra, 2022-06-09:
Here it is:
~~~
ace                       7.0.7                h0e60522_0    conda-forge
asio                      1.14.1               h0e60522_1    conda-forge
assimp                    5.2.4                hc2aa0de_0    conda-forge
blosc                     1.21.1               h74325e0_3    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.5.18.1          h5b45459_0    conda-forge
cairo                     1.16.0            h0ac17fb_1011    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
clang_variant             1.0                     default    conda-forge
clangdev                  5.0.0             h5aa907a_1012    conda-forge
cmake                     3.23.2               h39d44d4_0    conda-forge
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.4.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               h6676e37_9    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.11.0              h05b8ed6_5    conda-forge
geos                      3.10.3               h39d44d4_0    conda-forge
geotiff                   1.7.1                h38b14a8_1    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glfw                      3.3.7                h8ffe710_0    conda-forge
glm                       0.9.9.4              h1ad3211_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  3.0.0                had6c3a3_1    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gst-plugins-base          1.20.2               he07aa86_1    conda-forge
gstreamer                 1.20.2               hdff456e_1    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
harfbuzz                  4.3.0                h27de254_0    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
icu                       70.1                 h0e60522_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
imath                     3.1.5                h12d4b20_0    conda-forge
intel-openmp              2022.1.0          h57928b3_3787    conda-forge
ipopt                     3.14.6               hf6be2e5_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jasper                    2.0.33               h77af90b_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9e                   h8ffe710_1    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h8995ca9_4    conda-forge
[... 117 lines cut ...]
vc                        14.2                 hb210afc_5    conda-forge
vcstool                   0.3.0              pyhd8ed1ab_0    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vs2017_win-64             19.16.27038          h2e3bad8_2    conda-forge
vswhere                   3.0.3                h57928b3_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.12               h8ffe710_0    conda-forge
zstd                      1.5.2                h6255e5f_1    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

S-Dafarra, 2022-06-09:
``grphviz 3.0`` did not work either
```
87>LINK : warning LNK4217: symbol '??0QGVScene@@QEAA@AEBVQString@@PEAVQObject@@_N@Z (public: __cdecl QGVScene::QGVScene(class QString const &,class QObject *,bool))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"private: void __cdecl MainWindow::initScene(void)" (?initScene@MainWindow@@AEAAXXZ)'
87>LINK : warning LNK4217: symbol '??1QGVScene@@UEAA@XZ (public: virtual __cdecl QGVScene::~QGVScene(void))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: virtual void * __cdecl QGVScene::`scalar deleting destructor'(unsigned int)" (??_GQGVScene@@UEAAPEAXI@Z)'
87>LINK : warning LNK4217: symbol '?setGraphAttribute@QGVScene@@QEAAXAEBVQString@@0@Z (public: void __cdecl QGVScene::setGraphAttribute(class QString const &,class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?setNodeAttribute@QGVScene@@QEAAXAEBVQString@@0@Z (public: void __cdecl QGVScene::setNodeAttribute(class QString const &,class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?setEdgeAttribute@QGVScene@@QEAAXAEBVQString@@0@Z (public: void __cdecl QGVScene::setEdgeAttribute(class QString const &,class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?addNode@QGVScene@@QEAAPEAVQGVNode@@AEBVQString@@@Z (public: class QGVNode * __cdecl QGVScene::addNode(class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?addEdge@QGVScene@@QEAAPEAVQGVEdge@@PEAVQGVNode@@0AEBVQString@@@Z (public: class QGVEdge * __cdecl QGVScene::addEdge(class QGVNode *,class QGVNode *,class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?addSubGraph@QGVScene@@QEAAPEAVQGVSubGraph@@AEBVQString@@_N@Z (public: class QGVSubGraph * __cdecl QGVScene::addSubGraph(class QString const &,bool))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?applyLayout@QGVScene@@QEAAXXZ (public: void __cdecl QGVScene::applyLayout(void))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?clear@QGVScene@@QEAAXXZ (public: void __cdecl QGVScene::clear(void))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"private: void __cdecl MainWindow::initScene(void)" (?initScene@MainWindow@@AEAAXXZ)'
87>LINK : warning LNK4217: symbol '?enableBgGrid@QGVScene@@QEAAX_N@Z (public: void __cdecl QGVScene::enableBgGrid(bool))' defined in 'YARP_priv_qgvcore.lib(QGVScene.obj)' is imported by 'MainWindow.obj' in function '"private: void __cdecl MainWindow::onBackgroundGrid(void)" (?onBackgroundGrid@MainWindow@@AEAAXXZ)'
87>LINK : warning LNK4217: symbol '?label@QGVNode@@QEBA?AVQString@@XZ (public: class QString __cdecl QGVNode::label(void)const )' defined in 'YARP_priv_qgvcore.lib(QGVNode.obj)' is imported by 'MainWindow.obj' in function '"private: void __cdecl MainWindow::nodeDoubleClick(class QGVNode *)" (?nodeDoubleClick@MainWindow@@AEAAXPEAVQGVNode@@@Z)'
87>LINK : warning LNK4217: symbol '?setAttribute@QGVNode@@QEAAXAEBVQString@@0@Z (public: void __cdecl QGVNode::setAttribute(class QString const &,class QString const &))' defined in 'YARP_priv_qgvcore.lib(QGVNode.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
87>LINK : warning LNK4217: symbol '?setVertex@QGVNode@@QEAAXPEAX@Z (public: void __cdecl QGVNode::setVertex(void *))' defined in 'YARP_priv_qgvcore.lib(QGVNode.obj)' is imported by 'MainWindow.obj' in function '"public: void __cdecl MainWindow::drawGraph(class yarp::profiler::graph::Graph &)" (?drawGraph@MainWindow@@QEAAXAEAVGraph@graph@profiler@yarp@@@Z)'
[... cut ...]
87>YARP_priv_qgvcore.lib(QGVScene.obj) : error LNK2019: unresolved external symbol AgIoDisc referenced in function "public: void __cdecl QGVScene::loadLayout(class QString const &)" (?loadLayout@QGVScene@@QEAAXAEBVQString@@@Z)
87>YARP_priv_qgvcore.lib(QGVScene.obj) : error LNK2019: unresolved external symbol Agdirected referenced in function "public: __cdecl QGVScene::QGVScene(class QString const &,class QObject *,bool)" (??0QGVScene@@QEAA@AEBVQString@@PEAVQObject@@_N@Z)
87>C:\robotology-superbuild\build\src\YARP\bin\Release\yarpviz.exe : fatal error LNK1120: 4 unresolved externals
```

S-Dafarra, 2022-06-09:
With ``graphviz=2.50.0`` ``yarp`` compiled successfully.

traversaro, 2022-06-09:
Changelog of graphviz: https://gitlab.com/graphviz/graphviz/-/blob/main/CHANGELOG.md#400-2022-05-29 . Possibly related changes:
* `Breaking: The Agiodisc_t struct member putstr that was previously an
fputs analog is replaced by printf that is required to behave similar to
fprintf.`

If we fix this, it may be worth to suggest the changes also to https://github.com/Gepetto/qgv .

S-Dafarra, 2022-06-09:
Actually, it has just been reverted: https://gitlab.com/graphviz/graphviz/-/blob/a9c36efe4041cc16a54e834440e3a489eaa222e1/CHANGELOG.md#fixed

traversaro, 2022-06-09:
> Actually, it has just been reverted: https://gitlab.com/graphviz/graphviz/-/blob/a9c36efe4041cc16a54e834440e3a489eaa222e1/CHANGELOG.md#fixed

Cool, let's hope they soon do a release.

traversaro, 2022-06-10:
Bingo, we finally hit this in our CI (probably some migration completed and unblocked the use of graphviz 4) https://github.com/robotology/robotology-superbuild/runs/6827278767 : 
~~~
2022-06-10T08:40:25.4527691Z     Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/src/YARP/extern/qgv/CMakeLists.txt
2022-06-10T08:40:25.5482567Z cl : command line warning D9025: overriding '/W1' with '/w' [D:\a\robotology-superbuild\robotology-superbuild\build\src\YARP\extern\qgv\YARP_priv_qgvcore.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\YARP.vcxproj]
2022-06-10T08:40:25.5486716Z     QGVCore.cpp
2022-06-10T08:40:26.7750454Z D:\a\robotology-superbuild\robotology-superbuild\src\YARP\extern\qgv\qgv\QGVCore\private\QGVCore.h(90,19): error C2039: 'putstr': is not a member of 'Agiodisc_s' [D:\a\robotology-superbuild\robotology-superbuild\build\src\YARP\extern\qgv\YARP_priv_qgvcore.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\YARP.vcxproj]
2022-06-10T08:40:26.7751776Z   C:\Miniconda3\envs\test\Library\include\graphviz\cgraph.h(170): message : see declaration of 'Agiodisc_s' [D:\a\robotology-superbuild\robotology-superbuild\build\src\YARP\extern\qgv\YARP_priv_qgvcore.vcxproj]
2022-06-10T08:40:26.7753087Z D:\a\robotology-superbuild\robotology-superbuild\src\YARP\extern\qgv\qgv\QGVCore\private\QGVCore.h(90,37): error C2039: 'putstr': is not a member of 'Agiodisc_s' [D:\a\robotology-superbuild\robotology-superbuild\build\src\YARP\extern\qgv\YARP_priv_qgvcore.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\YARP.vcxproj]
2022-06-10T08:40:26.7757188Z   C:\Miniconda3\envs\test\Library\include\graphviz\cgraph.h(170): message : see declaration of 'Agiodisc_s' [D:\a\robotology-superbuild\robotology-superbuild\build\src\YARP\extern\qgv\YARP_priv_qgvcore.vcxproj]
2022-06-10T08:40:26.7758063Z     QGVGraphPrivate.cpp
2022-06-10T08:40:26.7758453Z     QGVEdgePrivate.cpp
~~~

As this change back soon as mentioned earlier, probably we can try to just install graphviz 3 (or 2.50) for the time being.

S-Dafarra, 2022-06-10:
I had missing symbols issues with the 3.

traversaro, 2022-06-10:
> I had missing symbols issues with the 3.

Yes, in that case those are not something that will go away like the `putstr` change, so I would at least try to fix them.

traversaro, 2022-06-10:
> > I had missing symbols issues with the 3.
> 
> Yes, in that case those are not something that will go away like the `putstr` change, so I would at least try to fix them.

This will be fixed by:
* https://github.com/robotology/ycm/pull/414
* https://github.com/robotology/yarp/pull/2851
will be merged and released.

however, it will take some time to get a YARP 3.7 patch release, so in the meanwhile we can switch to pin graphviz 2.50 .

traversaro, 2023-04-26:
The issue was fixed by all the aforementioned PRs, and https://github.com/robotology/robotology-superbuild/pull/1213 removed all the workarounds.

closed 2023-04-26 by traversaro