robotology/robotology-superbuild #1163: The ``robot-log-visualizer`` installed from the superbuild does not show the output of the camera

S-Dafarra, 2022-06-10:
I installed the ``robot-log-visualizer`` via the superbuild. It starts, but when I open a log file with a video, the video is not shown in the panel.

Looking at the terminal output, I see the following error
[image]

cc @GiulioRomualdi @traversaro

S-Dafarra, 2022-06-10:
I have installed ``libqt5multimedia5-plugins`` via ``apt``. Now, I don't see that error anymore, but the video is all black.

traversaro, 2022-06-10:
(Unrelated, but I saw the warning in the screenshot: https://github.com/robotology/idyntree/pull/997).

traversaro, 2022-06-10:
Just to double check, you can also try to install `robot-log-visualizer` in the same machine either in a pip environment or on a conda one (we should have packages for it both in PyPI and conda robotology channel) to see if the same problem happens there.

S-Dafarra, 2022-06-10:
Ah ok, so I can install it also using ``pip`` in an environment and it will not complain if we have it in the superbuild too (and hence in the path)?

traversaro, 2022-06-10:
> Ah ok, so I can install it also using `pip` in an environment and it will not complain if we have it in the superbuild too (and hence in the path)?

I am not really sure about this interactions, what I had in mind is to have a terminal in which you do not source the `setup.sh` of the robotology-superbuild, and there do this tests.

S-Dafarra, 2022-06-10:
I think I got it. The problem might be an incompatibility with the video generated by the version of ``blf`` we are currently using with the robot (https://github.com/ami-iit/bipedal-locomotion-framework/tree/add_cameras_logger). This version generates video files with the suffix ``camera1`` https://github.com/ami-iit/bipedal-locomotion-framework/blob/f0e7ebfccc4f3a53ed529f9866cc8914ce690b68/devices/YarpRobotLoggerDevice/app/robots/iCubGenova09/interface/webcams.xml#L8, but the visualizer is expecting ``webcam`` (https://github.com/ami-iit/robot-log-visualizer/blob/84e9edb452b7b4829f0e1c3c68a9f66f45b38796/robot_log_visualizer/ui/gui.py#L385). So probably the video is not loaded at all

S-Dafarra, 2022-06-13:
> Hi @S-Dafarra which commit are you using in blf? I think there is a bug in the visualizer I can fix

Hi @GiulioRomualdi, we changed the configuration files of the logger device in https://github.com/ami-iit/bipedal-locomotion-framework/commit/5a314e050fd891a0b28e693af1b19058cdf6076e (this is the commit we are currently using) in order to restore the ``_webcam`` suffix in the video. I think that the visualizer is not yet compatible with the multiple cameras option (https://github.com/ami-iit/robot-log-visualizer/blob/84e9edb452b7b4829f0e1c3c68a9f66f45b38796/robot_log_visualizer/ui/gui.py#L385)

GiulioRomualdi, 2022-06-13:
ook! that one was the problem. By the way, This PR fixes your problem https://github.com/ami-iit/robot-log-visualizer/pull/44. Indeed now the video file name is retrieved using a regexp. Furthermore, if you collect multiple videos from the logger, the visualizer will open all of them.

https://user-images.githubusercontent.com/16744101/173373344-a78757ab-2c3e-455b-a19c-e774420da743.mp4

S-Dafarra, 2022-06-13:
> ook! that one was the problem. By the way, This PR fixes your problem [ami-iit/robot-log-visualizer#44](https://github.com/ami-iit/robot-log-visualizer/pull/44). Indeed now the video file name is retrieved using a regexp. Furthermore, if you collect multiple videos from the logger, the visualizer will open all of them.
> 
> Here I just duplicated the same video to test the new feature
> 
>  simplescreenrecorder-2022-06-13_16.11.53.mp4

Super!

GiulioRomualdi, 2022-06-13:
If you want to test it on your setup please use following method

1. switch to https://github.com/ami-iit/robot-log-visualizer/pull/44
4. run `bash ./generate-ui.sh` in the root of the robot-log-visualizer (this is actually done by the CI once merged)
6. install it