robotology/robotology-superbuild #1168: Undefinded problems encountered during the build process

SSSSamZhu, 2022-06-13:
I'm trying to build the v2022.5.0 tag of the superbuild in a clean docker image with the following command:

```
RUN git clone  https://github.com/robotology/robotology-superbuild 
    cd robotology-superbuild 
    sudo bash ./scripts/install_apt_dependencies.sh 
    sudo rm -rf /var/lib/apt/lists/*

RUN mkdir build && cd build
    ccmake ./
    make
```
ROBOTOLOGY_ENABLE_DYNAMICS = ON
ROBOTOLOGY_ENABLE_DYNAMICS _FULL = ONgoi

Error 
```
[ 88%] Automatic MOC for target yarpmanager
[ 88%] Built target yarpmanager_autogen
Consolidate compiler generated dependencies of target yarpmanager
[ 88%] Linking CXX executable ../../bin/yarpmanager
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetnum@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tputs@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgoto@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetent@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetstr@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetflag@NCURSES6_TINFO_5.0.19991023'
collect2: error: ld returned 1 exit status
make[5]: *** [src/yarpmanager/CMakeFiles/yarpmanager.dir/build.make:667: bin/yarpmanager] Error 1
make[4]: *** [CMakeFiles/Makefile2:9546: src/yarpmanager/CMakeFiles/yarpmanager.dir/all] Error 2
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1051: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```
looking forward to your reply.

traversaro, 2022-06-13:
Hi @SSSSamZhu, which base docker image are you using? Thanks.

SSSSamZhu, 2022-06-14:
Hi @traversaro 
I built directly using Master.In addition, the host I configured has installed YARP3.6.0 before,YCM has also been upgraded to version 0.14.1.I'm not sure if this is the reason why the compile failed

traversaro, 2022-06-14:
> Hi @traversaro
> I built directly using Master.

Hi @SSSSamZhu, `master` of what? What I was referring was the base image that you used in your Dockerfile, i.e. the one specified after `FROM` in your Dockerfile. More in general, if you could share your Dockerfile (or at least a subset of it) it would be helpful to help you.

SSSSamZhu, 2022-06-14:
Hi @traversaro，
Maybe I gave you the wrong impression.I did not use Docker. I directly used clone files in base environment and did not use Conda to build virtual environment.

traversaro, 2022-06-14:
Hi @SSSSamZhu, sorry but your original message started with:
> I'm trying to build the v2022.5.0 tag of the superbuild in a clean docker image with the following command:
hence why I tought you used Docker. 

Anyhow, if you are not using Docker, can you report:
* Which distribution are you using (Ubuntu 20.04, Ubuntu 22.04, etc etc)
* Report the output of the `env` command in the terminal in which you are building the superbuild?

traversaro, 2022-06-14:
Hi @SSSSamZhu, as mentioned before, can you:
> Report the output of the `env` command in the terminal in which you are building the superbuild?

Thanks.

traversaro, 2022-06-14:
Hi @SSSSamZhu, with:
> Report the output of the env command in the terminal in which you are building the superbuild?

I meant type the `env` command in the terminal where you built the superbuild, and report the output of the `env` command.

traversaro, 2022-06-14:
From the output, and the fact that your username is prefixed by `(base)` it seems that you have an activated conda environment called `base`. It is not supported to compile `robotology-superbuild` using apt dependencies if there is a `conda` environment activated, as these can lead to the conflicts that you are seeing.
My suggestion is to:
* delete the existing build directory, as the existing build directory probably uses some libraries from conda and some from apt
* deactivate the conda `base` environment. To do that only for a terminal, just do `conda deactivated`, while if you want this to be the default for newly opened terminals, run once `conda config --set auto_activate_base false`.
* now, in a clean terminal (so where you do not have the `(base)` prefix) create a new build directory and compile the superbuild, you should not have any more problems.

SSSSamZhu, 2022-06-15:
Morning @traversaro，
Unfortunately, these actions did not work. To make matters worse, I tried to recompile the previously compiled YARP source file and the same error occurred again. I'm not sure if there's any correlation or conflict between the two.
Here's how I did it ,
```
(base) icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild$ rm -rf build/
(base) icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild$ conda deactivate
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild$ conda config --set auto_activate_base false
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild$ mkdir build && cd build
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ ccmake ../
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ cmake ../
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ make -j8
[  1%] Creating directories for 'YCM'
[  3%] Creating directories for 'icub-models'
[  5%] Performing download step (git clone) for 'YCM'
-- Avoiding repeated git clone, repository already exists
[  7%] Performing download step (git clone) for 'icub-models'
-- Avoiding repeated git clone, repository already exists
[  8%] No patch step for 'YCM'
[ 10%] Skipping update step for 'YCM'
[ 12%] No patch step for 'icub-models'
[ 14%] Skipping update step for 'icub-models'
[ 16%] Performing configure step for 'YCM'
Not searching for unused variables given on the command line.
loading initial cache file /home/icub/Desktop/robot/robotology-superbuild/build/src/YCM/CMakeFiles/YCMTmp/YCM-cache-Release.cmake
[ 17%] Performing configure step for 'icub-models'
Not searching for unused variables given on the command line.
loading initial cache file /home/icub/Desktop/robot/robotology-superbuild/build/src/icub-models/CMakeFiles/YCMTmp/icub-models-cache-Release.cmake
-- The C compiler identification is GNU 9.4.0
-- YCM Version: 0.14.1 (0.14.1)
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/GNUInstallDirs.cmake:241 (message):
  Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
  target architecture is known.  Please enable at least one language before
  including GNUInstallDirs.
Call Stack (most recent call first):
  CMakeLists.txt:29 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

.....
[ 61%] Built target YARP_dev
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetnum@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tputs@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgoto@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetent@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetstr@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetflag@NCURSES6_TINFO_5.0.19991023'
collect2: error: ld returned 1 exit status
make[5]: *** [src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/build.make:347: bin/yarpdataplayer] Error 1
make[4]: *** [CMakeFiles/Makefile2:9660: src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all] Error 2
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:943: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

SSSSamZhu, 2022-06-20:
Hi @traversaro，
Running the command you mentioned earlier yields the following result.
1. env
```
Setting up yarp and iCub env vars
icub@isaacliang-LC3:~$ env
SHELL=/bin/bash
ROS_VERSION=1
SESSION_MANAGER=local/isaacliang-LC3:@/tmp/.ICE-unix/3785,unix/isaacliang-LC3:/tmp/.ICE-unix/3785
GAZEBO_MASTER_URI=http://localhost:11345
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_MENU_PREFIX=gnome-
ROBOT_CODE=/usr/local/src/robot
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
CONDA_EXE=/home/icub/anaconda3/bin/conda
_CE_M=
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=/usr/local/src/robot/robotology-superbuild/build/install
PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig
ROS_PYTHON_VERSION=3
LANGUAGE=en_GB:en
TERMINATOR_DBUS_PATH=/net/tenshu/Terminator2
LC_ADDRESS=zh_CN.UTF-8
OBJ_SUBDIR=build
GNOME_SHELL_SESSION_MODE=ubuntu
LC_NAME=zh_CN.UTF-8
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
YARP_BUILD_DIR=/usr/local/src/robot/yarp/build
YARP_ROBOT_NAME=iCubShenzhen01
TERMINATOR_UUID=urn:uuid:f6675924-ce57-4bed-a31a-fea0f13913cc
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
LC_MONETARY=zh_CN.UTF-8
SSH_AGENT_PID=3750
ROS_PACKAGE_PATH=/opt/ros/noetic/share:/usr/local/src/robot/robotology-superbuild/build/install/share
CMAKE_BUILD_TYPE=Release
YARP_DIR=/usr/local/src/robot/yarp/build
GTK_MODULES=gail:atk-bridge
ROSLISP_PACKAGE_DIRECTORIES=
PWD=/home/icub
XDG_SESSION_DESKTOP=ubuntu
LOGNAME=icub
XDG_SESSION_TYPE=x11
CFW2CANAPI_DIR=/lib/modules/5.14.0-1042-oem/iCubDrivers//cfw002/
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
icub_firmware_shared_BUILD_DIR=/usr/local/src/robot/icub-firmware-shared/build
ICUB_SOURCE_DIR=/usr/local/src/robot/icub-main
XAUTHORITY=/home/icub/.Xauthority
XDG_GREETER_DATA_DIR=/var/lib/lightdm-data/icub
GDM_LANG=zh_CN
HOME=/home/icub
IM_CONFIG_PHASE=1
GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins::/usr/local/src/robot/robotology-superbuild/build/install/lib
LC_PAPER=zh_CN.UTF-8
LANG=en_GB.UTF-8
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
YARP_FORWARD_LOG_ENABLE=0
RTF_DIR=/usr/local/src/robot/robot-testing-framework/build
VTE_VERSION=6003
YARP_COLORED_OUTPUT=1
ICUBcontrib_DIR=/usr/local/src/robot/iCubContrib
AMENT_PREFIX_PATH=/usr/local/src/robot/robotology-superbuild/build/install
CMAKE_PREFIX_PATH=/opt/ros/noetic:/usr/local/src/robot/robotology-superbuild/build/install
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
https_proxy=http://127.0.0.1:12639
[... 113 lines cut ...]
# DebugStream customization
export YARP_VERBOSE_OUTPUT=0
export YARP_COLORED_OUTPUT=1
export YARP_TRACE_ENABLE=0
export YARP_FORWARD_LOG_ENABLE=0

# Drivers section
DRIVER_ROOT=/lib/modules/$(uname -r)/iCubDrivers/
# to be used if the corresponding hardware is mounted on the robot 
#export ESDCANAPI_DIR=${DRIVER_ROOT}/esdCanApi/
#export PLXCANAPI_DIR=${DRIVER_ROOT}/plxCanApi/
export CFW2CANAPI_DIR=${DRIVER_ROOT}/cfw002/
```

2. ldd /usr/lib/x86_64-linux-gnu/libedit.so
```
icub@isaacliang-LC3:~$ ldd /usr/lib/x86_64-linux-gnu/libedit.so
	linux-vdso.so.1 (0x00007ffc7a720000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f3105c71000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f3105c57000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3105a65000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3105d26000)
```

Thank you.

traversaro, 2022-06-20:
The `.bashrc_iCub` that you are using is the one meant to be used when YARP, ICUB are only build separately, not as part of the superbuild, if you are using the superbuild please used instead https://raw.githubusercontent.com/icub-tech-iit/icub-os-files/master/user-environment/bashrc_iCub_superbuild (see https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/user-env/#note).

For  your specific linking problem, I still does not understand what is going on, as additional info can you provide:
* 1. output of `readelf -s /lib/x86_64-linux-gnu/libedit.so.2 | grep tgetnum`
* 2. output of `readelf -s /lib/x86_64-linux-gnu/libtinfo.so | grep tgetnum`
* 3. The `CMakeCache.txt` contained in build/src/YARP
* 4. Can you go in `build/src/YARP`, run `cmake CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=ON .` and return the `compile_commands.json` that this command will generated in the build?

Sorry for asking so much info, I am quite confused as we continuously test this in CI and never saw something similar. The only difference between your case and the CI is that in your case you have ROS enabled, so if you want to try it could make sense to comment out from the `.bashrc` the ROS's `setup.sh` and generated a new build of the superbuild in a clean environment, but I would be quite surprised if this is related to ROS.

Another workaround you may try is to uninstall the `libedit-dev`. This should work, even if it is just a workaround that may hide the real problems.

SSSSamZhu, 2022-06-21:
Hi @traversaro，
Running the command you mentioned earlier yields the following result.
1,2.
```
icub@isaacliang-LC3:~$ readelf -s /lib/x86_64-linux-gnu/libedit.so.2 | grep tgetnum
    84: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND tgetnum@NCURSES6_TINFO_5.0.19991023 (4)
icub@isaacliang-LC3:~$ readelf -s /lib/x86_64-linux-gnu/libtinfo.so | grep tgetnum
    95: 0000000000014b00   285 FUNC    GLOBAL DEFAULT   15 tgetnum_sp@@NCURSES6_TINFO_5.8.20110226
   188: 0000000000014c20    22 FUNC    GLOBAL DEFAULT   15 tgetnum@@NCURSES6_TINFO_5.0.19991023

```
3.4. 
[film.zip](https://github.com/robotology/robotology-superbuild/files/8945776/film.zip)

Thank you.

traversaro, 2022-06-21:
Hi @SSSSamZhu, perhaps I am missing something but the `film.zip` file seems empty.

SSSSamZhu, 2022-06-21:
Hi @traversaro，
I'm awfully sorry for my oversight. I re-uploaded the folder.In addition, I would like to ask what is the meaning of `uninstall the libedit-dev`Do I need to delete this configuration file?

[New Folder.zip](https://github.com/robotology/robotology-superbuild/files/8947003/New.Folder.zip)

traversaro, 2022-06-21:
> In addition, I would like to ask what is the meaning of `uninstall the libedit-dev`

I meant running:
~~~
sudo apt remove libedit-dev
~~~

traversaro, 2022-06-21:
Thanks, as I mentioned before, the problem is always the same, there is still cross-talking between apt libraries and conda libraries. If you check your environment, you will see that even if you deactivated the environment, you still have `/home/icub/anaconda3/bin` in your `PATH`, and hence the problem (you can verify yourself by opening the `CMakeCache.txt` and noting how many times there is a part that start with `/home/icub/anaconda3`. Do you have any idea why this happens? 

If you are not using `anaconda3` for anything, I suggest you do uninstall it. If instead you need it to have it in the system, I suggest to try to understand who is adding `/home/icub/anaconda3/bin` in your `PATH`, and remove it.

SSSSamZhu, 2022-06-21:
Hi @traveraro，
Thanks for your patience
1. I tried to execute this command, but it had new problems during the build phase, so I ended up having to reinstall it.
`sudo apt remove libedit-dev`
2. I'm trying to understand what you mean.，maybe I'll be wrong.
` /home/icub/anaconda3/bin` is no longer in my `PATH`, which means there is no more crosstalk between apt and conda. So I tried to remove all the environment variables associated with `/home/icub/anaconda3/bin`, but unfortunately the problem continued and this is the final print.

```
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ make -j8
[  1%] Skipping update step for 'YCM'
[  3%] Skipping update step for 'icub-models'
[  5%] Performing configure step for 'YCM'
	
....
[ 51%] Built target YARP_companion
[ 51%] Building CXX object src/libYARP_profiler/src/CMakeFiles/YARP_profiler.dir/yarp/profiler/NetworkProfilerBasic.cpp.o
[ 59%] Built target YARP_dev
[ 59%] Building CXX object src/libYARP_profiler/src/CMakeFiles/YARP_profiler.dir/yarp/profiler/NetworkProfiler.cpp.o
[ 59%] Built target yarpdataplayer_autogen
[ 59%] Building CXX object src/yarp/CMakeFiles/yarp.dir/yarp.cpp.o
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetnum@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tputs@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgoto@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetent@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetstr@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetflag@NCURSES6_TINFO_5.0.19991023'
collect2: error: ld returned 1 exit status
make[5]: *** [src/yarplogger/CMakeFiles/yarplogger.dir/build.make:281: bin/yarplogger] Error 1
make[4]: *** [CMakeFiles/Makefile2:9588: src/yarplogger/CMakeFiles/yarplogger.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[ 59%] Building CXX object src/libYARP_profiler/src/CMakeFiles/YARP_profiler.dir/yarp/profiler/Graph.cpp.o
[ 59%] Linking CXX executable ../../bin/yarp
[ 59%] Built target yarp
[ 60%] Linking CXX static library ../../../lib/libYARP_profiler.a
[ 60%] Built target YARP_profiler
[ 61%] Linking CXX executable ../../bin/yarpmanager-console
[ 61%] Built target yarpmanager-console
[ 61%] Linking CXX executable ../../bin/yarpdataplayer-console
[ 61%] Built target yarpdataplayer-console
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:943: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ env
SHELL=/bin/bash
ROS_VERSION=1
SESSION_MANAGER=local/isaacliang-LC3:@/tmp/.ICE-unix/3785,unix/isaacliang-LC3:/tmp/.ICE-unix/3785
GAZEBO_MASTER_URI=http://localhost:11345
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_MENU_PREFIX=gnome-
ROBOT_CODE=/usr/local/src/robot
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=/usr/local/src/robot/robotology-superbuild/build/install
PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig
ROS_PYTHON_VERSION=3
LANGUAGE=en_GB:en
LC_ADDRESS=zh_CN.UTF-8
OBJ_SUBDIR=build
GNOME_SHELL_SESSION_MODE=ubuntu
LC_NAME=zh_CN.UTF-8
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
YARP_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/YARP
YARP_ROBOT_NAME=iCubShenzhen01
XMODIFIERS=@im=ibus
ICUBcontrib_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/ICUBcontrib
DESKTOP_SESSION=ubuntu
LC_MONETARY=zh_CN.UTF-8
[... 63 lines cut ...]
XDG_RUNTIME_DIR=/run/user/1000
YARP_SOURCE_DIR=/usr/local/src/robot/robotology-superbuild/src/YARP
LC_TIME=zh_CN.UTF-8
GAZEBO_MODEL_PATH=/usr/share/gazebo-11/models:/usr/share/gazebo-11/models::/usr/local/src/robot/robotology-superbuild/build/install/share/gazebo/models:/usr/local/src/robot/robotology-superbuild/build/install/share/iCub/robots:/usr/local/src/robot/robotology-superbuild/build/install/share:/usr/local/src/robot/robotology-superbuild/build/install/share/gazebo/models:/usr/local/src/robot/robotology-superbuild/build/install/share/iCub/robots:/usr/local/src/robot/robotology-superbuild/build/install/share
JOURNAL_STREAM=8:59396
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/local/share:/usr/share:/var/lib/snapd/desktop:/usr/local/src/robot/robotology-superbuild/build/install/share:/usr/local/src/robot/robotology-superbuild/build/install/share
YARP_DATA_DIRS=/home/roboverse/yarp/build/share/yarp:/home/roboverse/iCub/icub-main/build/share/iCub:/home/roboverse/iCub/iCubContrib/share/ICUBcontrib:/usr/local/src/robot/robotology-superbuild/build/install/share/yarp:/usr/local/src/robot/robotology-superbuild/build/install/share/iCub:/usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib:/usr/local/src/robot/robotology-superbuild/build/install/share/yarp:/usr/local/src/robot/robotology-superbuild/build/install/share/iCub:/usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib
[... cut ...]
3.  So far, I wonder if the configuration of my computer itself conflicts with YARP. The same error occurs when compiling a single YRAP independently, as I reported earlier. It is simple and feasible to compile YARP separately before compiling superbuild. As for `Conda`, I have tried to recompile directly from a new environment, but this problem still occurs. 

4. I have a question that I would like to ask. Can I try to change the method of building using Conda-Forge on this host? Will it conflict with the APT method I used before ? https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md

In conclusion, thank you very much for your patience

traversaro, 2022-06-21:
> I have a question that I would like to ask. Can I try to change the method of building using Conda-Forge on this host? Will it conflict with the APT method I used before ? https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md

You can try, as long as you do not source the `setup.sh` of the superbuild compiled with apt dependencies. However, you will then need to create a clean environemnt using only the `conda-forge` environement and not the `defaults` one that comes by default if you install anaconda, so also that may not be trivial.

> I'm trying to understand what you mean.，maybe I'll be wrong.
/home/icub/anaconda3/bin is no longer in my PATH, which means there is no more crosstalk between apt and conda. So I tried to remove all the environment variables associated with /home/icub/anaconda3/bin, but unfortunately the problem continued and this is the final print.

Did you remove your build directory and created a new one after you removed `/home/icub/anaconda3/bin` from your path? If you did not, this explains the errors, as CMake caches a lot of paths so you will still have references to  `/home/icub/anaconda3/bin` around.

SSSSamZhu, 2022-06-22:
Hi @traveraro，
My original problem was solved after I removed the build and rebuilt it. In this case, anaconda3 does affect compilation.
But we meet new problems now. The problem we had in the beginning was this

```
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ make -j64
[  1%] Skipping update step for 'YCM'
[  3%] Skipping update step for 'icub-models'
[  5%] Performing configure step for 'YCM'
......
-- Generating done
-- Build files have been written to: /home/icub/Desktop/robot/robotology-superbuild/build/src/bipedal-locomotion-framework
[ 94%] Performing build step for 'bipedal-locomotion-framework'
Consolidate compiler generated dependencies of target VectorsCollection
Consolidate compiler generated dependencies of target TextLogging
[  2%] Built target VectorsCollection
[  3%] Linking CXX shared library ../../lib/libBipedalLocomotionFrameworkTextLogging.so
/usr/bin/ld: /usr/local/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[5]: *** [src/TextLogging/CMakeFiles/TextLogging.dir/build.make:130: lib/libBipedalLocomotionFrameworkTextLogging.so.0.6.0] Error 1
make[4]: *** [CMakeFiles/Makefile2:1116: src/TextLogging/CMakeFiles/TextLogging.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/bipedal-locomotion-framework.dir/build.make:85: src/bipedal-locomotion-framework/CMakeFiles/YCMStamp/bipedal-locomotion-framework-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:2297: CMakeFiles/bipedal-locomotion-framework.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```
We have referred to this issue and it is useful to some extent. 
https://github.com/gabime/spdlog/issues/1405#issue-556962601
I ran the command
`cmake -DCMAKE_POSITION_INDEPENDENT_CODE=ON .. && make -j`
But then he had a new problem
```
icub@isaacliang-LC3:~/Desktop/robot/robotology-superbuild/build$ make -j64
[ 73%] Skipping update step for 'osqp'
[ 73%] Skipping update step for 'YCM'
[ 73%] Skipping update step for 'qpOASES'
[ 73%] Skipping update step for 'CppAD'
[ 73%] Skipping update step for 'icub-models'
[ 73%] Skipping update step for 'BlockFactory'
[ 73%] Skipping update step for 'matioCpp'
[ 73%] Skipping update step for 'manif'
[ 73%] Skipping update step for 'qhull'
[ 73%] Performing configure step for 'osqp'
....
[ 73%] Performing install step for 'qhull'
Not searching for unused variables given on the command line.
CMake Deprecation Warning at CMakeLists.txt:31 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:35 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

loading initial cache file /home/icub/Desktop/robot/robotology-superbuild/build/src/OsqpEigen/CMakeFiles/YCMTmp/OsqpEigen-cache-Release.cmake
CMake Error at cmake/language_support.cmake:32 (file):
  file failed to open for writing (Permission denied):

    /home/icub/Desktop/robot/robotology-superbuild/build/src/casadi/language_tests/Fortran/CMakeLists.txt
Call Stack (most recent call first):
[... 53 lines cut ...]
  	git config --global --add safe.directory
  /exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-src

make[5]: *** [CMakeFiles/json-populate.dir/build.make:130: json-populate-prefix/src/json-populate-stamp/json-populate-update] Error 1
make[5]: Leaving directory '/exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-subbuild'
make[4]: *** [CMakeFiles/Makefile2:83: CMakeFiles/json-populate.dir/all] Error 2
make[4]: Leaving directory '/exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-subbuild'
make[3]: *** [Makefile:91: all] Error 2
make[3]: Leaving directory '/exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-subbuild'

CMake Error at /usr/share/cmake-3.24/Modules/FetchContent.cmake:1572 (message):
  Build step for json failed: 2
Call Stack (most recent call first):
  /usr/share/cmake-3.24/Modules/FetchContent.cmake:1712:EVAL:2 (__FetchContent_directPopulate)
  /usr/share/cmake-3.24/Modules/FetchContent.cmake:1712 (cmake_language)
  CMakeLists.txt:82 (FetchContent_Populate)

-- Configuring incomplete, errors occurred!
See also "/home/icub/Desktop/robot/robotology-superbuild/build/src/YARP_telemetry/CMakeFiles/CMakeOutput.log".
See also "/home/icub/Desktop/robot/robotology-superbuild/build/src/YARP_telemetry/CMakeFiles/CMakeError.log".
make[2]: *** [CMakeFiles/YARP_telemetry.dir/build.make:95: src/YARP_telemetry/CMakeFiles/YCMStamp/YARP_telemetry-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:1921: CMakeFiles/YARP_telemetry.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```
In conclusion, thank you very much for your patience

traversaro, 2022-06-22:
## spdlog problem

Beside the workaround you found, another fix is to just compiled spdlog as a shared library. However, why do you have a custom install of spdlog in `/usr/local`, instead of using the spdlog installed by apt? 

## fatal: unsafe repository error (is owned by someone else)

This is a problem that I never saw, I need to think about it.

traversaro, 2022-06-22:
> ## fatal: unsafe repository error (is owned by someone else)
> 
> This is a problem that I never saw, I need to think about it.

Did you added a symbolic link `/home/icub/Desktop/robot/robotology-superbuild`  to `/exports/code/robotology-superbuild` ? Perhaps this can create problems.

SSSSamZhu, 2022-06-23:
Hi @traveraro，
> Did you added a symbolic link /home/icub/Desktop/robot/robotology-superbuild to /exports/code/robotology-superbuild ? Perhaps this can create problems.

Did you mean
```
cd  /home/icub/Desktop/robot/robotology-superbuild
ln -s  /exports/code/robotology-superbuild dist_file
```
 (dist_file: The newly created link file)?

traversaro, 2022-06-23:
> > Beside the workaround you found, another fix is to just compiled spdlog as a shared library. However, why do you have a custom install of spdlog in /usr/local, instead of using the spdlog installed by apt?
> 
> At first I thought I was missing the dependency to install spdlog, so I installed a new spdlog in the root directory. Now it seems that this operation is a little redundant, it didn't work.

As you may have noticed, the operation of installing  manually in `/usr/local`  a different version of a library that is already installed by apt it not only redundant, can lead to errors. Please go to the build directory from which spdlog was installed, and uninstall it with (from https://gitlab.kitware.com/cmake/community/-/wikis/FAQ#can-i-do-make-uninstall-with-cmake):
~~~
xargs rm < install_manifest.txt
~~~ 

Then, make sure that `libspdlog-dev` is installed with:
~~~
sudo apt install libspdlog-dev
~~~
Notice that should be already done by the `sudo bash ./scripts/install_apt_dependencies.sh` step of the installation of the superbuild, are you sure you executed it?

For what regards the strange error related to YARP_telemetry, did you try to execute the workaround the error is suggesting, i.e.:
~~~
git config --global --add safe.directory /exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-src
~~~
?

SSSSamZhu, 2022-06-23:
Hello @traveraro，
Thanks, it works !
> For what regards the strange error related to YARP_telemetry, did you try to execute the workaround the error is suggesting, i.e.:
> `git config --global --add safe.directory /exports/code/robotology-superbuild/build/src/YARP_telemetry/_deps/json-src`
> ?
After I repeat make a few times, the` YARP_telemetry `problem doesn't appear on the terminal again. So I didn't do it again.
What I end up doing is
1. Create a file called cmake_uninstall.cmake.in in the top directory of the package with the following contents
```
if(NOT EXISTS "@CMAKE_BINARY_DIR@/install_manifest.txt")
  message(FATAL_ERROR "Cannot find install manifest: @CMAKE_BINARY_DIR@/install_manifest.txt")
endif()

file(READ "@CMAKE_BINARY_DIR@/install_manifest.txt" files)
string(REGEX REPLACE "\n" ";" files "${files}")
foreach(file ${files})
  message(STATUS "Uninstalling $ENV{DESTDIR}${file}")
  if(IS_SYMLINK "$ENV{DESTDIR}${file}" OR EXISTS "$ENV{DESTDIR}${file}")
    exec_program(
      "@CMAKE_COMMAND@" ARGS "-E remove \"$ENV{DESTDIR}${file}\""
      OUTPUT_VARIABLE rm_out
      RETURN_VALUE rm_retval
      )
    if(NOT "${rm_retval}" STREQUAL 0)
      message(FATAL_ERROR "Problem when removing $ENV{DESTDIR}${file}")
    endif()
  else(IS_SYMLINK "$ENV{DESTDIR}${file}" OR EXISTS "$ENV{DESTDIR}${file}")
    message(STATUS "File $ENV{DESTDIR}${file} does not exist.")
  endif()
endforeach()
```

2. There is a cmakelists. TXT file in the same directory, add the following at the end:
```
# uninstall target
if(NOT TARGET uninstall)
  configure_file(
    "${CMAKE_CURRENT_SOURCE_DIR}/cmake_uninstall.cmake.in"
    "${CMAKE_CURRENT_BINARY_DIR}/cmake_uninstall.cmake"
    IMMEDIATE @ONLY)

  add_custom_target(uninstall
    COMMAND ${CMAKE_COMMAND} -P ${CMAKE_CURRENT_BINARY_DIR}/cmake_uninstall.cmake)
endif()
```

3. Find the install_manifest.txt file, usually in the build folder, and execute in that directory
`xargs rm < install_manifest.txt `
4.Finally, delete the source code package

Finally I have a question about the readme file, can I not execute `sudo make install` under apt method? Or the document may have omitted the command.
[image]

traversaro, 2022-06-23:
Good to know @SSSSamZhu . For your own knoledge, for the uninstall the `xargs rm < install_manifest.txt` is sufficient, no need to to the step 1. and 2. . 

If everything works, feel free to close the issue.

SSSSamZhu, 2022-06-23:
Hello @traveraro，
Unfortunately, he had a problem verifying the installation

traversaro, 2022-06-23:
> Does that mean I need to delete icub-main, etc., which I compiled separately before? 

That would be ideal. 

> Is it possible to fix this problem by modifying environment variables?

You can try to remove any reference to manually build YARP, icub, etc etc from environment variables, but my suggestion is just to remove every library that you manually built, and re-run a clean build of the superbuild just to be sure.

traversaro, 2022-06-23:
By  the way, I noticed that the location of the custom built iCub is `/home/roboverse/iCub/icub-main/build`, are you mixing the use of two different users (`roboverse`  and `icub` )? In that case, this could explain the "unsafe directory" problem we saw before.

SSSSamZhu, 2022-06-23:
> By the way, I noticed that the location of the custom built iCub is , are you mixing the use of two different users ( and )? In that case, this could explain the "unsafe directory" problem we saw before.`/home/roboverse/iCub/icub-main/build``roboverse``icub`
The device's user name was changed once, from Roboverse to icub. I modified what I could find. But not quite.

SSSSamZhu, 2022-06-24:
Morning @traveraro，
Now the terminal can run iCub_SIM, but when running yarpMotorGUI and YarpManager, there is an error in the picture.Obviously this error has a lot to do with YARP. I tried to execute `xargs rm < install_manifest.txt.` in` ~robot/yarp/build`. However, the terminal report does not contain this file `install_manifest.txt` in build. 

I have thought about the following reasons: 
1. There was no TXT file when the build was generated, as shown in the third picture below (this is a screenshot from icUB standard configuration notebook) 
2. As I mentioned before, the TXT file was overwritten when I recompiled, but because I failed to compile it, His cache in the system still exists, causing this problem.

[image]
[image]
[image]

traversaro, 2022-06-24:
If there is no `install_manifest.txt`, probably you never installed it in the first place. Are you sure that you do not have any env variable that refer to the `yarp/build`  directory?

SSSSamZhu, 2022-06-27:
Morning @traveraro，

```
Setting up yarp and iCub env vars
icub@isaacliang-LC3:~$ env
SHELL=/bin/bash
ROS_VERSION=1
SESSION_MANAGER=local/isaacliang-LC3:@/tmp/.ICE-unix/4310,unix/isaacliang-LC3:/tmp/.ICE-unix/4310
GAZEBO_MASTER_URI=http://localhost:11345
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_MENU_PREFIX=gnome-
ROBOT_CODE=/usr/local/src/robot
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=/home/icub/Desktop/robot/robotology-superbuild/build/install
PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig
ROS_PYTHON_VERSION=3
LANGUAGE=en_GB:en
LC_ADDRESS=zh_CN.UTF-8
OBJ_SUBDIR=build
GNOME_SHELL_SESSION_MODE=ubuntu
LC_NAME=zh_CN.UTF-8
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
YARP_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/YARP
YARP_ROBOT_NAME=iCubShenzhen01
XMODIFIERS=@im=ibus
ICUBcontrib_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/ICUBcontrib
DESKTOP_SESSION=ubuntu
LC_MONETARY=zh_CN.UTF-8
SSH_AGENT_PID=4274
ROS_PACKAGE_PATH=/opt/ros/noetic/share:/home/icub/Desktop/robot/robotology-superbuild/build/install/share
CMAKE_BUILD_TYPE=Release
GTK_MODULES=gail:atk-bridge
ROSLISP_PACKAGE_DIRECTORIES=
PWD=/home/icub
LOGNAME=icub
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
CFW2CANAPI_DIR=/lib/modules/5.14.0-1042-oem/iCubDrivers//cfw002/
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
icub_firmware_shared_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/icub_firmware_shared
ICUB_SOURCE_DIR=/usr/local/src/robot/robotology-superbuild/src/ICUB
XAUTHORITY=/home/icub/.Xauthority
XDG_GREETER_DATA_DIR=/var/lib/lightdm-data/icub
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
GDM_LANG=zh_CN
HOME=/home/icub
IM_CONFIG_PHASE=1
GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins::/home/icub/Desktop/robot/robotology-superbuild/build/install/lib:/home/icub/Desktop/robot/robotology-superbuild/build/install/lib
LC_PAPER=zh_CN.UTF-8
LANG=en_GB.UTF-8
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
YARP_FORWARD_LOG_ENABLE=0
VTE_VERSION=6003
YARP_COLORED_OUTPUT=1
ICUBcontrib_DIR=/usr/local/src/robot/robotology-superbuild/src/icub-contrib-common
AMENT_PREFIX_PATH=/home/icub/Desktop/robot/robotology-superbuild/build/install:/home/icub/Desktop/robot/robotology-superbuild/build/install
CMAKE_PREFIX_PATH=/opt/ros/noetic:/home/icub/Desktop/robot/robotology-superbuild/build/install:/home/icub/Desktop/robot/robotology-superbuild/build/install
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/102d6709_ccd1_4929_a9d0_8b181389c7f4
GAZEBO_RESOURCE_PATH=/usr/share/gazebo-11:/usr/share/gazebo-11::/home/icub/Desktop/robot/robotology-superbuild/build/install/share/gazebo/worlds:/home/icub/Desktop/robot/robotology-superbuild/build/install/share/gazebo/worlds
INVOCATION_ID=3c0e10e689654fefb529b2bb44cd6ad3
MANAGERPID=3735
GJS_DEBUG_OUTPUT=stderr
ROBOTOLOGY_SUPERBUILD_SOURCE_DIR=/home/icub/Desktop/robot/robotology-superbuild
[... 14 lines cut ...]
QT_IM_MODULE=ibus
LC_MEASUREMENT=zh_CN.UTF-8
BLOCKFACTORY_PLUGIN_PATH=/home/icub/Desktop/robot/robotology-superbuild/build/install/lib/blockfactory:/home/icub/Desktop/robot/robotology-superbuild/build/install/lib/blockfactory
ROBOTOLOGY_SUPERBUILD_INSTALL_DIR=/usr/local/src/robot/robotology-superbuild/build/install
PAPERSIZE=a4
YARP_VERBOSE_OUTPUT=0
YARP_TRACE_ENABLE=0
LD_LIBRARY_PATH=/opt/ros/noetic/lib:/home/icub/Desktop/robot/robotology-superbuild/build/install/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins::/home/icub/Desktop/robot/robotology-superbuild/build/install/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins::/home/icub/Desktop/robot/robotology-superbuild/build/install/lib
XDG_RUNTIME_DIR=/run/user/1000
[... cut ...]
PATH=/opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/icub/Desktop/robot/robotology-superbuild/build/install/bin:/home/icub/Desktop/robot/robotology-superbuild/build/install/bin
GDMSESSION=ubuntu
ICUB_BUILD_DIR=/usr/local/src/robot/robotology-superbuild/build/src/ICUB
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
LC_NUMERIC=zh_CN.UTF-8
OGRE_RESOURCE_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.0
_=/usr/bin/env
```

traversaro, 2022-06-27:
Indeed, it seems that there are no strange env variable. However, it is still possible to accidentally installed a YARP in `/usr/local`. As the problem is in the yarpmanager, probably we can start debugging that one. What is the output of:
~~~
ldd `which yarpmanager`
~~~

SSSSamZhu, 2022-06-27:
```
Setting up yarp and iCub env vars
icub@isaacliang-LC3:~$ ldd `which yarpmanager`
	linux-vdso.so.1 (0x00007fffbdedc000)
	libYARP_init.so.3 => /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libYARP_init.so.3 (0x00007fd7d5708000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fd7d5023000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fd7d4a38000)
	libYARP_math.so.3 => /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libYARP_math.so.3 (0x00007fd7d49e6000)
	libYARP_sig.so.3 => /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libYARP_sig.so.3 (0x00007fd7d4997000)
	libtinyxml.so.2.6.2 => /lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007fd7d497d000)
	libYARP_companion.so.3 => /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libYARP_companion.so.3 (0x00007fd7d48f6000)
	libYARP_os.so.3 => /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3 (0x00007fd7d4715000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fd7d41cc000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd7d3fea000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd7d3e9b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd7d3e7e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd7d3c8c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd7d3c69000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd7d3be1000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fd7d3ba9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd7d3b8d000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fd7d3a86000)
	libjpeg.so.8 => /lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fd7d3a01000)
	libsoxr.so.0 => /lib/x86_64-linux-gnu/libsoxr.so.0 (0x00007fd7d3996000)
	libavcodec.so.58 => /lib/x86_64-linux-gnu/libavcodec.so.58 (0x00007fd7d23d5000)
	libavutil.so.56 => /lib/x86_64-linux-gnu/libavutil.so.56 (0x00007fd7d22ac000)
	libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2 (0x00007fd7d2272000)
	libACE-6.4.5.so => /lib/libACE-6.4.5.so (0x00007fd7d20dd000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd7d5937000)
	libicui18n.so.66 => /lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007fd7d1dde000)
	libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007fd7d1bf8000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd7d1bf2000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007fd7d1b6f000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007fd7d1b57000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd7d1a2e000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fd7d1976000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fd7d1942000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fd7d1883000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fd7d1854000)
	libgomp.so.1 => /lib/x86_64-linux-gnu/libgomp.so.1 (0x00007fd7d1812000)
	libswresample.so.3 => /lib/x86_64-linux-gnu/libswresample.so.3 (0x00007fd7d17f0000)
	libvpx.so.6 => /lib/x86_64-linux-gnu/libvpx.so.6 (0x00007fd7d15b8000)
	libwebpmux.so.3 => /lib/x86_64-linux-gnu/libwebpmux.so.3 (0x00007fd7d15ac000)
	libwebp.so.6 => /lib/x86_64-linux-gnu/libwebp.so.6 (0x00007fd7d1542000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fd7d1517000)
	librsvg-2.so.2 => /lib/x86_64-linux-gnu/librsvg-2.so.2 (0x00007fd7d0bec000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fd7d0b8c000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007fd7d0a69000)
	libzvbi.so.0 => /lib/x86_64-linux-gnu/libzvbi.so.0 (0x00007fd7d09da000)
	libsnappy.so.1 => /lib/x86_64-linux-gnu/libsnappy.so.1 (0x00007fd7d09cf000)
	libaom.so.0 => /lib/x86_64-linux-gnu/libaom.so.0 (0x00007fd7d052f000)
	libcodec2.so.0.9 => /lib/x86_64-linux-gnu/libcodec2.so.0.9 (0x00007fd7cf74b000)
	libgsm.so.1 => /lib/x86_64-linux-gnu/libgsm.so.1 (0x00007fd7cf73c000)
	libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007fd7cf6c4000)
	libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007fd7cf66e000)
	libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007fd7cf60f000)
	libshine.so.3 => /lib/x86_64-linux-gnu/libshine.so.3 (0x00007fd7cf402000)
	libspeex.so.1 => /lib/x86_64-linux-gnu/libspeex.so.1 (0x00007fd7cf3e4000)
	libtheoraenc.so.1 => /lib/x86_64-linux-gnu/libtheoraenc.so.1 (0x00007fd7cf3a7000)
	libtheoradec.so.1 => /lib/x86_64-linux-gnu/libtheoradec.so.1 (0x00007fd7cf387000)
	libtwolame.so.0 => /lib/x86_64-linux-gnu/libtwolame.so.0 (0x00007fd7cf361000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007fd7cf333000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007fd7cf286000)
	libwavpack.so.1 => /lib/x86_64-linux-gnu/libwavpack.so.1 (0x00007fd7cf25a000)
	libx264.so.155 => /lib/x86_64-linux-gnu/libx264.so.155 (0x00007fd7cef9c000)
	libx265.so.179 => /lib/x86_64-linux-gnu/libx265.so.179 (0x00007fd7ce02e000)
	libxvidcore.so.4 => /lib/x86_64-linux-gnu/libxvidcore.so.4 (0x00007fd7cdf1b000)
	libva.so.2 => /lib/x86_64-linux-gnu/libva.so.2 (0x00007fd7cdef2000)
	libva-drm.so.2 => /lib/x86_64-linux-gnu/libva-drm.so.2 (0x00007fd7cdeeb000)
	libva-x11.so.2 => /lib/x86_64-linux-gnu/libva-x11.so.2 (0x00007fd7cdee3000)
[... 18 lines cut ...]
[... cut ...]
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fd7cb729000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fd7cb71e000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fd7cb718000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fd7cb710000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007fd7cb6b9000)
	libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007fd7cb629000)
	libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007fd7cb61d000)
icub@isaacliang-LC3:~$ 

```

traversaro, 2022-06-27:
Nothing strange here, I wonder if the segmentation fault is caused by something else and not by some incompatibility with a previously compiled YARP library.

SSSSamZhu, 2022-06-27:
Hi @traveraro，
> If there is no , probably you never installed it in the first place. Are you sure that you do not have any env variable that refer to the directory?
In fact, I confirm that YARP has been compiled and installed separately. I remember that yarpmanager could be run directly at that time .Although YARP directly seen in the terminal is version 3.7 under Superbuild, so I think that YARP compiled separately before (version 3.6) affects YARP under Superbuild.

> Nothing strange here, I wonder if the segmentation fault is caused by something else and not by some incompatibility with a previously compiled YARP library.
Could it be that we're missing something that needs to be configured?

traversaro, 2022-06-27:
> In fact, I confirm that YARP has been compiled and installed separately. I remember that yarpmanager could be run directly at that time .Although YARP directly seen in the terminal is version 3.7 [...]

Ok, in that case can you open a terminal without sourcing the `setup.sh`  of the robotology-superbuild (for example modifying the `.bashrc` if you have that there), and run `which yarp` or something similar? This should point us to the other YARP installation if this is true.

SSSSamZhu, 2022-06-27:
The original 3.6.0 version of YARP doesn't seem to open anymore ,So far all points to superbuild YARP .

SSSSamZhu, 2022-06-28:
```
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libOsqpEigen.so.0.7.0
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/libOsqpEigen.so
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/OsqpEigen.h
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Constants.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/SparseMatrixHelper.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/SparseMatrixHelper.tpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Data.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Data.tpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Settings.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Solver.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Solver.tpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/include/OsqpEigen/Debug.hpp
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/cmake/OsqpEigen/OsqpEigenConfigVersion.cmake
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/cmake/OsqpEigen/OsqpEigenConfig.cmake
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/cmake/OsqpEigen/OsqpEigenTargets.cmake
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/lib/cmake/OsqpEigen/OsqpEigenTargets-release.cmake
-- Up-to-date: /home/icub/Desktop/robot/robotology-superbuild/build/install/share/cmake/osqp-eigen/package.xml
[ 31%] Completed 'OsqpEigen'
[ 33%] Built target OsqpEigen
[ 33%] Skipping update step for 'iDynTree'
[ 33%] Performing configure step for 'iDynTree'
Not searching for unused variables given on the command line.
loading initial cache file /home/icub/Desktop/robot/robotology-superbuild/build/src/iDynTree/CMakeFiles/YCMTmp/iDynTree-cache-Release.cmake
CMake Error at /usr/local/lib/cmake/YARP/YARPConfig.cmake:40 (message):
  File or directory /usr/local/share/yarp referenced by variable
  YARP_DATA_INSTALL_DIR_FULL does not exist !
Call Stack (most recent call first):
  /usr/local/lib/cmake/YARP/YARPConfig.cmake:97 (set_and_check)
  cmake/iDynTreeDependencies.cmake:37 (find_package)
  cmake/iDynTreeDependencies.cmake:69 (idyntree_handle_dependency)
  CMakeLists.txt:39 (include)

-- Configuring incomplete, errors occurred!
See also "/home/icub/Desktop/robot/robotology-superbuild/build/src/iDynTree/CMakeFiles/CMakeOutput.log".
make[2]: *** [CMakeFiles/iDynTree.dir/build.make:93: src/iDynTree/CMakeFiles/YCMStamp/iDynTree-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:1441: CMakeFiles/iDynTree.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

traversaro, 2022-06-28:
It seems that the installation of YARP in `/usr/local` is corrupted. If you are sure that installation is coming from a YARP 3.6.0, what I suggest to cleanup the system is to re-install a YARP 3.6.0 build in `/usr/local` , and then uninstall the files with `xargs rm < install_manifest.txt`.

SSSSamZhu, 2022-06-29:
> If you are sure that installation is coming from a YARP 3.6.0, what I suggest to cleanup the system is to re-install a YARP 3.6.0 build in , and then uninstall the files with 

This is what I intended to do at first, but when I recompiled the source file, he reported this error again. So I can't compile. 
Refer to the steps here :  https://www.yarp.it/latest//install_yarp_linux.html
```

--  [ ]   Plugin: yarpmod_grabber (ENABLE_yarpmod_grabber)
--  [ ]   Plugin: yarpmod_fakeMotor (ENABLE_yarpmod_fakeMotor)
--  [ ]   Plugin: yarpmod_test_motor (ENABLE_yarpmod_test_motor)
--  [ ] Bindings (YARP_COMPILE_BINDINGS)
--  [ ]   Java bindings (CREATE_JAVA)
--  [ ]   Python bindings (CREATE_PYTHON)
--  [ ]   Perl bindings (CREATE_PERL)
--  [ ]   CSharp bindings (CREATE_CSHARP)
--  [ ]   TCL bindings (CREATE_TCL)
--  [ ]   Ruby bindings (CREATE_RUBY)
--  [ ]   Lua bindings (CREATE_LUA)
--  [ ]   Octave bindings (CREATE_OCTAVE)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/icub/Desktop/robot/yarp/build
icub@isaacliang-LC3:~/Desktop/robot/yarp/build$ make
[  0%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/base64.cpp.o
[  0%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcClient.cpp.o
[  0%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcServerConnection.cpp.o
....
[ 16%] Built target YARP_init
[ 16%] Building CXX object src/yarpidl_rosmsg/CMakeFiles/yarpidl_rosmsg.dir/src/main.cpp.o
[ 16%] Building CXX object src/yarpidl_rosmsg/CMakeFiles/yarpidl_rosmsg.dir/src/RosType.cpp.o
[ 16%] Building CXX object src/yarpidl_rosmsg/CMakeFiles/yarpidl_rosmsg.dir/src/RosTypeCodeGenYarp.cpp.o
[ 16%] Building C object src/yarpidl_rosmsg/CMakeFiles/yarpidl_rosmsg.dir/__/__/extern/md5/md5.c.o
[ 16%] Linking CXX executable ../../bin/yarpidl_rosmsg
[ 16%] Built target yarpidl_rosmsg
[ 16%] Generating code from ../../../extern/ros/geometry2/tf2_msgs/msg/TF2Error.msg
Failed to open tf2_msgs_TF2Error.h for writing
make[2]: *** [src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_tf2_msgs.dir/build.make:105: src/libYARP_rosmsg/src/yarp/rosmsg/tf2_msgs/TF2Error.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:4524: src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_tf2_msgs.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
icub@isaacliang-LC3:~/Desktop/robot/yarp/build$ 
```

traversaro, 2022-06-29:
Are you sure you have space in the disk?

SSSSamZhu, 2022-06-29:
Sure, we still have 200 gigabytes of disk capacity .
[image]

traversaro, 2022-06-29:
Then I am not sure what could be the problem, as the superbuild instead works correctly. Are you sure that you have a clean environment, i.e. you do not source `setup.sh` from the superbuild? 

You could also try to use a different build directory, or clone the yarp repo completly in another location.

SSSSamZhu, 2022-06-29:
> Are you sure that you have a clean environment, i.e. you do not source from the superbuild?setup.sh

You are right, because I already executed these commands when I compiled YARP separately from the root directory. Thank you for your warning. 
https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/user-env/?h=env
So I changed the environment variable of superbuild into one that compiles YARP alone, and successfully installed yarp that compiles yarp alone. If nothing else, I confirm that the separately compiled YARP is completely uninstalled

```
Setting up yarp and iCub env vars
icub@isaacliang-LC3:~/Desktop/robot/yarp/build$ sudo xargs rm < install_manifest.txt
[sudo] password for icub: 
rm: cannot remove '/usr/local/include/yarp/os/ConstString.h': No such file or directory
icub@isaacliang-LC3:~/Desktop/robot/yarp/build$ sudo xargs rm < install_manifest.txt
rm: cannot remove '/usr/local/share/yarp/cmake/YarpPlugin.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/YarpIDL.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/YarpInstallationHelpers.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/YarpPrintFeature.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/YarpDeprecatedOption.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_carrier.cpp.in': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_RFModule.cpp.in': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_device.cpp.in': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_portmonitor.cpp.in': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_library.cpp.in': No such file or directory
rm: cannot remove '/usr/local/share/yarp/cmake/template/yarp_plugin_yarpdev_main.cpp.in': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_priv_xmlrpcpp.a': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_xmlrpcpp/YARP_priv_xmlrpcppConfigVersion.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_xmlrpcpp/YARP_priv_xmlrpcppConfig.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_xmlrpcpp/YARP_priv_xmlrpcppTargets.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_xmlrpcpp/YARP_priv_xmlrpcppTargets-release.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_priv_qgvcore.a': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_qgvcore/YARP_priv_qgvcoreConfigVersion.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_qgvcore/YARP_priv_qgvcoreConfig.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_qgvcore/YARP_priv_qgvcoreTargets.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_qgvcore/YARP_priv_qgvcoreTargets-release.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_priv_hmac.a': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_hmac/YARP_priv_hmacConfigVersion.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_hmac/YARP_priv_hmacConfig.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_hmac/YARP_priv_hmacTargets.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_priv_hmac/YARP_priv_hmacTargets-release.cmake': No such file or directory
rm: cannot remove '/usr/local/share/yarp/plugins/yarp.ini': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/dirs.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/string.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/filesystem.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/environment.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/system.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/options.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/compiler.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/api.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/version.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/conf/numeric.h': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_conf/YARP_confConfigVersion.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_conf/YARP_confConfig.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/cmake/YARP_conf/YARP_confTargets.cmake': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_os.so.3.6.0': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_os.so.3': No such file or directory
rm: cannot remove '/usr/local/lib/libYARP_os.so': No such file or directory
rm: cannot remove '/usr/local/include/yarp/os/impl/AuthHMAC.h': No such file or directory
rm: cannot remove '/usr/local/include/yarp/os/impl/BottleImpl.h': No such file or directory
[... cut ...]
  cmake/iDynTreeDependencies.cmake:37 (find_package)
  cmake/iDynTreeDependencies.cmake:69 (idyntree_handle_dependency)
  CMakeLists.txt:39 (include)

-- Configuring incomplete, errors occurred!
See also "/home/icub/Desktop/robot/robotology-superbuild/build/src/iDynTree/CMakeFiles/CMakeOutput.log".
make[2]: *** [CMakeFiles/iDynTree.dir/build.make:93: src/iDynTree/CMakeFiles/YCMStamp/iDynTree-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:1441: CMakeFiles/iDynTree.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

traversaro, 2022-06-29:
For some reason the system installed YARP installed its cmake files in `/usr/local/lib/x86_64-linux-gnu/cmake`, no idea why. Probably at this point it is easier to manually remove this files, i.e. :
~~~
cd 
/usr/local/lib/x86_64-linux-gnu/cmake
sudo rm -r ./YARP
sudo rm -r ./YARP_*
~~~

SSSSamZhu, 2022-06-30:
```

Setting up yarp and iCub env vars
icub@isaacliang-LC3:~$ yarpserver 

     ==========================
    | __  __ ___  ____   ____  |
    | \ \/ //   ||  _ \ |  _ \ |
    |  \  // /| || |/ / | |/ / |
    |  / // ___ ||  _ \ |  _/  |
    | /_//_/  |_||_| \_\|_|    |
    |                          |
     ==========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
[WARNING] |yarp.os.MultiNamespace| Could not find namespace the_robot_network_namespace
[ERROR] |yarp.os.Port|/| Port / failed to activate at tcp://10.43.239.142:10000/ (address conflict)
[ERROR] Name server failed to open
icub@isaacliang-LC3:~$ yarpmanager
[ERROR] |yarp.os.Property| cannot read from ymanager.ini
[WARNING] |yarp.os.MultiNamespace| Could not find namespace the_robot_network_namespace
icub@isaacliang-LC3:~$ 

```

traversaro, 2022-06-30:
Hi @SSSSamZhu, can you open a separate issue in https://github.com/robotology/community for this problem? I think that we debugged enough problems in this issue, and for clarity it is good to have an issue for each problem, thanks!

SSSSamZhu, 2022-06-30:
> Hi @SSSSamZhu, can you open a separate issue in https://github.com/robotology/community for this problem? I think that we debugged enough problems in this issue, and for clarity it is good to have an issue for each problem, thanks!

ok, I will close this issue. Thanks!

closed 2022-06-30 by SSSSamZhu