robotology/robotology-superbuild #1187: Clarification on the usage of XDG_DATA_DIRS

xEnVrE, 2022-06-27:
Recently I have tried to setup a `robotology-superbuild` install on my system (Arch Linux with `5.18.7-arch1-1` kernel).

After sourcing the `build/install/share/robotology-superbuild/setup.sh` file, as per https://github.com/robotology/robotology-superbuild#configure-your-environment, I found out that my system could not run any UI dialog reliably anymore as many of the png files associated to icons could not be found. This was causing erratic crashes or strange behaviors.

After investigating a bit I found that the `setup.sh` file sets the `XDG_DATA_DIRS` as follows:

https://github.com/robotology/robotology-superbuild/blob/bab995bf57661af9f20d0907431910bac17096d1/cmake/template/setup.sh.in#L17

Unfortunately, in some systems, as the one I am using apparently, if the `XDG_DATA_DIRS` is not set or empty, a default value of `/usr/local/share/:/usr/share/` is assumed. However, when the `XDG_DATA_DIRS` is populated by the `setup.sh`, given that it is empty, the new value will just contain `<path_to_ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX>/share`, i.e. the reference to `/usr/local/share` and `/usr/share` is missing.

Of course, this seems to be very system-dependent and might be easily solved if the user exports `XDG_DATA_DIRS` correctly. However, on some systems this might create strange behaviors and it is difficult to track them down and associate them with the `robotology-superbuild` installation.

Wondering if you could suggest anything to mitigate the problem, e.g. adding even a simple disclaimer in the README.

Thanks

cc @traversaro

traversaro, 2022-06-27:
Thanks for reporting the issue so clearly. This is a indeed a pain point, in the `setup.sh` we should just append to `XDG_DATA_DIRS` if it is not defined, otherwise we should first define it as `/usr/local/share/:/usr/share/` and only later append, I will try to fix this issue.

For reference, this is the reason why this was added in the first place:
* https://github.com/robotology/robotology-superbuild/issues/622
* https://github.com/robotology/robotology-superbuild/pull/759

@HosameldinMohamed just a curiosity, did you ever experienced something like that?

xEnVrE, 2022-06-27:
Thanks for the explanation @traversaro

xEnVrE, 2022-06-27:
> Interestingly, I have an Archlinux system at home and I didn't encounter this issue there as well, but probably I have installed something that set XDG_DATA_DIRS before configuring the superbuild, such as Snap or Flatpak.

I can confirm I did not install any of those and by default my system does not have the `XDG_DATA_DIRS` exported.

traversaro, 2022-06-27:
Interesting, I wonder that if this packages set the `XDG_DATA_DIRS`, they make sure to set it to `/usr/local/share:/usr/share` before appending more values.

traversaro, 2022-09-13:
> > Interesting, I wonder that if this packages set the `XDG_DATA_DIRS`, they make sure to set it to `/usr/local/share:/usr/share` before appending more values.
> 
> Probably that's what you are looking for https://github.com/flatpak/flatpak/blob/main/profile/flatpak.sh

Indeed, the line https://github.com/flatpak/flatpak/blob/3a93ef4842f7dac3eed02e2a45091cd358162b6a/profile/flatpak.sh#L25 seems to do almost exactly that.

traversaro, 2022-09-13:
With @RiccardoZuppetti we experienced another nasty bug relatedo to this, in a Docker image in which `snapd` was not installed and so where `XDG_DATA_DIRS` was not defined. In particular, there `matplotlib` was not working with error: 
~~~
 File "/home/riccardozuppetti/git/adherent/src/adherent/trajectory_control/trajectory_controller.py", line 671, in retrieve_contacts
    plt.figure()
  File "/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/pyplot.py", line 787, in figure
    manager = new_figure_manager(
  File "/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/pyplot.py", line 306, in new_figure_manager
    return _backend_mod.new_figure_manager(*args, **kwargs)
  File "/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/backend_bases.py", line 3494, in new_figure_manager
    return cls.new_figure_manager_given_figure(num, fig)
  File "/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/backend_bases.py", line 3500, in new_figure_manager_given_figure
    manager = cls.FigureManager(canvas, num)
  File "/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/backends/backend_gtk3.py", line 324, in __init__
    self.window.set_icon_from_file(window_icon)
gi.repository.GLib.Error: gdk-pixbuf-error-quark: Couldn’t recognize the image file format for file “/home/riccardozuppetti/.local/lib/python3.8/site-packages/matplotlib/mpl-data/images/matplotlib.svg” (3)
~~~

Better to finally fix this before we waste more time on it.