robotology/robotology-superbuild #1190: Update WSL2 documentation to cover WSLg

traversaro, 2022-06-28:
Reported by @FabioBergonti . Basically if a user is on Win11 and can use WSLg, we can skip setting the DISPLAY env variable and install a X server, see https://github.com/robotology/robotology-superbuild#wsl2 .

traversaro, 2023-01-25:
Ok. I probably forgot about this because I wanted to suggest IIT users to update to Windows 11 and WSLg before tackling this, but this proveed to be harder than I expected due to some internal policy. Anyhow, now WSLg is also available on Windows 10 (see https://devblogs.microsoft.com/commandline/the-windows-subsystem-for-linux-in-the-microsoft-store-is-now-generally-available-on-windows-10-and-11/), so probably we can just document WSLg workflow. 

@martinaxgloria just for reference, which Windows version did you got installed in your laptop? 

fyi @pattacini @davidelasagna

martinaxgloria, 2023-01-25:
Hi @traversaro, I had some problems with WSL on Windows 10, but I have updated to Windows 11 and now everything works correctly!

traversaro, 2023-02-14:
I double checked with @davidelasagna, and we (i.e. IIT) currently install on new laptops Windows 10 Version 22H2 (OS build 19045), while WSLg is available on Windows 10 since build 19044 (see https://learn.microsoft.com/en-us/windows/wsl/tutorials/gui-apps). So I think we can safely update the docs to just refer to WSLg.

closed 2023-02-14 by traversaro