robotology/robotology-superbuild #1191: Non-working WSL2 configuration as of June 2022

traversaro, 2022-06-28:
Both me and @FabioBergonti are not able to configure a mixed WSL2/Windows host system with a YARP network as described in https://github.com/robotology/robotology-superbuild#connect-to-a-yarp-server-on-a-windows-host-on-wsl2 . In particular, the Linux system is not able to connect to a yarpserver launched on Windows, while a Windows client can connect to a yarpserver launched on Linux. 

A simpler version of this problem is that Linux can't ping the Windows host, so the issue can be related to https://github.com/microsoft/WSL/issues/4171 .

traversaro, 2022-06-28:
I am quite sure that this configuration was working fine back in https://github.com/robotology/robotology-superbuild/pull/468 when I wrote that documentation, as  for sure I tried back then. 

@pattacini @Nicogene (and perhaps @sgiraz) do you have any experience with this problem?

pattacini, 2022-06-28:
Hi @traversaro 

I've just made a quick test with this scenario:
- WINDOWS
    - YARP version: YARP version 3.7.0+46-20220616.6+gitdd9bc1f65.
    - Launched `yarpserver`
    - Launched `yarp read /read`
- WSL2
    - robotology-superbuild v2022.05.0
    - `yarp where` can see the server correctly 👍🏻 
    - `yarp write ... /read` can send messages correctly 👍🏻

traversaro, 2022-06-28:
Thanks @pattacini ! Can you also test on Linux/WSL2 if:
~~~
ping $(grep nameserver /etc/resolv.conf | awk '{print $2}')
~~~
is working? If that is working on your system (it does not work for me), I guess this may be related to firewall settings.

pattacini, 2022-06-28:
Yep, I can confirm that the ping works smoothly.

sgiraz, 2022-06-29:
cc @vvasco