robotology/robotology-superbuild #1192: Setting LD_LIBRARY_PATH in gazebo-yarp-plugins creates incompatibility with system programs

traversaro, 2022-06-28:
@FabioBergonti reported that programs such as `gedit`, `htop`  were not working in an environment when robotology packages were installed. It turns out that this is due to `LD_LIBRARY_PATH` being set. We should investigate how to avoid setting it, even if we may need to solve https://github.com/robotology/robotology-superbuild/issues/1175 .

traversaro, 2022-06-28:
Note that `LD_LIBRARY_PATH`  is probably set by https://github.com/osrf/gazebo/blob/e2849f96ac638bdd21b79e2300c1433beb2e2001/cmake/setup.sh.in#L19 , but it contains `$CONDA_PREFIX/lib`  only if `gazebo-yarp-plugins` is installed as well.

traversaro, 2022-06-28:
> Note that `LD_LIBRARY_PATH` is probably set by https://github.com/osrf/gazebo/blob/e2849f96ac638bdd21b79e2300c1433beb2e2001/cmake/setup.sh.in#L19 , but it contains `$CONDA_PREFIX/lib` only if `gazebo-yarp-plugins` is installed as well.

This can be fixed by changing:
~~~
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${GAZEBO_PLUGIN_PATH}
~~~
to
~~~
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:${GAZEBO_SETUP_SH_PLUGIN_PATH}
~~~

Nicogene, 2026-02-16:
I don't think that this issue is still valid, in any case we are moving to gz-sim