robotology/robotology-superbuild #1220: how to get actual graping force from tactile sensor in icub robot in python

ArfnZl, 2022-07-29:
hello sir
i have  addee tactile sensors in icub in python code also it is gripping from different axix but it is giving values only 0 and 255.i want to add pressure sensor to measure grasping force which will give values of grasping force.kindly help me with working example.here is the relevant my code 
arp.Network.init()
            # Initialise YARP
            # create the network
yarp.Network.init()

# define ports
inport = yarp.Port()
bottle = yarp.Bottle()

# activate ports
inport.open("/reader")
yarp.Network.connect("/icubSim/skin/right_hand_comp","/reader")

# create connection betweeen ports inside the network:
# note: connect(writer, reader)  ok
#      connect(reader, writer)  not ok
yarp.Network.connect("/writer", inport.getName())

# read the messge

# close the network
yarp.Network.fini()

RightArm=yarp.Port()
RightArm.open('/python/RightArm')
yarp.Network.connect('/icubSim/right_arm','/python/RightArm')
        # prepare a property object           
armOptions = yarp.Property()
armOptions.put('device','remote_controlboard')
armOptions.put('local','/client/right_arm')
armOptions.put('remote','/icubSim/right_arm')
        # create remote driver
armDriver = yarp.PolyDriver(armOptions)
        #query motor control interfacesarmPos = armDriver.viewIPositionControl()

armPos = armDriver.viewIPositionControl()
armEncs = armDriver.viewIEncoders()
        # retrieve the number of joints
jntsArm=armPos.getAxes()

        
tmp=yarp.Vector(jntsArm)

a=[180, 10, 0, 10, 60, 40, 0, 40, 0, 0, 0, 0, 0, 0, 0, 0, 90]
for i in range(0,17):
    sigma=15.0
    xrange=np.arange(0, 300, 50)
    center=(tmp.get(i)+a[i])
    tmp.set(i, center)
 
    gaussianResp=gaussian1D_bank(center,sigma,xrange)
    inport.read(bottle)
    print("Received ", bottle.toString())
    yarp.delay(5)
this code is giving values of tactile sensors 0 and 255.i want to find the values of grasping force

ArfnZl, 2022-07-31:
@traversaro 
i have sent my code please reply me

traversaro, 2022-08-22:
See https://github.com/robotology/robotology-superbuild/issues/1232#issuecomment-1222375132 .

closed 2022-08-22 by traversaro