robotology/robotology-superbuild #1221: yarpmanager :Unable to open graphical interface in Windows10 OS

SSSSamZhu, 2022-07-29:
I've been running `yarpmanager` on my Windows host without getting a response. The result is the same this time. It seems that the result is not normal. Could you provide me with a solution? Thanks!

```
(robsub) C:\Users\samgmzhu>yarpmanager

(robsub) C:\Users\samgmzhu>yarpmanager --apppath ./

(robsub) C:\Users\samgmzhu>yarpmanager --apppath ./

(robsub) C:\Users\samgmzhu>
```

Thanks.

traversaro, 2022-07-29:
Hi @SSSSamZhu, can you provide:
* Version of YARP used
* If you installed yarp from binaries or from source
* Version of conda dependencies installed, by providing the output of `mamba list`

Thanks!

SSSSamZhu, 2022-07-29:
I am so sorry that I forgot to provide my basic information again. Thank you for your warning.

```
(robsub) C:\Users\samgmzhu\robotology-superbuild>git branch
* (HEAD detached at v2022.05.0)
  master

(robsub) C:\Users\samgmzhu\robotology-superbuild>yarp --version
YARP version 3.7.0

(robsub) C:\Users\samgmzhu>mamba list
# packages in environment at C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub:
#
# Name                    Version                   Build  Channel
ace                       7.0.7                h0e60522_0    conda-forge
asio                      1.22.1               h0e60522_0    conda-forge
assimp                    5.2.4                hc2aa0de_0    conda-forge
blosc                     1.21.1               h74325e0_3    conda-forge
boost                     1.74.0          py310hc781a3c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.6.15            h5b45459_0    conda-forge
cairo                     1.16.0            h0ac17fb_1011    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.2               h39d44d4_0    conda-forge
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.4.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               h6676e37_9    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.11.0              hfac46e6_6    conda-forge
geos                      3.10.3               h39d44d4_0    conda-forge
geotiff                   1.7.1                h714bc5f_2    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
git                       2.37.1               h57928b3_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glfw                      3.3.7                h8ffe710_0    conda-forge
glib                      2.70.2               h7755175_4    conda-forge
glib-tools                2.70.2               h7755175_4    conda-forge
glm                       0.9.9.4              h1ad3211_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  2.50.0               had6c3a3_2    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gst-plugins-base          1.20.3               he07aa86_0    conda-forge
gstreamer                 1.20.3               hdff456e_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
harfbuzz                  4.3.0                h27de254_0    conda-forge
[... 124 lines cut ...]
urdfdom_headers           1.1.0                h5362a0b_0    conda-forge
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vs2017_win-64             19.16.27033          hb90652a_6    conda-forge
vswhere                   3.0.3                h57928b3_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xerces-c                  3.2.3                h0e60522_5    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.12               h8ffe710_1    conda-forge
zstd                      1.5.2                h6255e5f_1    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

traversaro, 2022-07-29:
Thanks @SSSSamZhu . It is indeed strange that `yarpmanager` does not start. On my system, `yarpmanager` indeed returns immediatly, but the `yarpmanager` windows is opened, so a bit of questions:
* Are you sure that the yarpmanager window is not open somewhere even if the application is returning?
* Did you tried to ensure that yarpserver is running before launching yarpmanager?
* Does `yarpmanager-console` work?
* Does `yarpview` work?
* Does `yarpmotorgui` work?

traversaro, 2022-07-29:
(Related issue: https://github.com/robotology/walking-teleoperation/issues/98#issuecomment-1198943807 ).

SSSSamZhu, 2022-07-29:
Hi @traversaro ,
> Did you tried to ensure that yarpserver is running before launching yarpmanager?

Yes

> Does yarpmanager-console work?

```
(robsub) C:\Users\samgmzhu>yarpmanager-console
[ERROR] |yarp.os.Property| cannot read from ymanager.ini

__   __
\ \ / / __ ___   __ _ _ __   __ _  __ _  ___ _ __
 \ V / '_ ` _ \ / _` | '_ \ / _` |/ _` |/ _ \ '__|
  | || | | | | | (_| | | | | (_| | (_| |  __/ |
  |_||_| |_| |_|\__,_|_| |_|\__,_|\__, |\___|_|
                                  |___/

type "help" for more information.

>>
```

> Does `yarpview` work?

[image]

> Does `yarpmotorgui` work?

traversaro, 2022-07-29:
> > Does `yarpview` work?
> 
> [image]

This is by far the most informative error. This could be related to https://github.com/robotology/robotology-superbuild/issues/871, https://github.com/robotology/robotology-superbuild/pull/883 and https://github.com/robotology/robotology-superbuild/issues/882 .

Can you report the output of `set` ? I think there is some other `Qt` library in the path that is being used instead of the one of conda.

SSSSamZhu, 2022-08-02:
Hi @traversaro ,
Sorry for the delay due to some accidents. This is the output of `set`. Thanks
 
```
(robsub) C:\Users\samgmzhu>set
ALLUSERSPROFILE=C:\ProgramData
AMENT_PREFIX_PATH=;C:\Users\samgmzhu\robotology-superbuild\build\install
APPDATA=C:\Users\samgmzhu\AppData\Roaming
BITS=64
BLOCKFACTORY_PLUGIN_PATH=C:\Users\samgmzhu\robotology-superbuild\build\install\bin\blockfactory
CC=cl.exe
CMAKE_GEN=Visual Studio 15 2017 Win64
CMAKE_GENERATOR=Visual Studio 15 2017 Win64
CMAKE_PLAT=x64
CMAKE_PREFIX_PATH=;C:\Users\samgmzhu\robotology-superbuild\build\install
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=samgmzhu-PC0
ComSpec=C:\windows\system32\cmd.exe
CONDA_BAT=C:\Users\samgmzhu\AppData\Local\mambaforge\condabin\conda.bat
CONDA_DEFAULT_ENV=robsub
CONDA_EXE=C:\Users\samgmzhu\AppData\Local\mambaforge\Scripts\conda.exe
CONDA_PREFIX=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub
CONDA_PROMPT_MODIFIER=(robsub)
CONDA_PYTHON_EXE=C:\Users\samgmzhu\AppData\Local\mambaforge\python.exe
CONDA_SHLVL=1
CPATH=;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\include
CXX=cl.exe
DISTUTILS_USE_SDK=1
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
GAZEBO_MASTER_URI=http://localhost:11345
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
GAZEBO_MODEL_PATH=C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/share/gazebo-11/models;;C:\Users\samgmzhu\robotology-superbuild\build\install\share\gazebo\models;C:\Users\samgmzhu\robotology-superbuild\build\install\share\iCub\robots;C:\Users\samgmzhu\robotology-superbuild\build\install\share
GAZEBO_PLUGIN_PATH=C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/bin/gazebo-11/plugins;
GAZEBO_RESOURCE_PATH=C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/share/gazebo-11;
GDAL_DATA=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\share\gdal
GEM_HOME=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\share\rubygems
GEOTIFF_CSV=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\share\epsg_csv
HOME=C:\Users\samgmzhu
HOMEDRIVE=C:
HOMEPATH=\Users\samgmzhu
LD_LIBRARY_PATH=;C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/bin/gazebo-11/plugins
LIB=;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\lib
LOCALAPPDATA=C:\Users\samgmzhu\AppData\Local
LOGONSERVER=\\GM-LOBSTER
MSSdk=1
MSYS2_ARG_CONV_EXCL=/AI;/AL;/OUT;/out
MSYS2_ENV_CONV_EXCL=CL
NEWER_VS_WITH_OLDER_VC=0
NUMBER_OF_PROCESSORS=8
OGRE_RESOURCE_PATH=C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/bin
OS=Windows_NT
Path=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\share\rubygems\bin;;C:/Users/samgmzhu/AppData/Local/mambaforge/envs/robsub/Library/bin/gazebo-11/plugins;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\mingw-w64\bin;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\usr\bin;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\bin;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Scripts;C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\bin;C:\Users\samgmzhu\AppData\Local\mambaforge\condabin;D:\Anaconda3;D:\Anaconda3\Library\mingw-w64\bin;D:\Anaconda3\Library\usr\bin;D:\Anaconda3\Library\bin;D:\Anaconda3\Scripts;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0;C:\windows\System32\OpenSSH;D:\Microsoft VS Code\bin;C:\Program Files\Bandizip;C:\Program Files\Git\bin;C:\Program Files\Git\mingw64\libexec\git-core;C:\Program Files\CMake\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Users\samgmzhu\AppData\Local\Microsoft\WindowsApps;.;C:\Users\samgmzhu\robotology-superbuild\build\install\bin
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 9, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=9e09
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROJ_LIB=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\share\proj
PROJ_NETWORK=ON
PROMPT=(robsub) $P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\windows\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
PY_VCRUNTIME_REDIST=\bin\vcruntime140.dll
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=C:\Users\samgmzhu\robotology-superbuild\build\install
ROBOTOLOGY_SUPERBUILD_SOURCE_DIR=C:\Users\samgmzhu\robotology-superbuild
ROS_PACKAGE_PATH=;C:\Users\samgmzhu\robotology-superbuild\build\install\share
SDF_PATH=C:\Users\samgmzhu\AppData\Local\mambaforge\envs\robsub\Library\share\sdformat9\1.7
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\windows
TEMP=C:\Users\samgmzhu\AppData\Local\Temp
TMP=C:\Users\samgmzhu\AppData\Local\Temp
USERDNSDOMAIN=TENCENT.COM
USERDOMAIN=TENCENT
USERDOMAIN_ROAMINGPROFILE=TENCENT
USERNAME=samgmzhu
USERPROFILE=C:\Users\samgmzhu
USE_NEW_CMAKE_GEN_SYNTAX=0
VSINSTALLDIR=C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\
VS_MAJOR=15
VS_VERSION=15.0
VS_YEAR=2017
windir=C:\windows
WindowsSdkDir=C:\Program Files (x86)\Windows Kits\10\
WindowsSDKVer=10.0.19041.0
WXDRIVE_START_ARGS=--wxdrive-setting=0 --disable-gpu --disable-software-rasterizer --enable-features=NetworkServiceInProcess
XDG_DATA_DIRS=;C:\Users\samgmzhu\robotology-superbuild\build\install\share
YARP_DATA_DIRS=;C:\Users\samgmzhu\robotology-superbuild\build\install\share\yarp;C:\Users\samgmzhu\robotology-superbuild\build\install\share\iCub;C:\Users\samgmzhu\robotology-superbuild\build\install\share\ICUBcontrib
ZES_ENABLE_SYSMAN=1
```

SSSSamZhu, 2022-08-02:
After I removed the environment variable for anaconda3, an error message returned after running `yarpmanager`. I'm sorry for the inconvenience caused by the Chinese system. The system prompts that QT5Gui cannot be found. It is suggested that I reinstall the program, and what should I do next?

 

 

 
[image]

closed 2022-08-02 by SSSSamZhu

traversaro, 2022-08-02:
Yes, in general it is not a good idea to install two conda distribution such as `Anaconda3` and `mambaforge`, and add both of them to the `PATH`, for example via default activation of the `base` environment, as they could install different version of the same dll, as you see in this case. This is the reason why we suggest to set `conda config --set auto_activate_base false` in https://github.com/robotology/robotology-superbuild/blob/fa1505f6de204c00673fe2f9435f795c1f794ab9/doc/install-mambaforge.md .

SSSSamZhu, 2022-08-02:
All in all，thank you very much.