robotology/robotology-superbuild #1232: how to get pressure sensor commands for grasping  in python icub simulator

ArfnZl, 2022-08-19:
Hello sir
i am searching pressure sensor commands for grasping in python icub simulator. i am sending code of my code 

yarp.Network.fini()

RightArm=yarp.Port()
RightArm.open('/python/RightArm')
yarp.Network.connect('/icubSim/right_arm','/python/RightArm')
# prepare a property object
armOptions = yarp.Property()
armOptions.put('device','remote_controlboard')
armOptions.put('local','/client/right_arm')
armOptions.put('remote','/icubSim/right_arm')
# create remote driver
armDriver = yarp.PolyDriver(armOptions)
#query motor control interfacesarmPos = armDriver.viewIPositionControl()

armPos = armDriver.viewIPositionControl()
armEncs = armDriver.viewIEncoders()
# retrieve the number of joints
jntsArm=armPos.getAxes()

tmp=yarp.Vector(jntsArm)

a=[180, 10, 0, 10, 60, 40, 0, 40, 0, 0, 0, 0, 0, 0, 0, 0, 90]
for i in range(0,17):
sigma=15.0
xrange=np.arange(0, 300, 50)
center=(tmp.get(i)+a[i])
tmp.set(i, center)

gaussianResp=gaussian1D_bank(center,sigma,xrange)
inport.read(bottle)
print("Received ", bottle.toString())
yarp.delay(5)

traversaro, 2022-08-22:
Hi @ArfnZl, I repeately asked you question that you did not replied to in the following issues:
* https://github.com/robotology/robotology-superbuild/issues/1098#issuecomment-1102618534 
* https://github.com/robotology/robotology-superbuild/issues/1082#issuecomment-1087230288
* https://github.com/robotology/robotology-superbuild/issues/1056#issuecomment-1060677969
* https://github.com/robotology/robotology-superbuild/issues/1119#issuecomment-1126935169
* https://github.com/robotology/robotology-superbuild/issues/1188#issuecomment-1167394749

To be honest, it is not completly polite to continue to open new issues without replying to old ones.

If you want that someone reply to your questions, you should:
* **MOST IMPORTANT: Stop opening new issues without replying to old ones.**
* Open the issues in the appropriate place. As I repeatedly, this is not an issue tracker for generic questions on iCub. For that, you can open an issue in https://github.com/robotology/community/discussions .
* Please explain the context. What is your goal? Which simulator are you using? Which version of the software? If you have a snippet of code, explain how are you launching it: which program did you started before? `yarpserver` ? Furthermore, please attach exactly the code that you are running as a .py, so that if somebody wants can run it as " Minimal, Reproducible Example ", see https://stackoverflow.com/help/minimal-reproducible-example .

traversaro, 2023-09-11:
Closing as the original author did not provided the requested feedback.

closed 2023-09-11 by traversaro