robotology/robotology-superbuild #1234: Homebrew CI Failure mid August 2022

traversaro, 2022-08-21:
Since ~2022/08/15 the Homebrew CI failed, with this error:
~~~
2022-08-21T03:29:06.3086910Z [3/128] Building CXX object src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseDynamicsWithCompliantContacts.cpp.o
2022-08-21T03:29:06.3187630Z FAILED: src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseDynamicsWithCompliantContacts.cpp.o 
2022-08-21T03:29:06.3292980Z /Applications/Xcode_13.2.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DContinuousDynamicalSystem_EXPORTS -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -D_USE_MATH_DEFINES -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ContinuousDynamicalSystem/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ContactModels/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Math/include -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Conversions/include -isystem /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/local/include/eigen3 -isystem /usr/local/include -O3 -DNDEBUG -isysroot /Applications/Xcode_13.2.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.1.sdk -mmacosx-version-min=11.6 -fPIC -std=gnu++17 -MD -MT src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseDynamicsWithCompliantContacts.cpp.o -MF src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseDynamicsWithCompliantContacts.cpp.o.d -o src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseDynamicsWithCompliantContacts.cpp.o -c /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ContinuousDynamicalSystem/src/FloatingBaseDynamicsWithCompliantContacts.cpp
2022-08-21T03:29:06.3389430Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ContinuousDynamicalSystem/src/FloatingBaseDynamicsWithCompliantContacts.cpp:15:
2022-08-21T03:29:06.3491400Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include/BipedalLocomotion/TextLogging/Logger.h:12:
2022-08-21T03:29:06.3592160Z In file included from /usr/local/include/spdlog/fmt/ostr.h:20:
2022-08-21T03:29:06.3693300Z In file included from /usr/local/include/fmt/ostream.h:14:
2022-08-21T03:29:06.3793700Z In file included from /usr/local/include/fmt/format.h:48:
2022-08-21T03:29:06.3895280Z /usr/local/include/fmt/core.h:1732:3: error: static_assert failed due to requirement 'formattable' "Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt"
2022-08-21T03:29:06.3995330Z   static_assert(
2022-08-21T03:29:06.4096120Z   ^
2022-08-21T03:29:06.4198000Z /usr/local/include/fmt/core.h:1753:10: note: in instantiation of function template specialization 'fmt::detail::make_value<fmt::basic_format_context<fmt::appender, char>, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>> &>' requested here
2022-08-21T03:29:06.4297820Z   return make_value<Context>(val);
2022-08-21T03:29:06.4398120Z          ^
2022-08-21T03:29:06.4500470Z /usr/local/include/fmt/core.h:1875:23: note: in instantiation of function template specialization 'fmt::detail::make_arg<true, fmt::basic_format_context<fmt::appender, char>, fmt::detail::type::custom_type, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>, 0>' requested here
2022-08-21T03:29:06.4599340Z         data_{detail::make_arg<
2022-08-21T03:29:06.4700460Z                       ^
2022-08-21T03:29:06.4801900Z /usr/local/include/fmt/core.h:1894:10: note: in instantiation of function template specialization 'fmt::format_arg_store<fmt::basic_format_context<fmt::appender, char>, const char *, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>::format_arg_store<const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.4901110Z   return {std::forward<Args>(args)...};
2022-08-21T03:29:06.5001640Z          ^
2022-08-21T03:29:06.5103020Z /usr/local/include/spdlog/logger.h:370:52: note: in instantiation of function template specialization 'fmt::make_format_args<fmt::basic_format_context<fmt::appender, char>, const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.5202840Z             fmt::detail::vformat_to(buf, fmt, fmt::make_format_args(std::forward<Args>(args)...));
2022-08-21T03:29:06.5303020Z                                                    ^
2022-08-21T03:29:06.5404990Z /usr/local/include/spdlog/logger.h:90:9: note: in instantiation of function template specialization 'spdlog::logger::log_<const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.5504840Z         log_(loc, lvl, fmt, std::forward<Args>(args)...);
2022-08-21T03:29:06.5605100Z         ^
2022-08-21T03:29:06.5706330Z /usr/local/include/spdlog/logger.h:96:9: note: in instantiation of function template specialization 'spdlog::logger::log<const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.5806130Z         log(source_loc{}, lvl, fmt, std::forward<Args>(args)...);
2022-08-21T03:29:06.5906470Z         ^
2022-08-21T03:29:06.6007830Z /usr/local/include/spdlog/logger.h:158:9: note: in instantiation of function template specialization 'spdlog::logger::log<const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.6107760Z         log(level::info, fmt, std::forward<Args>(args)...);
2022-08-21T03:29:06.6207990Z         ^
2022-08-21T03:29:06.6309450Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ContinuousDynamicalSystem/src/FloatingBaseDynamicsWithCompliantContacts.cpp:44:16: note: in instantiation of function template specialization 'spdlog::logger::info<const char *const &, Eigen::Transpose<Eigen::Matrix<double, 3, 1, 0>>>' requested here
2022-08-21T03:29:06.6409480Z         log()->info("{} The gravity vector  not found. The default one will be "
2022-08-21T03:29:06.6509750Z                ^
2022-08-21T03:29:06.6610840Z 1 error generated.
2022-08-21T03:29:06.6712020Z [4/128] Building CXX object src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FloatingBaseSystemKinematics.cpp.o
2022-08-21T03:29:06.6813520Z [5/128] Building CXX object src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/LinearTimeInvariantSystem.cpp.o
2022-08-21T03:29:06.6913080Z [6/128] Building CXX object src/ContinuousDynamicalSystem/CMakeFiles/ContinuousDynamicalSystem.dir/src/FirstOrderSmoother.cpp.o
2022-08-21T03:29:06.7014770Z ninja: build stopped: subcommand failed.
~~~

At a first glance, this could be a problem in fmt 9.0.0 that was indeed updated in Homebrew 5 days ago, see https://github.com/Homebrew/homebrew-core/pull/105132 . I checked why this was not happening on conda-forge, as in theory fmt was updated also there: https://github.com/conda-forge/fmt-feedstock/pull/30 . Apparently the spdlog of conda-forge is still using an old vendored version of fmt, see https://github.com/conda-forge/spdlog-feedstock/blob/4e0b46c6f245efc44a930d6a0de3858bfb2926cc/recipe/meta.yaml#L19 . This is probably an issue on the conda-forge side. However, fmt 9.0.0 eventually will arrive also on other distros, so perhaps we may need to look into that?

fyi @GiulioRomualdi @S-Dafarra

traversaro, 2022-08-21:
Indeed, the same error can be found in blf CI: https://github.com/ami-iit/bipedal-locomotion-framework/runs/7935832360?check_suite_focus=true .

traversaro, 2022-08-21:
> Apparently the spdlog of conda-forge is still using an old vendored version of fmt, see https://github.com/conda-forge/spdlog-feedstock/blob/4e0b46c6f245efc44a930d6a0de3858bfb2926cc/recipe/meta.yaml#L19 . This is probably an issue on the conda-forge side.

Regarding that, I opened an issue in https://github.com/conda-forge/spdlog-feedstock/issues/49 .

GiulioRomualdi, 2022-09-05:
Fixed in https://github.com/ami-iit/bipedal-locomotion-framework/pull/556 I think we can reenable brew CI

traversaro, 2022-09-07:
> Fixed in [ami-iit/bipedal-locomotion-framework#556](https://github.com/ami-iit/bipedal-locomotion-framework/pull/556) I think we can reenable brew CI

Also latest releases are tested, so better wait for the next blf release.

traversaro, 2022-10-28:
blf 0.10.0 was released at the end of september, we can renable ci on homebrew.

closed 2022-11-04 by traversaro