robotology/robotology-superbuild #1251: Windows Conda CI failure of start of September 2022

traversaro, 2022-09-07:
The Windows Conda CI failures started to fail with error:
~~~
-- Found PkgConfig: C:/Miniconda3/envs/test/Library/bin/pkg-config.exe (found version "0.29.2") 
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/Miniconda3/envs/test/Library/share/cmake-3.24/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  C:/Miniconda3/envs/test/Library/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/_deps/ycm-src/modules/FindOrBuildPackage.cmake:205 (include)
  cmake/RobotologySuperbuildLogic.cmake:19 (find_or_build_package)
  CMakeLists.txt:66 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: C:/Miniconda3/envs/test/Library/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
CMake Error at C:/Miniconda3/envs/test/Library/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Protobuf (missing: Protobuf_LIBRARIES) (found version
  "3.21.5")
Call Stack (most recent call first):
-- Configuring incomplete, errors occurred!
  C:/Miniconda3/envs/test/Library/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Miniconda3/envs/test/Library/share/cmake-3.24/Modules/FindProtobuf.cmake:650 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
See also "D:/a/robotology-superbuild/robotology-superbuild/build/CMakeFiles/CMakeOutput.log".
  C:/Miniconda3/envs/test/Library/lib/cmake/gazebo/gazebo-config.cmake:193 (find_package)
~~~

a similar error occured in the Gazebo CI job in which conda-forge is used (fyi @scpeters @j-rivero):
~~~
2022-09-07T03:10:18.5700886Z ====== Finding 3rd Party Packages ======
2022-09-07T03:10:18.5817163Z -- Operating system is Windows
2022-09-07T03:10:18.6769436Z -- Found PkgConfig: C:/Miniconda/envs/test/Library/bin/pkg-config.exe (found version "0.29.2") 
2022-09-07T03:10:18.6947969Z -- Checking for module 'freeimage>=3.9.0'
2022-09-07T03:10:18.7092645Z --   No package 'freeimage' found
2022-09-07T03:10:18.7094364Z --   freeimage.pc not found, trying freeimage_include_dir and freeimage_library_dir flags.
2022-09-07T03:10:18.8067305Z --   Found C:/Miniconda/envs/test/Library/include/FreeImage.h
2022-09-07T03:10:20.0990093Z --   Looking for FreeImage.h - found
2022-09-07T03:10:20.3772161Z --   Looking for libfreeimage - found
2022-09-07T03:10:22.8500472Z CMake Error at C:/Miniconda/envs/test/Library/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
2022-09-07T03:10:22.8501501Z   Could NOT find Protobuf (missing: Protobuf_LIBRARIES) (found version
2022-09-07T03:10:22.8502070Z   "3.21.5")
2022-09-07T03:10:22.8502526Z Call Stack (most recent call first):
2022-09-07T03:10:22.8503168Z   C:/Miniconda/envs/test/Library/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
2022-09-07T03:10:22.8503874Z   C:/Miniconda/envs/test/Library/share/cmake-3.24/Modules/FindProtobuf.cmake:650 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
2022-09-07T03:10:22.8504464Z   cmake/SearchForStuff.cmake:34 (find_package)
2022-09-07T03:10:22.8505111Z   CMakeLists.txt:163 (include)
2022-09-07T03:10:22.8505428Z -- Configuring incomplete, errors occurred!
~~~

The diff in the environment between the latest working and the first not-working job is:
~~~diff
--- <unnamed>
+++ <unnamed>
@@ -79,7 +79,7 @@
 imath                     3.1.5                h12d4b20_0    conda-forge
 intel-openmp              2022.1.0          h57928b3_3787    conda-forge
 ipopt                     3.14.9               h0ff4939_1    conda-forge
-ipython                   8.4.0              pyh08f2357_1    conda-forge
+ipython                   8.5.0              pyh08f2357_1    conda-forge
 irrlicht                  1.8.5                h739eaf8_2    conda-forge
 jasper                    2.0.33               h77af90b_0    conda-forge
 jedi                      0.18.1             pyhd8ed1ab_2    conda-forge
@@ -126,7 +126,7 @@
 libopencv                 4.6.0           py310hb1fdf93_4    conda-forge
 libpng                    1.6.37               h1d00b33_4    conda-forge
 libpq                     14.5                 hfcc5ef8_0    conda-forge
-libprotobuf               3.21.5               h12be248_0    conda-forge
+libprotobuf               3.21.5               h12be248_1    conda-forge
 libraw                    0.20.2               hee1bdec_1    conda-forge
 librttopo                 1.1.0               h2842628_11    conda-forge
 libsdformat               9.8.0                h33150cf_2    conda-forge
@@ -219,8 +219,8 @@
 stack_data                0.5.0              pyhd8ed1ab_0    conda-forge
 svt-av1                   1.2.1                h0e60522_0    conda-forge
 swig                      4.0.2                h57928b3_2    conda-forge
-tbb                       2021.5.0             h2d74725_1    conda-forge
-tbb-devel                 2021.5.0             h2d74725_1    conda-forge
+tbb                       2021.5.0             h91493d7_2    conda-forge
+tbb-devel                 2021.5.0             h91493d7_2    conda-forge
 tiledb                    2.11.1               h5689973_0    conda-forge
 tiny-process-library      2.0.4                h0e60522_0    conda-forge
 tinyxml                   2.6.2                h2d74725_2    conda-forge
~~~

So the problem is probably in the build `h12be248_1` of libprotobuf, that was triggered by https://github.com/conda-forge/libprotobuf-feedstock/pull/128 . This introduced the bug described in https://github.com/conda-forge/libprotobuf-feedstock/pull/128#issuecomment-1238531229, so now the change is being reverted in https://github.com/conda-forge/libprotobuf-feedstock/pull/130, so we just need to wait some time and the issue should be solved.

h-vetinari, 2022-09-07:
> so now the change is being reverted in [conda-forge/libprotobuf-feedstock#130](https://github.com/conda-forge/libprotobuf-feedstock/pull/130), so we just need to wait some time for the PR to be merged and the issue should be solved.

The reversion was only for unix. I've commented about the windows bits on the PR.

traversaro, 2022-09-08:
The problem was fixed by https://github.com/conda-forge/libprotobuf-feedstock/pull/131, see https://github.com/robotology/robotology-superbuild/actions/runs/3012141729 . Thanks a lot @h-vetinari !

closed 2022-09-08 by traversaro