robotology/robotology-superbuild #1253: Inserting any iCub model into Gazebo world crashes Gazebo with FatalException

nunoguedelha, 2022-09-08:
Steps for reproducing the issue...

- Install Gazebo and build the superbuild
```sh
conda create -n robenv
conda activate robenv
mamba update --all
mamba install -c conda-forge gazebo=11.9.1
mamba install -c conda-forge cmake compilers make ninja pkg-config
mamba install -c conda-forge ace asio assimp boost eigen freetype gazebo glew glfw glm graphviz gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json opencv pkg-config portaudio sdl sdl2 sqlite tinyxml spdlog lua soxr
cd ~/dev
git clone https://github.com/robotology/robotology-superbuild
cd robotology-superbuild
mkdir build
cd build
cmake ..
cmake --build . --config Release --target update-all
cmake --build . --config Release
```
- run Gazebo
```
gazebo --verbose
```
- insert an iCub model (e.g. iCubGazeboV3)
=> Gazebo crashes with the error:
```
[libprotobuf ERROR google/protobuf/descriptor_database.cc:559] Invalid file descriptor data passed to EncodedDescriptorDatabase::Add().
[libprotobuf FATAL google/protobuf/descriptor.cc:1357] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size):
libc++abi.dylib: terminating with uncaught exception of type google::protobuf::FatalException: CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size):
```

#### Details

- When installing the superbuild dependencies through conda, we dropped `qt-main`.
- Superbuild options set through `ccmake` interface:
```
 ROBOTOLOGY_ENABLE_DYNAMICS       ON
 ROBOTOLOGY_ENABLE_DYNAMICS_FUL   ON
 ROBOTOLOGY_PROJECT_TAGS          Stable
 ROBOTOLOGY_USES_MATLAB           ON
```

<details>
<summary>conda info:</summary>

```sh
     active environment : robotologydeps3
    active env location : /Users/nunoguedelha/miniforge3/envs/robotologydeps3
            shell level : 1
       user config file : /Users/nunoguedelha/.condarc
 populated config files : /Users/nunoguedelha/miniforge3/.condarc
                          /Users/nunoguedelha/.condarc
          conda version : 4.14.0
    conda-build version : not installed
         python version : 3.8.13.final.0
       virtual packages : __osx=10.15.7=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/nunoguedelha/miniforge3  (writable)
      conda av data dir : /Users/nunoguedelha/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/nunoguedelha/miniforge3/pkgs
                          /Users/nunoguedelha/.conda/pkgs
       envs directories : /Users/nunoguedelha/miniforge3/envs
[... 240 lines cut ...]
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.7.2                h0d85af4_0    conda-forge
xorg-libxau               1.0.9                h35c211d_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
xorg-libxext              1.3.4                h0d85af4_1    conda-forge
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxmu               1.1.3                h0d85af4_0    conda-forge
xorg-libxpm               3.5.13               h0d85af4_0    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-libxt                1.2.1                h0d85af4_2    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.12               hfe4f2af_2    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

</details>

nunoguedelha, 2022-09-08:
The **Protobuf** headers and libraries were being found in the system (/usr/local/include/google/protobuf) instead of Conda install prefix, which can be seen in the Cmake cache:
```
 Protobuf_DIR                     Protobuf_DIR-NOTFOUND
 Protobuf_INCLUDE_DIR             /usr/local/include
 Protobuf_LIBRARY_DEBUG           /usr/local/lib/libprotobuf.dylib
 Protobuf_LIBRARY_RELEASE         /usr/local/lib/libprotobuf.dylib
 Protobuf_LITE_LIBRARY_DEBUG      /usr/local/lib/libprotobuf-lite.dylib
 Protobuf_LITE_LIBRARY_RELEASE    /usr/local/lib/libprotobuf-lite.dylib
 Protobuf_PROTOC_EXECUTABLE       /Users/nunoguedelha/miniforge3/envs/robotologybin3/bin/protoc
 Protobuf_PROTOC_LIBRARY_DEBUG    /usr/local/lib/libprotoc.dylib
 Protobuf_PROTOC_LIBRARY_RELEAS   /usr/local/lib/libprotoc.dylib
```
I added to the CMake command line DCMAKE_PREFIX_PATH=$CONDA_PREFIX to force CMake to find protobuf in conda environment.

I now get the CMake config:
```
Protobuf_DIR                     Protobuf_DIR-NOTFOUND
 Protobuf_INCLUDE_DIR             /Users/nunoguedelha/miniforge3/envs/robotologydeps3/include
 Protobuf_LIBRARY_DEBUG           /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotobuf.dylib
 Protobuf_LIBRARY_RELEASE         /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotobuf.dylib
 Protobuf_LITE_LIBRARY_DEBUG      /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotobuf-lite.dylib
 Protobuf_LITE_LIBRARY_RELEASE    /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotobuf-lite.dylib
 Protobuf_PROTOC_EXECUTABLE       /Users/nunoguedelha/miniforge3/envs/robotologydeps3/bin/protoc
 Protobuf_PROTOC_LIBRARY_DEBUG    /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotoc.dylib
 Protobuf_PROTOC_LIBRARY_RELEAS   /Users/nunoguedelha/miniforge3/envs/robotologydeps3/lib/libprotoc.dylib
```

Rebuilding...

traversaro, 2023-09-11:
I guess this was fixed by rebuilding against the correct libprotobuf libraries, if anyone is still experiencing this issue, please open a new issue, thanks!

closed 2023-09-11 by traversaro