robotology/robotology-superbuild #1291: Mid-October Unstable Branches CI failure

traversaro, 2022-10-13:
See https://github.com/robotology/robotology-superbuild/actions/runs/3234982258/jobs/5298788047 :
~~~
2022-10-13T03:32:06.2628796Z [163/477] Building CXX object src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParser.cpp.o
2022-10-13T03:32:06.2629258Z FAILED: src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParser.cpp.o 
2022-10-13T03:32:06.2635487Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DEMBOBJ_DLL -DEMBOBJ_dontuseexternalincludes -DEOPROT_CFG_OVERRIDE_CALLBACKS_IN_RUNTIME -D__ACE_INLINE__ -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/../motionControlLib -I/embObjAnalog/usrcbk -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/../skinLib -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/core/core -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/core/exec/yarp -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/icub -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/opcprot -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/protocol/api -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/protocol/cfg -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/protocol/src -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/comm-v2/transport -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/embobj/plus/utils -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/canProtocolLib -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParser.cpp.o -MF src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParser.cpp.o.d -o src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParser.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp
2022-10-13T03:32:06.2639942Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp: In member function 'bool ServiceParser::parseService(yarp::os::Searchable&, servConfigPOS_t&)':
2022-10-13T03:32:06.2640895Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1771:25: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2641703Z  1771 |     posBoardConfig_ptr->version.firmware.major = asServBoardInfo_ptr->firmware.major;
2022-10-13T03:32:06.2642026Z       |                         ^~~~~~~
2022-10-13T03:32:06.2642710Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1772:25: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2643385Z  1772 |     posBoardConfig_ptr->version.firmware.minor = asServBoardInfo_ptr->firmware.minor;
2022-10-13T03:32:06.2643718Z       |                         ^~~~~~~
2022-10-13T03:32:06.2644370Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1773:25: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2645053Z  1773 |     posBoardConfig_ptr->version.firmware.build = asServBoardInfo_ptr->firmware.build;
2022-10-13T03:32:06.2645395Z       |                         ^~~~~~~
2022-10-13T03:32:06.2646078Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1774:25: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2646734Z  1774 |     posBoardConfig_ptr->version.protocol.major = asServBoardInfo_ptr->protocol.major;
2022-10-13T03:32:06.2647071Z       |                         ^~~~~~~
2022-10-13T03:32:06.2647753Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1775:25: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2648431Z  1775 |     posBoardConfig_ptr->version.protocol.minor = asServBoardInfo_ptr->protocol.minor;
2022-10-13T03:32:06.2648748Z       |                         ^~~~~~~
2022-10-13T03:32:06.2649426Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1788:29: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'boardInfo'
2022-10-13T03:32:06.2650087Z  1788 |         posBoardConfig_ptr->boardInfo.canloc[i].addr= sensor.location.can.addr;
2022-10-13T03:32:06.2650417Z       |                             ^~~~~~~~~
2022-10-13T03:32:06.2651072Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:1789:29: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'boardInfo'
2022-10-13T03:32:06.2651742Z  1789 |         posBoardConfig_ptr->boardInfo.canloc[i].port= sensor.location.can.port;
2022-10-13T03:32:06.2652080Z       |                             ^~~~~~~~~
2022-10-13T03:32:06.2652883Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp: In member function 'bool ServiceParser::parseService(yarp::os::Searchable&, servConfigMC_t&)':
2022-10-13T03:32:06.2653822Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:3980:22: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'boardInfo'
2022-10-13T03:32:06.2654516Z  3980 |                 pos->boardInfo.canloc[i].port = mc_service.properties.poslocations[i].port;
2022-10-13T03:32:06.2654856Z       |                      ^~~~~~~~~
2022-10-13T03:32:06.2655517Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:3981:22: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'boardInfo'
2022-10-13T03:32:06.2656206Z  3981 |                 pos->boardInfo.canloc[i].addr = mc_service.properties.poslocations[i].addr;
2022-10-13T03:32:06.2656546Z       |                      ^~~~~~~~~
2022-10-13T03:32:06.2657230Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:3982:22: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'boardInfo'
2022-10-13T03:32:06.2658051Z  3982 |                 pos->boardInfo.canloc[i].insideindex = mc_service.properties.poslocations[i].insideindex;
2022-10-13T03:32:06.2658410Z       |                      ^~~~~~~~~
2022-10-13T03:32:06.2659088Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:3986:18: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2659790Z  3986 |             pos->version.firmware.major = mc_service.properties.canboards.at(0).firmware.major;
2022-10-13T03:32:06.2660245Z       |                  ^~~~~~~
2022-10-13T03:32:06.2660924Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:3987:18: error: 'struct eOmn_serv_config_data_as_pos_t' has  [... cut ...]
2022-10-13T03:32:06.2679772Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/serviceParser.cpp:4041:18: error: 'struct eOmn_serv_config_data_as_pos_t' has no member named 'version'
2022-10-13T03:32:06.2680450Z  4041 |             pos->version.protocol.minor = mc_service.properties.canboards.at(0).protocol.minor;
2022-10-13T03:32:06.2680793Z       |                  ^~~~~~~
2022-10-13T03:32:06.2681182Z [164/477] Building CXX object src/libraries/icubmod/embObjLib/CMakeFiles/ethResources.dir/serviceParserMultipleFt.cpp.o
~~~

Probably it is a icub-firmware-shared/ICUB incompatibility?

traversaro, 2022-10-13:
Probably it is due to the fact that https://github.com/robotology/icub-firmware-shared/pull/71 was merged and https://github.com/robotology/icub-main/pull/835 was not merged.

pattacini, 2022-10-13:
PR merged ✅
This CI will confirm if everything works ok:
- https://github.com/robotology/icub-main/actions/runs/3241023263

pattacini, 2022-10-13:
The CI is fine @traversaro 👍🏻

traversaro, 2022-10-13:
Fixed, see for example https://github.com/robotology/robotology-superbuild/actions/runs/3240981421/jobs/5312309548 . Thanks @pattacini @marcoaccame !

closed 2022-10-26 by traversaro