robotology/robotology-superbuild #1303: CI Failure Unstable all OS failure Mid November 2022

traversaro, 2022-11-15:
@davidetome reported the failure in Unstable:

[image]

It seems that icub-main was using the `PreciselyTimed.h` header that was deprecated in YARP 3.3.0 (see https://github.com/robotology/yarp/blob/v3.7.2/doc/release/v3_3_0.md?plain=1#L76) and that was removed today in https://github.com/robotology/yarp/pull/2893 . However, I also checked how it was possible that icub-main was compiled for so long with that deprecation warning, and it seems that no deprecation warning was printed when `PreciselyTimed.h` was used (log from https://github.com/robotology/robotology-superbuild/actions/runs/3466927105/jobs/5791289549):
~~~
2022-11-15T03:23:13.3765496Z [65/477] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/ThreadTable2.cpp.o
2022-11-15T03:23:13.3765973Z [66/477] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/ThreadPool2.cpp.o
2022-11-15T03:23:13.3766478Z [67/477] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/yarp_plugin_icubmod_canmotioncontrol.cpp.o
2022-11-15T03:23:13.3766977Z [68/477] Building CXX object src/libraries/actionPrimitives/CMakeFiles/actionPrimitives.dir/src/actionPrimitives.cpp.o
2022-11-15T03:23:13.3767347Z [69/477] Linking CXX static library libactionPrimitives.a
2022-11-15T03:23:13.3767760Z [70/477] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/SmithPredictor.cpp.o
2022-11-15T03:23:13.3768329Z [71/477] Building CXX object src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerserver.dir/yarp_plugin_icubmod_cartesiancontrollerserver.cpp.o
2022-11-15T03:23:13.3768926Z [72/477] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o
2022-11-15T03:23:13.3769314Z [73/477] Linking CXX shared module lib/iCub/canmotioncontrol.so
~~~

So for some reason the `YARP_COMPILER_WARNING` (https://github.com/robotology/yarp/blob/v3.7.2/src/libYARP_dev/src/yarp/dev/PreciselyTimed.h#L11) from YARP is not working as expected (at least in icub-main) .

fyi @randaz81 @pattacini

traversaro, 2022-11-15:
CI verified the fail: https://github.com/robotology/robotology-superbuild/actions/runs/3470474963/jobs/5798719702 .

randaz81, 2022-11-15:
> So for some reason the `YARP_COMPILER_WARNING` (https://github.com/robotology/yarp/blob/v3.7.2/src/libYARP_dev/src/yarp/dev/PreciselyTimed.h#L11) from YARP is not working as expected (at least in icub-main) .

This is very interesting, I'll check it

pattacini, 2022-11-15:
https://github.com/robotology/icub-main/pull/841 is now merged and fixed the CI failures.

davidetome, 2022-11-17:
I still have problems during compilation : 

[image]

I'm not able to re-open this issue.

cc @traversaro @randaz81 @pattacini

pattacini, 2022-11-17:
Open a PR:
- https://github.com/robotology/yarp-device-xsensmt/pull/36

randaz81, 2022-11-17:
In preparation of the next release yarp 3.8, yarp master branch may experience similar issues in the upcoming weeks due to general cleanups.

traversaro, 2022-11-17:
I triggered a new CI run to actually check if the issue is solved: https://github.com/robotology/robotology-superbuild/actions/runs/3486938556 .

traversaro, 2022-11-17:
There are still failures: https://github.com/robotology/robotology-superbuild/actions/runs/3486938556 .

traversaro, 2022-11-17:
You can notice by the linked PRs that to fully fix this for all the components will take some time.

closed 2022-12-12 by traversaro