robotology/robotology-superbuild #1311: CI Stable branches all configuration failures end of November 2022

traversaro, 2022-11-28:
See https://github.com/robotology/robotology-superbuild/actions/runs/3556352694 .

The error is:
~~~
2022-11-28T03:10:43.2952890Z CMake Error at conf/iCubFindDependencies.cmake:71 (message):
2022-11-28T03:10:43.2953429Z   An old version of icub-firmware-shared has been detected: at least 1.28.2
2022-11-28T03:10:43.2953765Z   is required
~~~

traversaro, 2022-11-28:
fyi @pattacini @Nicogene

pattacini, 2022-11-28:
Yep, I'm on it.
Just need to align master and devel of icub-firmware-shared.

cc @marcoaccame @sgiraz

pattacini, 2022-11-28:
>Just need to align master and devel of icub-firmware-shared.

Done by @marcoaccame 👍🏻 
`icub-main` CI just failed on Windows for an unrelated problem: see https://github.com/robotology/icub-main/actions/runs/3563300154/jobs/5985928361.

pattacini, 2022-11-28:
It should be fine now: see https://github.com/robotology/icub-main/actions/runs/3563300154/jobs/5987007551.

@traversaro you may want to give this CI another whirl.

traversaro, 2022-12-06:
Fixed, see https://github.com/robotology/robotology-superbuild/actions/runs/3570441493 .

closed 2022-12-06 by traversaro