robotology/robotology-superbuild #1313: CI Stable/Unstable branch failure on Ubuntu 18.04 and conda/Linux

traversaro, 2022-12-06:
Ubuntu 18.04 :
~~~
2022-12-06T02:37:29.9010699Z [24/474] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o
2022-12-06T02:37:29.9011039Z FAILED: src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o 
2022-12-06T02:37:29.9012647Z /usr/bin/c++ -DHAVE_CSTDDEF -D_USE_MATH_DEFINES -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/usr/include/coin -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include -g -fPIC -std=c++1z -MD -MT src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o -MF src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o.d -o src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o -c /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp
2022-12-06T02:37:29.9013896Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:13: warning: In the GNU C Library, "major" is defined
2022-12-06T02:37:29.9014314Z  by <sys/sysmacros.h>. For historical compatibility, it is
2022-12-06T02:37:29.9014612Z  currently defined by <sys/types.h> as well, but we plan to
2022-12-06T02:37:29.9014914Z  remove this soon. To use "major", include <sys/sysmacros.h>
2022-12-06T02:37:29.9015297Z  directly. If you did not intend to use a system-defined macro
2022-12-06T02:37:29.9015601Z  "major", you should undefine it after including <sys/types.h>.
2022-12-06T02:37:29.9015936Z  iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:37:29.9016287Z              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                             
2022-12-06T02:37:29.9016867Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:13: warning: In the GNU C Library, "minor" is defined
2022-12-06T02:37:29.9017447Z  by <sys/sysmacros.h>. For historical compatibility, it is
2022-12-06T02:37:29.9017741Z  currently defined by <sys/types.h> as well, but we plan to
2022-12-06T02:37:29.9018045Z  remove this soon. To use "minor", include <sys/sysmacros.h>
2022-12-06T02:37:29.9018419Z  directly. If you did not intend to use a system-defined macro
2022-12-06T02:37:29.9018715Z  "minor", you should undefine it after including <sys/types.h>.
2022-12-06T02:37:29.9019012Z  iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:37:29.9019442Z              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                             
2022-12-06T02:37:29.9019871Z In file included from /usr/include/x86_64-linux-gnu/sys/types.h:205:0,
2022-12-06T02:37:29.9020152Z                  from /usr/include/stdlib.h:394,
2022-12-06T02:37:29.9020393Z                  from /usr/include/c++/7/cstdlib:75,
2022-12-06T02:37:29.9020848Z                  from /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:11:
2022-12-06T02:37:29.9021577Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp: In constructor 'iCub::iKin::iKinLimbVersion::iKinLimbVersion()':
2022-12-06T02:37:29.9022374Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:38: error: class 'iCub::iKin::iKinLimbVersion' does not have any field named 'gnu_dev_major'
2022-12-06T02:37:29.9022826Z  iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:37:29.9023066Z                                       ^
2022-12-06T02:37:29.9023691Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:48: error: class 'iCub::iKin::iKinLimbVersion' does not have any field named 'gnu_dev_minor'
2022-12-06T02:37:29.9024131Z  iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:37:29.9024379Z                                                 ^
2022-12-06T02:37:29.9024691Z [25/474] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/skinPart.cpp.o
~~~

conda/Linux:
~~~
2022-12-06T02:55:15.6710208Z FAILED: src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o 
2022-12-06T02:55:15.6712358Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -D_USE_MATH_DEFINES -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/usr/share/miniconda3/envs/test/include/coin-or -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o -MF src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o.d -o src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp
2022-12-06T02:55:15.6713953Z In file included from /usr/share/miniconda3/envs/test/x86_64-conda-linux-gnu/sysroot/usr/include/sys/types.h:222,
2022-12-06T02:55:15.6714462Z                  from /usr/share/miniconda3/envs/test/x86_64-conda-linux-gnu/sysroot/usr/include/stdlib.h:314,
2022-12-06T02:55:15.6715254Z                  from /usr/share/miniconda3/envs/test/x86_64-conda-linux-gnu/include/c++/10.4.0/cstdlib:75,
2022-12-06T02:55:15.6715876Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:11:
2022-12-06T02:55:15.6716766Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp: In constructor 'iCub::iKin::iKinLimbVersion::iKinLimbVersion()':
2022-12-06T02:55:15.6717681Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:38: error: class 'iCub::iKin::iKinLimbVersion' does not have any field named 'gnu_dev_major'
2022-12-06T02:55:15.6718219Z  1546 | iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:55:15.6718540Z       |                                      ^~~~~
2022-12-06T02:55:15.6719207Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/src/iKinFwd.cpp:1546:48: error: class 'iCub::iKin::iKinLimbVersion' does not have any field named 'gnu_dev_minor'
2022-12-06T02:55:15.6719797Z  1546 | iKinLimbVersion::iKinLimbVersion() : major(0), minor(0)
2022-12-06T02:55:15.6720110Z       |                                                ^~~~~
2022-12-06T02:55:15.6720476Z [26/446] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinHlp.cpp.o
~~~

traversaro, 2022-12-06:
See: 
* https://stackoverflow.com/questions/20024722/c-error-constructor-parameter#comment29820491_20024722
* https://bugzilla.redhat.com/show_bug.cgi?id=130601
* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=17130

traversaro, 2022-12-06:
I reported just the gcc version in the original comment, but probably it plays a role also the glibc version.

pattacini, 2022-12-06:
Possible patch proposed in https://github.com/robotology/icub-main/pull/847.

pattacini, 2022-12-06:
PR merged ✅