robotology/robotology-superbuild #1332: Documentation – Problems with setting `DISPLAY` on WSL2

pattacini, 2023-01-25:
The documentation for WSL2 says that `DISPLAY` should be set pointing to the host as follows:

https://github.com/robotology/robotology-superbuild/blob/a7cdb9e15ce0525c0b7e61837a59e410836df144/README.md?plain=1#L174

However, applications such as `yarpview` and `gazebo` won't run correctly this way on my WSL2 system.
I didn't seem to catch error messages though.

By contrast, leaving the env var `DISPLAY` to its original value (i.e., `:0`) makes the apps work again.

@traversaro, did you happen to come across the same behavior?

cc @martinaxgloria @davidelasagna

traversaro, 2023-01-25:
Thanks for the feedback @pattacini ! This is kind of related to https://github.com/robotology/robotology-superbuild/issues/1190 .

pattacini, 2023-01-25:
@traversaro, you can close this as a duplicate then 👍🏻

traversaro, 2023-01-25:
Ok! Let's continue the discussion in https://github.com/robotology/robotology-superbuild/issues/1190 .

closed 2023-01-25 by traversaro