robotology/robotology-superbuild #1338: Compilation failure on clean Windows 11 installation

S-Dafarra, 2023-02-06:
I just tried to install the latest release (https://github.com/robotology/robotology-superbuild/tree/releases/2022.11) on Windows 11 using Visual Studio 2022. The machine is new and I tried to perform a first installation. 

It failed on ``yarp`` with the errors
```
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\1006a7d5eb8eeb0ddfe54312318c55a3\GoalID.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\1006a7d5eb8eeb0ddfe54312318c55a3\GoalStatus.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\1006a7d5eb8eeb0ddfe54312318c55a3\GoalStatusArray.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_actionlib_msgs.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_actionlib_msgs.vcxproj]
15>  sx_instantiator.cpp
14>  Generating code from ../../../extern/ros/common_msgs/diagnostic_msgs/srv/AddDiagnostics.srv
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\d8e7f1970e3bce5be8ffb743f983bcdc\AddDiagnostics.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\d8e7f1970e3bce5be8ffb743f983bcdc\DiagnosticArray.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\d8e7f1970e3bce5be8ffb743f983bcdc\DiagnosticStatus.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\d8e7f1970e3bce5be8ffb743f983bcdc\KeyValue.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\d8e7f1970e3bce5be8ffb743f983bcdc\SelfTest.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_diagnostic_msgs.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_diagnostic_msgs.vcxproj]
14>  Generating code from ../../../extern/ros/common_msgs/geometry_msgs/msg/Accel.msg
15>C:\robotology-superbuild\src\casadi\casadi\core\exception.hpp(105,37): warning C4996: 'localtime': This function or variable may be unsafe. Consider using localtime_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\robotology-superbuild\build\src\casadi\casadi\core\casadi.vcxproj]
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Accel.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\AccelStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\AccelWithCovariance.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\AccelWithCovarianceStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Inertia.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\InertiaStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Point.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Point32.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PointStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Polygon.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PolygonStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Pose.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Pose2D.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PoseArray.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PoseStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PoseWithCovariance.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\PoseWithCovarianceStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Quaternion.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\QuaternionStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Transform.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\TransformStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Twist.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\TwistStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\TwistWithCovariance.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\TwistWithCovarianceStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Vector3.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Vector3Stamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\Wrench.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\70615ed70e9edbf9b90ce31789e4bf13\WrenchStamped.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_geometry_msgs.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_geometry_msgs.vcxproj]
14>  Generating code from duration
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\f58e8139a7dc4e3075c17bf172bb6312\TickDuration.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\f58e8139a7dc4e3075c17bf172bb6312\TickTime.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_native.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_native.vcxproj]
14>  Generating code from ../../../extern/ros/common_msgs/nav_msgs/srv/GetMap.srv
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\GetMap.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\GetPlan.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\GridCells.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\MapMetaData.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\OccupancyGrid.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\Odometry.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\Path.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\85f621e1bc953a08891d76388c5dc461\SetMap.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_nav_msgs.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_nav_msgs.vcxproj]
14>  Generating code from ../../../extern/ros/common_msgs/sensor_msgs/msg/BatteryState.msg
14>C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): error MSB8066: Custom build for 'C:\robotology-superbuild\build\src\YARP\CMakeFiles\9b1758c1d033caf03922a85ef4bcebe5\BatterySt [... cut ...] s\fdf6724d72c38174c8d782772dd24e6e\MenuEntry.h.rule;C:\robotology-superbuild\build\src\YARP\CMakeFiles\bc16a037eee8ab7274a3455315243782\YARP_rosmsg_gen_visualization_msgs.rule;C:\robotology-superbuild\src\YARP\src\libYARP_rosmsg\src\CMakeLists.txt' exited with code -1073741515. [C:\robotology-superbuild\build\src\YARP\src\libYARP_rosmsg\src\YARP_rosmsg_gen_visualization_msgs.vcxproj]
14>  Building Custom Rule C:/robotology-superbuild/src/YARP/src/libYARP_sig/src/CMakeLists.txt
14>  Image.cpp
```

S-Dafarra, 2023-02-06:
Interestingly, I tried to compile ``YARP`` alone afterwards, but there were no issues.

S-Dafarra, 2023-02-06:
Recompiling the superbuild seemed to fix the issue

traversaro, 2023-02-14:
Ok, thanks @S-Dafarra for reporting the issue. Let's close this until we find a way to reproduce the problem.

closed 2023-02-14 by traversaro