robotology/robotology-superbuild #1341: Yarp c# Binding  in superbuild

omareldardear, 2023-02-10:
Hello Everyone, 

We were trying to compile the c# binding on Linux, 
But there was no cmake flag for the c# , I only find python , Matlab, and Octave. But not Python.  

Would you please consider adding a flag for the c# ? 

This issue is connected to the open issue in Yarp as we are trying to have csharp binding work in both windows and linux. 
https://github.com/robotology/yarp/issues/2937

traversaro, 2023-02-11:
Hello @omareldardear ! We can look into adding such an option to the superbuild if it is used in a consistent way. If this is just for a one time experiment, you can also just enable it in YARP build directory: 

~~~
cd robotology-superbuild/build/src/YARP
# enable the option you want to enable via ccmake . , then generate by pressing c several times, and then g
ccmake . 
# now the option is enabled, so unless the superbuild is overwriting it, you can go back and compile the superbuils as usual
cd ../..
cmake --build . --config Release
~~~

traversaro, 2023-02-14:
Hello @omareldardear, did you checked if the suggested workaround is working for your objectives?

omareldardear, 2023-02-14:
CREATE_CSHARP 
Which will appear after enabling  YARP_COMPILE_BINDINGS

It is the same as Python or Octave (CREATE_PYTHON , CREATE_OCTAVE)

traversaro, 2023-02-16:
Further improvements can be discussed in https://github.com/robotology/robotology-superbuild/issues/1347 .

closed 2023-02-16 by traversaro