robotology/robotology-superbuild #1350: Get rid of manually compiled qhull?

traversaro, 2023-02-20:
Back in https://github.com/robotology/robotology-superbuild/issues/517 we added support for compiling qhull as part of the superbuild. As `qhull`  is also installed via apt, we should see if it is possible to get rid of it and just use the qhull installed by apt, for better compatibility with other apt-provided packages that uses qhull (such as PCL or MoveIt!).

See also https://github.com/robotology/robotology-superbuild/pull/1275, that basically implemented this for all the non-apt platforms.

cc @GiulioRomualdi

traversaro, 2023-02-21:
On Ubuntu 20.04 the blf build is failing with:
~~~
2023-02-20T17:11:00.8898742Z CMake Error at cmake/BipedalLocomotionFrameworkFindDependencies.cmake:58 (find_package):
2023-02-20T17:11:00.8899236Z   By not providing "FindQhull.cmake" in CMAKE_MODULE_PATH this project has
2023-02-20T17:11:00.8899681Z   asked CMake to find a package configuration file provided by "Qhull", but
2023-02-20T17:11:00.8900036Z   CMake did not find one.
2023-02-20T17:11:00.8900195Z 
2023-02-20T17:11:00.8900408Z   Could not find a package configuration file provided by "Qhull" (requested
2023-02-20T17:11:00.8900729Z   version 8.0.0) with any of the following names:
2023-02-20T17:11:00.8900889Z 
2023-02-20T17:11:00.8900980Z     QhullConfig.cmake
2023-02-20T17:11:00.8901234Z     qhull-config.cmake
2023-02-20T17:11:00.8901369Z 
2023-02-20T17:11:00.8901529Z   Add the installation prefix of "Qhull" to CMAKE_PREFIX_PATH or set
2023-02-20T17:11:00.8901873Z   "Qhull_DIR" to a directory containing one of the above files.  If "Qhull"
2023-02-20T17:11:00.8902218Z   provides a separate development package or SDK, be sure it has been
2023-02-20T17:11:00.8902471Z   installed.
2023-02-20T17:11:00.8902687Z Call Stack (most recent call first):
2023-02-20T17:11:00.8903047Z   cmake/BipedalLocomotionFrameworkDependencies.cmake:32 (checkandset_dependency)
2023-02-20T17:11:00.8903474Z   CMakeLists.txt:21 (include)
2023-02-20T17:11:00.8903644Z 
2023-02-20T17:11:00.8903650Z 
2023-02-20T17:11:00.8903838Z -- Configuring incomplete, errors occurred!
~~~

I am not sure if it is just that QhullConfig.cmake is missing (and that could be fixed by providing a `FindQhull.cmake`) or even by linking the library correctly the compilation still fails.

GiulioRomualdi, 2023-02-21:
Only `.cmake` is missing. If I'm not mistaken is privately linked in blf so we can avoid installing the ``FindQhull.cmake``

traversaro, 2023-02-27:
Apparently in Ubuntu 20.04 only the qhullcpp headers are installed, but no library contained the compiled symbols, so it is impossible to use apt-provided package via qhullcpp .

traversaro, 2023-02-27:
> Apparently in Ubuntu 20.04 only the qhullcpp headers are installed, but no library contained the compiled symbols, so it is impossible to use apt-provided package via qhullcpp .

So, to minimize the possible problems, I still think that it make sense to at least solve the problem on Ubuntu 22.04 and future version. To do so, I modified https://github.com/robotology/robotology-superbuild/pull/1351 to:
* Always install `libqhull-dev` via apt
* Only build qhull on Ubuntu 20.04 (Focal) 

Note that this may be problematic on 20.04 as both qhull from apt and qhull from superbuild is installed. However, this is already the case in CI and in the laptop of most developers, as libqhull-dev is a dependency of libpcl-dev . So, the PR does not make the situation worse on 20.04, and instead permit  to avoid building qhull on 22.04, so I think it is a net gain.