robotology/robotology-superbuild #1366: CI Failure in conda-forge and Homebrew with CMake 3.26 March 2023

traversaro, 2023-03-16:
CMake 3.26 was recently released, and it seems there was a related regression in the configuration of `ICUBContrib` :

~~~
2023-03-16T03:06:01.1914833Z [289/424] Performing configure step for 'ICUBcontrib'
2023-03-16T03:06:01.1915375Z FAILED: src/ICUBcontrib/CMakeFiles/YCMStamp/ICUBcontrib-configure /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUBcontrib/CMakeFiles/YCMStamp/ICUBcontrib-configure 
2023-03-16T03:06:01.1917514Z cd /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUBcontrib && /usr/share/miniconda3/envs/test/bin/cmake --no-warn-unused-cli "-DCMAKE_PREFIX_PATH:PATH=/home/runner/work/robotology-superbuild/robotology-superbuild/build/install;/usr/share/miniconda3/envs/test;/usr/share/miniconda3/envs/test/x86_64-conda-linux-gnu/sysroot/usr" -DMatlab_ROOT_DIR:PATH=/home/runner/work/robotology-superbuild/robotology-superbuild/msdk_R2020b_mexa64 -DMatlab_MEX_EXTENSION:STRING=mexa64 -C/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUBcontrib/CMakeFiles/YCMTmp/ICUBcontrib-cache-Release.cmake -GNinja /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUBcontrib && /usr/share/miniconda3/envs/test/bin/cmake -E touch /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUBcontrib/CMakeFiles/YCMStamp/ICUBcontrib-configure
2023-03-16T03:06:01.1917768Z Not searching for unused variables given on the command line.
2023-03-16T03:06:01.1918229Z loading initial cache file /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUBcontrib/CMakeFiles/YCMTmp/ICUBcontrib-cache-Release.cmake
2023-03-16T03:06:01.1918363Z CMake Error at CMakeLists.txt:8 (find_package):
2023-03-16T03:06:01.1918535Z   Could not find a configuration file for package "YARP" that is compatible
2023-03-16T03:06:01.1918722Z   with requested version "3.3.2".
2023-03-16T03:06:01.1918733Z 
2023-03-16T03:06:01.1919077Z   The following configuration files were considered but not accepted:
2023-03-16T03:06:01.1919084Z 
2023-03-16T03:06:01.1919674Z     /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake, version: 3.8.0 (64bit)
2023-03-16T03:06:01.1919686Z 
2023-03-16T03:06:01.1919692Z 
2023-03-16T03:06:01.1919697Z 
2023-03-16T03:06:01.1920068Z -- Configuring incomplete, errors occurred!
~~~

Everything was working fine with CMake 3.25.* .

traversaro, 2023-03-16:
The difference between `YARPConfigVersion.cmake` generated with 3.25 (working) and `YARPConfigVersion.cmake` generated with 3.26 (not working) is:
~~~patch
--- <unnamed>
+++ <unnamed>
@@ -52,8 +52,13 @@
 endif()
 
 
-# if the installed project requested no architecture check, don't perform the check
-if("FALSE")
-  return()
-endif()
-
 # if the installed or the using project don't have CMAKE_SIZEOF_VOID_P set, ignore it:
-if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "" OR "8" STREQUAL "")
+if(CMAKE_SIZEOF_VOID_P STREQUAL "" OR "8" STREQUAL "")
   return()
 endif()
~~~

The problem in particular is the last diff line. Basically if a noarch C++ packages consumes a 64-bit C++ package, `CMAKE_SIZEOF_VOID_P` is an empty string, and the flow should enter in the `if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "" OR "8" STREQUAL "")` . 
The problem is that in CMake language (apparently) if `CMAKE_SIZEOF_VOID_P` is an empty string, `"${CMAKE_SIZEOF_VOID_P}" STREQUAL ""`, while `CMAKE_SIZEOF_VOID_P STREQUAL ""` appers to be false. I am not sure if `CMAKE_SIZEOF_VOID_P STREQUAL ""` being false is expect to happen, anyhow this seems to be an upstream CMake 3.26 problem.

traversaro, 2023-03-16:
The regression seems to be caused by https://gitlab.kitware.com/cmake/cmake/-/merge_requests/8133 . As this requires an upstream fix, it may take some time. Probably we can just pin CMake to <= 3.25 in conda-forge, while in homebrew we can't do that.

traversaro, 2023-03-16:
I opened an upstream issue in CMake: https://gitlab.kitware.com/cmake/cmake/-/issues/24608 .

traversaro, 2023-03-24:
CMake 3.26.0 was released yesterday, so now the issue is fixed in Homebrew! In a funny (for a given definition of funny) coincidence, exactly the same data the GitHub Actions Ubuntu images have been updated to use CMake 3.26.0, so now we have the problem in the `ubuntu-20.04` image. The same problem occurs also in other projects now, see https://github.com/robotology/assistive-rehab/actions/runs/4507102467/jobs/7934556272 .

cc @pattacini @Nicogene

traversaro, 2023-03-24:
I opened an issue upstream in https://github.com/actions/runner-images/issues/7336, but I guess a few weeks may be necessary to update GitHub Actions images. The possible workaround/alternatives in the meanwhile are:
* Remove CMake installed via this script https://github.com/actions/runner-images/blob/0f6480a303676a2b8fe9b53b160112c1293fe699/images/linux/scripts/installers/cmake.sh#L4 (not sure how you do it, probably a lot of manual `rm` calls)
* Switch to use a docker image, see https://github.com/robotology/robotology-superbuild/blob/master/.github/workflows/ci.yml#L185 . This will permit to use a vanilla Ubuntu 20.04 image, with no new CMake installed.

traversaro, 2023-03-24:
> Is `ubuntu-latest` still `20.04` on GH workspaces?

No, but the problem is the same on 22.04 . I was just discussing 20.04 as in the robotology-superbuild build we are using the GitHub Actions images only for 20.04, for 22.04 we are just using the Docker images.

pattacini, 2023-03-24:
On second thought, given that the affected repos are only `robots-configuration` and `assistive-rehab`, I would simply disable the CI rather than implement any workaround.

traversaro, 2023-03-24:
> * Remove CMake installed via this script https://github.com/actions/runner-images/blob/0f6480a303676a2b8fe9b53b160112c1293fe699/images/linux/scripts/installers/cmake.sh#L4 (not sure how you do it, probably a lot of manual `rm` calls)

In https://github.com/actions/runner-images/issues/7336#issuecomment-1482778271 they provided a much better workaround, i.e. overwriting the existing cmake with a newer one:
~~~
curl -sL https://github.com/Kitware/CMake/releases/download/v3.26.1/cmake-3.26.1-linux-x86_64.sh -o cmakeinstall.sh \
&& chmod +x cmakeinstall.sh \
&& ./cmakeinstall.sh --prefix=/usr/local --exclude-subdir \
&& rm cmakeinstall.sh
~~~

traversaro, 2023-04-02:
> This should be fixed by [actions/runner-images#7341](https://github.com/actions/runner-images/pull/7341) .

The images have been updated. However, probably there is a new regression in Ubuntu 20.04 in blf due to CMake 3.26.1, that was indeed already detected in https://github.com/ami-iit/bipedal-locomotion-framework/pull/635#issuecomment-1485020874 : 

~~~
2023-04-02T03:25:03.8400239Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include/BipedalLocomotion/ParametersHandler/IParametersHandler.h:16,
2023-04-02T03:25:03.8401374Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include/BipedalLocomotion/System/LinearTask.h:18,
2023-04-02T03:25:03.8402335Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/System/src/LinearTask.cpp:13:
2023-04-02T03:25:03.8403575Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:212:43: error: expected template-name before ‘<’ token
2023-04-02T03:25:03.8404446Z   212 | struct is_string : public std::disjunction<std::is_same<char*, typename std::decay<T>::type>,
2023-04-02T03:25:03.8404832Z       |                                           ^
2023-04-02T03:25:03.8405762Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:212:43: error: expected ‘{’ before ‘<’ token
2023-04-02T03:25:03.8407059Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:224:16: error: ‘string_view’ in namespace ‘std’ does not name a type
2023-04-02T03:25:03.8407713Z   224 | constexpr std::string_view
2023-04-02T03:25:03.8407991Z       |                ^~~~~~~~~~~
2023-04-02T03:25:03.8409086Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include/BipedalLocomotion/GenericContainer/TemplateHelpers.h:224:11: note: ‘std::string_view’ is only available from C++17 onwards
2023-04-02T03:25:03.8409746Z   224 | constexpr std::string_view
~~~

traversaro, 2023-04-02:
The problem is that someone is explicitly passing `-std=c++14` option:
~~~
usr/include/eigen3 -O3 -DNDEBUG -std=c++17 -fPIC -fvisibility=hidden -std=c++14 -flto -fno-fat-lto-objects -MD -MT bindings/python/CMakeFiles/pybind11_blf.dir/System/src/Advanceable.cpp.o -MF bindings/python/CMakeFiles/pybind11_blf.dir/System/src/Advanceable.cpp.o.d -o bindings/python/CMakeFiles/pybind11_blf.dir/System/src/Advanceable.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/bindings/python/System/src/Advanceable.cpp
~~~

traversaro, 2023-04-02:
> The problem is that someone is explicitly passing `-std=c++14` option:
>
> ```
> usr/include/eigen3 -O3 -DNDEBUG -std=c++17 -fPIC -fvisibility=hidden -std=c++14 -flto -fno-fat-lto-objects -MD -MT [...]

The version of pybind11 is passing the `-std=c++14` option explicitly, and for some reason with CMake 3.26.* now the `-std=c++14` takes the precendence over `-std=c++17`.

traversaro, 2023-04-02:
> The version of pybind11 is passing the `-std=c++14` option explicitly, and for some reason with CMake 3.26.* now the `-std=c++14` takes the precendence over `-std=c++17`.

A fix/workaround for this issue is provided in https://github.com/ami-iit/bipedal-locomotion-framework/pull/643 .

traversaro, 2023-04-06:
We still have some failures in `LatestReleases`, as we do not have a release of blf that contains https://github.com/ami-iit/bipedal-locomotion-framework/pull/643 . Anyhow, this is not a big problem.

closed 2023-04-26 by traversaro