robotology/robotology-superbuild #1375: conda-forge CI on macOS regression end of March 2023

traversaro, 2023-03-29:
Today several subprojects of the superbuild started experiencing failures such as:
~~~
2023-03-29T13:21:42.8417150Z /Users/runner/miniconda3/envs/test/bin/x86_64-apple-darwin13.4.0-clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DLIBBULLET_VERSION=3.20 -DLIBBULLET_VERSION_GT_282 -DSPDLOG_FMT_EXTERNAL -Dgazebo_yarp_singleton_EXPORTS -I/Users/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/libraries/singleton/include/GazeboYarpPlugins -I/Users/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/libraries/singleton/include -isystem /Users/runner/miniconda3/envs/test/include/gazebo-11 -isystem /Users/runner/miniconda3/envs/test/include/bullet -isystem /Users/runner/miniconda3/envs/test/include/simbody -isystem /Users/runner/miniconda3/envs/test/include/sdformat-9.8 -isystem /Users/runner/miniconda3/envs/test/include/ignition/math6 -isystem /Users/runner/miniconda3/envs/test/include/OGRE -isystem /Users/runner/miniconda3/envs/test/include/OGRE/OSX -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Bites -isystem /Users/runner/miniconda3/envs/test/include/SDL2 -isystem /Users/runner/miniconda3/envs/test/include/OGRE/HLMS -isystem /Users/runner/miniconda3/envs/test/include/OGRE/MeshLodGenerator -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Overlay -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Paging -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Property -isystem /Users/runner/miniconda3/envs/test/include/OGRE/RTShaderSystem -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Terrain -isystem /Users/runner/miniconda3/envs/test/include/OGRE/Volume -isystem /Users/runner/miniconda3/envs/test/include/ignition/transport8 -isystem /Users/runner/miniconda3/envs/test/include/ignition/msgs5 -isystem /Users/runner/miniconda3/envs/test/include/ignition/common3 -isystem /Users/runner/miniconda3/envs/test/include/ignition/fuel_tools4 -isystem /Users/runner/miniconda3/envs/test/include/eigen3 -isystem /Users/runner/miniconda3/envs/test/include/sdformat-9.8/sdf/.. -isystem /Users/runner/miniconda3/envs/test/include/ignition/cmake2 -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -O2 -pipe -stdlib=libc++ -fvisibility-inlines-hidden -fmessage-length=0 -isystem /Users/runner/miniconda3/envs/test/include -O3 -DNDEBUG -std=gnu++17 -isysroot /Applications/Xcode_14.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk -fPIC -mfpmath=sse -msse -msse2 -msse3 -mssse3 -MD -MT libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o -MF libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o.d -o libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o -c /Users/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/libraries/singleton/src/Handler.cc
2023-03-29T13:21:42.8523790Z In file included from /Users/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/libraries/singleton/src/Handler.cc:7:
2023-03-29T13:21:42.8626500Z In file included from /Users/runner/work/gazebo-yarp-plugins/gazebo-yarp-plugins/libraries/singleton/include/GazeboYarpPlugins/Handler.hh:43:
2023-03-29T13:21:42.8728580Z In file included from /Users/runner/miniconda3/envs/test/include/gazebo-11/gazebo/common/Events.hh:23:
2023-03-29T13:21:42.8829980Z In file included from /Users/runner/miniconda3/envs/test/include/gazebo-11/gazebo/common/Console.hh:27:
2023-03-29T13:21:42.8930570Z In file included from /Users/runner/miniconda3/envs/test/include/boost/thread.hpp:13:
2023-03-29T13:21:42.9031940Z In file included from /Users/runner/miniconda3/envs/test/include/boost/thread/thread.hpp:12:
2023-03-29T13:21:42.9132430Z In file included from /Users/runner/miniconda3/envs/test/include/boost/thread/thread_only.hpp:22:
2023-03-29T13:21:42.9232970Z In file included from /Users/runner/miniconda3/envs/test/include/boost/thread/detail/thread.hpp:41:
2023-03-29T13:21:42.9334200Z In file included from /Users/runner/miniconda3/envs/test/include/boost/functional/hash.hpp:6:
2023-03-29T13:21:42.9435130Z /Users/runner/miniconda3/envs/test/include/boost/container_hash/hash.hpp:132:33: error: no template named 'unary_function' in namespace 'std'; did you mean '__unary_function'?
2023-03-29T13:21:42.9540510Z         struct hash_base : std::unary_function<T, std::size_t> {};
2023-03-29T13:21:42.9636350Z                            ~~~~~^
2023-03-29T13:21:42.9737740Z /Users/runner/miniconda3/envs/test/bin/../include/c++/v1/__functional/unary_function.h:46:1: note: '__unary_function' declared here
2023-03-29T13:21:42.9837660Z using __unary_function = __unary_function_keep_layout_base<_Arg, _Result>;
2023-03-29T13:21:42.9938690Z ^
2023-03-29T13:21:43.0039800Z 1 error generated.
2023-03-29T13:21:44.1771450Z [13/112] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o
2023-03-29T13:21:44.1873210Z FAILED: libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o 
~~~

traversaro, 2023-03-29:
The difference between working and not working environment is:
~~~diff
--- <unnamed>
+++ <unnamed>
@@ -26,7 +26,7 @@
 ld64_osx-64                        609  hfd63004_13         conda-forge/osx-64       1MB
 libclang-cpp14                  14.0.6  default_h55ffa42_0  conda-forge/osx-64      15MB
 libcurl                         7.88.1  h6df9250_1          conda-forge/osx-64     344kB
-libcxx                          15.0.7  h71dddab_0          conda-forge/osx-64       1MB
+libcxx                          16.0.0  h71dddab_0          conda-forge/osx-64       1MB
 libedit                   3.1.20191231  h0678c8f_2          conda-forge/osx-64     105kB
 libev                             4.33  haf1e3a3_1          conda-forge/osx-64     101kB
 libgfortran                      5.0.0  11_3_0_h97931a8_31  conda-forge/osx-64     160kB
@@ -198,7 +198,7 @@
 ruby                            3.1.2  hb3742b3_1          conda-forge/osx-64       8MB
 sdl                            1.2.60  hf0c8a7f_0          conda-forge/osx-64     168kB
 sdl2                           2.26.4  hf0c8a7f_0          conda-forge/osx-64       1MB
-setuptools                     67.6.0  pyhd8ed1ab_0        conda-forge/noarch     579kB
+setuptools                     67.6.1  pyhd8ed1ab_0        conda-forge/noarch     580kB
 simbody                           3.7  h43072b6_3          conda-forge/osx-64      39MB
 snappy                         1.1.10  h225ccf5_0          conda-forge/osx-64      35kB
 soxr                            0.1.3  hbf74d83_3          conda-forge/osx-64     128kB
@@ -212,8 +212,8 @@
 tinyxml                         2.6.2  h65a07b1_2          conda-forge/osx-64      52kB
 tinyxml2                        9.0.0  he49afe7_2          conda-forge/osx-64     108kB
 tk                             8.6.12  h5dbffcc_0          conda-forge/osx-64       4MB
-tzcode                          2023b  hb7f2c08_0          conda-forge/osx-64      63kB
-tzdata                          2023b  h71feb2d_0          conda-forge/noarch     118kB
+tzcode                          2023c  hb7f2c08_0          conda-forge/osx-64      63kB
+tzdata                          2023c  h71feb2d_0          conda-forge/noarch     118kB
 urdfdom                         3.1.0  hd1da41a_1          conda-forge/osx-64     117kB
 urdfdom_headers                 1.1.0  h1b54a9f_0          conda-forge/osx-64      19kB
 wheel                          0.40.0  pyhd8ed1ab_0        conda-forge/noarch      56kB
~~~

The most relevant change seems the libcxx change, so probably there was a regression there.

traversaro, 2023-03-29:
Ok, the issue  on the boost contained_hash library seems to be this one: https://github.com/boostorg/container_hash/issues/24 .

traversaro, 2023-03-29:
Ok, backporting stuff to boost-cpp 1.78 is essentially blocked by https://github.com/conda-forge/boost-cpp-feedstock/pull/130/files . So at this point I am out of ideas, let's open a bug report on libcxx-feedstock.

traversaro, 2023-03-29:
> let's open a bug report on libcxx-feedstock.

Done in https://github.com/conda-forge/libcxx-feedstock/issues/114 .

h-vetinari, 2023-03-30:
> So at this point I am out of ideas, let's open a bug report on libcxx-feedstock.

Add a constraint to `libcxx<16`?

h-vetinari, 2023-03-30:
The release notes you dug up also mention a special symbol, which, when defined, might be able to unblock your build for the time being: `_LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY`

traversaro, 2023-04-06:
Fixed by https://github.com/conda-forge/boost-cpp-feedstock/pull/135, I think that we can close.

closed 2023-04-06 by traversaro