robotology/robotology-superbuild #1380: Early April 2023 CI Failure on Stable branches

traversaro, 2023-04-05:
Since this night stable branches are failing with error:
~~~
2023-04-05T02:48:35.7166599Z   CMake Error at conf/iCubFindDependencies.cmake:69 (message):
2023-04-05T02:48:35.7167180Z     An old version of icub-firmware-shared has been detected: at least 1.33.1
2023-04-05T02:48:35.7167695Z     is required
~~~

traversaro, 2023-04-05:
cc @Nicogene @pattacini do you have any idea on this? Unstable (i.e. `devel`) branches are not affected.

Nicogene, 2023-04-05:
It seems that icub-firmware-shared was 1.33.0 (see https://github.com/robotology/icub-firmware-shared/commit/9694c9034af1f53bdaa3a7ae3682019bbbcc64af) on master but icub-main master requires 1.33.1.

I just released icub-firmware-shared 1.34.0 so this failure should be fixed now

pattacini, 2023-04-05:
Updated the required version via https://github.com/robotology/icub-main/commit/75502fbfe98885656821f46c528bc555d872cfd6.

traversaro, 2023-04-06:
This was fixed by the linked changes, thanks!

closed 2023-04-06 by traversaro