robotology/robotology-superbuild #1388: CI Failure Mid April 2023 ROBOTOLOGY_ENABLE_DYNAMICS Stable and Unstbale branches

traversaro, 2023-04-16:
The error is:

~~~
2023-04-16T03:43:05.2637891Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -DTSID_EXPORTS -D_USE_MATH_DEFINES -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Contacts/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Math/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/ParametersHandler/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/GenericContainer/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TextLogging/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/System/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/Conversions/include -isystem /usr/share/miniconda3/envs/test/include/eigen3 -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/osqp -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -std=c++17 -fPIC -MD -MT src/TSID/CMakeFiles/TSID.dir/src/SO3Task.cpp.o -MF src/TSID/CMakeFiles/TSID.dir/src/SO3Task.cpp.o.d -o src/TSID/CMakeFiles/TSID.dir/src/SO3Task.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp
2023-04-16T03:43:05.2642783Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp: In member function 'virtual bool BipedalLocomotion::TSID::SO3Task::initialize(std::weak_ptr<const BipedalLocomotion::ParametersHandler::IParametersHandler>)':
2023-04-16T03:43:05.2644400Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp:139:29: error: no matching function for call to 'LieGroupControllers::ProportionalDerivativeController<manif::SO3<double> >::setGains(<brace-enclosed initializer list>)'
2023-04-16T03:43:05.2645150Z   139 |     m_SO3Controller.setGains({kpAngular, kdAngular});
2023-04-16T03:43:05.2645460Z       |     ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2646224Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ProportionalDerivativeController/ControllerBase.h:11,
2023-04-16T03:43:05.2647274Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ProportionalDerivativeController/Controller.h:15,
2023-04-16T03:43:05.2648223Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/ProportionalDerivativeController.h:11,
2023-04-16T03:43:05.2649241Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/include/BipedalLocomotion/TSID/SO3Task.h:17,
2023-04-16T03:43:05.2650089Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp:9:
2023-04-16T03:43:05.2651766Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ControllerBase.h:129:6: note: candidate: 'void LieGroupControllers::ControllerBase<_Derived>::setGains(Ts&& ...) [with Ts = {}; _Derived = LieGroupControllers::ProportionalDerivativeController<manif::SO3<double> >]'
2023-04-16T03:43:05.2652535Z   129 | void ControllerBase<_Derived>::setGains(Ts&&... gains)
2023-04-16T03:43:05.2652833Z       |      ^~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2653614Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ControllerBase.h:129:6: note:   candidate expects 0 arguments, 1 provided
2023-04-16T03:43:05.2654879Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp: In member function 'virtual bool BipedalLocomotion::TSID::SO3Task::update()':
2023-04-16T03:43:05.2656305Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp:160:29: error: no matching function for call to 'LieGroupControllers::ProportionalDerivativeController<manif::SO3<double> >::setState(<brace-enclosed initializer list>)'
2023-04-16T03:43:05.2656968Z   160 |     m_SO3Controller.setState(
2023-04-16T03:43:05.2657209Z       |     ~~~~~~~~~~~~~~~~~~~~~~~~^
2023-04-16T03:43:05.2657511Z   161 |         {BipedalLocomotion::Conversions::toManifRot(
2023-04-16T03:43:05.2657818Z       |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2658234Z   162 |              m_kinDyn->getWorldTransform(m_frameIndex).getRotation()),
2023-04-16T03:43:05.2658582Z       |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2659048Z   163 |          iDynTree::toEigen(m_kinDyn->getFrameVel(m_frameIndex).getAngularVec3())});
2023-04-16T03:43:05.2659415Z       |          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2660142Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ProportionalDerivativeController/ControllerBase.h:11,
2023-04-16T03:43:05.2661098Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ProportionalDerivativeController/Controller.h:15,
2023-04-16T03:43:05.2662291Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/ProportionalDerivativeController.h:11,
2023-04-16T03:43:05.2663285Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/include/BipedalLocomotion/TSID/SO3Task.h:17,
2023-04-16T03:43:05.2664147Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp:9:
2023-04-16T03:43:05.2665565Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ControllerBase.h:109:6: note: candidate: 'bool LieGroupControllers::ControllerBase<_Derived>::setState(Ts&& ...) [with Ts = {}; _Derived = LieGroupControllers::ProportionalDerivativeController<manif::SO3<double> >]'
2023-04-16T03:43:05.2666531Z   109 | bool ControllerBase<_Derived>::setState(Ts&&... state)
2023-04-16T03:43:05.2666818Z       |      ^~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.2667495Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ControllerBase.h:109:6: note:   candidate expects 0 arguments, 1 provided
2023-04-16T03:43:05.2668724Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp: In member function 'bool BipedalLocomotion::TSID::SO3Task::setSetPoint(const SO3d&, const Tangent&, const Tangent&)':
2023-04-16T03:43:05.2670174Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/bipedal-locomotion-framework/src/TSID/src/SO3Task.cpp:189:47: error: no matching function for call to 'LieGroupControllers::ProportionalDerivativeController<manif::SO3<double> >::setDesiredState(<brace-enclosed initializer list>)'
2023-04-16T03:43:05.2670924Z   189 |     ok = ok && m_SO3Controller.setDesiredState({I_R_F, angularVelocity});
2023-04-16T03:43:05.2671277Z       |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
2023-04-16T03:43:05.3085326Z   116 | bool ControllerBase<_Derived>::setDesiredState(Ts&&... state)
2023-04-16T03:43:05.3085640Z       |      ^~~~~~~~~~~~~~~~~~~~~~~~
2023-04-16T03:43:05.3086352Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/LieGroupControllers/impl/ControllerBase.h:116:6: note:   candidate expects 0 arguments, 1 provided
2023-04-16T03:43:05.3086863Z ninja: build stopped: subcommand failed.
2023-04-16T03:43:29.0160255Z [365/432] Performing build step for 'GazeboYARPPlugins'
~~~

traversaro, 2023-04-16:
I guess we changed the API in https://github.com/ami-iit/lie-group-controllers/pull/12, but we did not updated blf accordingly?

traversaro, 2023-04-16:
As expected, `LatestReleases` is working fine.

traversaro, 2023-04-17:
Do you have any idea on this @RiccardoGrieco @GiulioRomualdi ?

GiulioRomualdi, 2023-04-17:
Sorry @traversaro. I fixed in  https://github.com/ami-iit/bipedal-locomotion-framework/pull/653

closed 2023-04-17 by GiulioRomualdi

traversaro, 2023-04-17:
> Sorry @traversaro. I fixed in [ami-iit/bipedal-locomotion-framework#653](https://github.com/ami-iit/bipedal-locomotion-framework/pull/653)

Great, thanks!