robotology/robotology-superbuild #1411: Typos in setup.bat - Windows

EhsanRanjbari, 2023-05-15:
I discovered two typos in the setup.bat file as below: 

1- https://github.com/robotology/robotology-superbuild/blob/ea59067a8913722bcf829c8fc407a99b51cda4c4/cmake/template/setup.bat.in#LL8C101-L8C101

2- https://github.com/robotology/robotology-superbuild/blob/ea59067a8913722bcf829c8fc407a99b51cda4c4/cmake/template/setup.bat.in#LL17C265-L17C265

in both cases `"` misplaced. In particular, I discovered the second one when I tried to load the ergoCub model in Gazebo and the path was not added.

traversaro, 2024-01-30:
I guess this was fixed by @S-Dafarra in https://github.com/robotology/robotology-superbuild/pull/1588, right @EhsanRanjbari ?

EhsanRanjbari, 2024-01-31:
> I guess this was fixed by @S-Dafarra in #1588, is this correct @EhsanRanjbari ?

Yes. Thanks!

Closing the issue as completed.

closed 2024-01-31 by EhsanRanjbari