robotology/robotology-superbuild #1418: Fail target ICUB in ubuntu 20.04

corsinone, 2023-05-27:
Hello, 
I am trying to install the `robotology-superbuild` (I am using commit:[3a5db10](https://github.com/robotology/robotology-superbuild/commit/3a5db10e41b778a21f703f705b4fee5711280995)) and I get the following error:
```
[ 75%] Completed 'ICUB'
[ 75%] Built target ICUB
make: *** [Makefile:95 : all] Error 2
```
My machine is running `ubuntu 20.04.6` and I am using `MATLAB R2021b`.

When running `ccmake ../` in the `build` directory I have **enabled** the following options:
`ROBOTOLOGY_ENABLE_DYNAMICS`, `ROBOTOLOGY_USES_GAZEBO`, `ROBOTOLOGY_USES_MATLAB`, and `ROBOTOLOGY_USES_PYTHON`.

Any idea of what could have caused the error?
In case more details are needed, I can provide them.
If I scroll the terminal, I cannot see explicit errors during the building process.

Thanks in advance for any help.

traversaro, 2023-05-28:
Can you provide the complete log? Thanks!

corsinone, 2023-05-28:
Hi @traversaro, thanks for your quick reply.
I made a log by running  `make 2>&1 | tee log_make_command.txt`. In attachment, such text file.
[log_make_command.txt](https://github.com/robotology/robotology-superbuild/files/11584305/log_make_command.txt)

I've just noticed that opencv was not installed on the machine where I was running the compilation, but I haven't had the time to run it again after installing it.

Can you spot any other issues?

Thanks again :)

traversaro, 2023-05-28:
The real error is:
~~~
/home/cdutau/robotology-superbuild/src/yarp-matlab-bindings/matlab/autogenerated/yarpMATLAB_wrap.cxx: In function ‘int _wrap_Contactable_setCallbackLock__SWIG_0(int, mxArray**, int, mxArray**)’:
/home/cdutau/robotology-superbuild/src/yarp-matlab-bindings/matlab/autogenerated/yarpMATLAB_wrap.cxx:15673:13: error: ‘Mutex’ is not a member of ‘yarp::os’
15673 |   yarp::os::Mutex *arg2 = (yarp::os::Mutex *) 0 ;
      |             ^~~~~
/home/cdutau/robotology-superbuild/src/yarp-matlab-bindings/matlab/autogenerated/yarpMATLAB_wrap.cxx:15673:20: error: ‘arg2’ was not declared in this scope; did you mean ‘arg1’?
15673 |   yarp::os::Mutex *arg2 = (yarp::os::Mutex *) 0 ;
      |                    ^~~~
      |                    arg1

~~~

It seems that for some reason you have a quite old yarp-matlab-bindings version that is not compatible with the YARP version you are installing. To understand more, can you try to run:
~~~
cd robotology-superbuild
bash ./scripts/robotologyGitStatus.sh
~~~
to get the version of the superbuild subprojects you are using? Thanks!

corsinone, 2023-06-01:
@charlesdutau provide a txt file with the output of that command, so that @traversaro can help you at best.

Thanks Silvio for your time, I know you are currently busy with the demo at ICRA, so I really appreciate your effort to find the time to check this issue :)

charlesdutau, 2023-06-01:
Alright, I just ran the command again and this is the output
[log_git_status_script.txt](https://github.com/robotology/robotology-superbuild/files/11624578/log_git_status_script.txt)

It may be worth pointing out that all the lines beginning with ?? were not here when I first launched the script, and thus are probably the result of me trying to run ccmake ./ and make on the YARP, yarp-matlab-bindings, and YCM source directories.

traversaro, 2023-06-04:
Thanks @corsinone, that is exactly what I was going to suggest. If the issue is fixed, feel free to close the issue, thanks!

closed 2023-06-06 by traversaro