robotology/robotology-superbuild #1426: Stable branchs CI Failure Early June 2023

traversaro, 2023-06-08:
See https://github.com/robotology/robotology-superbuild/actions/runs/5206701031 .

When compiling together icub-main master and icub-firmware-shared master, the configuration is failing with:
~~~
2023-06-08T02:59:30.4612670Z -- Qt5: found
2023-06-08T02:59:30.4612928Z CMake Error at conf/iCubFindDependencies.cmake:69 (message):
2023-06-08T02:59:30.4613344Z   An old version of icub-firmware-shared has been detected: at least 1.34.2
2023-06-08T02:59:30.4613605Z   is required
2023-06-08T02:59:30.4613808Z Call Stack (most recent call first):
2023-06-08T02:59:30.4614039Z   CMakeLists.txt:57 (include)
2023-06-08T02:59:30.4614171Z 
2023-06-08T02:59:30.4614186Z 
~~~

Perhaps we need to merge devel in master for icub-firmware-shared in view of the 2023.05 distro release?

fyi @Nicogene @marcoaccame @pattacini

pattacini, 2023-06-08:
Yep, I was preparing `icub-main` for the distro, hence I fast-forwarded `master` to `devel` overlooking the interaction between `icub-main@master` and `icub-firmware-shared@master`.

Advancing `master` to `devel` for `icub-firmware-shared` should solve the issue, as you mentioned. Especially, as I'm afraid that tagging will take still some time.

>Perhaps we need to merge devel in master for icub-firmware-shared in view of the 2023.05 distro release?

@Nicogene @marcoaccame can we do this?

Nicogene, 2023-06-08:
> @Nicogene @marcoaccame can we do this?

Yes, but first we need to await:
- https://github.com/robotology/icub-firmware-shared/pull/83
- https://github.com/robotology/icub-firmware-shared/pull/84

Is it a problem @traversaro?

cc @valegagge @marcoaccame @MSECode @pattacini

valegagge, 2023-06-08:
> first we need to await:

There is also the diagnostic ready to merge!
I'm waiting the calib14: today @MSECode 'll test on ergocub, then his PRs can be merged.
Ather that, I'll update the `fw-version` increasing the version and I'll update the dependency file in `icub-main`.
I already tested on iCubGenova11.

Nicogene, 2023-06-08:
> > Is it a problem @traversaro?
> 
> Can't we already merge devel in master once now (with no release) so the CI is fixed, and merge again once those PR are merged? I prefer to avoid having a broken CI as that could hide other problems.

Yes! I did it now 👍🏻

traversaro, 2023-06-08:
Great, thanks!

closed 2023-06-08 by traversaro