robotology/robotology-superbuild #1429: VS2022 CI Build Failure in ROBOTOLOGY_ENABLE_TESTING

traversaro, 2023-06-16:
Part of the error is:
~~~
2023-06-15T04:07:47.9314513Z D:\a\robotology-superbuild\robotology-superbuild\src\blocktest-yarp-plugins\src\actionSendDirectPosition.cpp(126,41): warning C4267: 'argument': conversion from 'size_t' to 'const int', possible loss of data [D:\a\robotology-superbuild\robotology-superbuild\build\src\blocktest-yarp-plugins\blocktestyarpplugins.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\blocktest-yarp-plugins.vcxproj]
2023-06-15T04:07:49.0206752Z C:/Miniconda3/envs/test/Library/include\boost/asio/detail/executor_function.hpp(101,1): error C2872: 'impl': ambiguous symbol [D:\a\robotology-superbuild\robotology-superbuild\build\src\blocktest-yarp-plugins\blocktestyarpplugins.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\blocktest-yarp-plugins.vcxproj]
2023-06-15T04:07:49.0229233Z   D:/a/robotology-superbuild/robotology-superbuild/build/install/include\yarp/os/Log.h(44,11): message : could be 'yarp::os::impl' [D:\a\robotology-superbuild\robotology-superbuild\build\src\blocktest-yarp-plugins\blocktestyarpplugins.vcxproj]
2023-06-15T04:07:49.0232104Z   D:/a/robotology-superbuild/robotology-superbuild/build/install/include\yarp/dev/ImplementInteractionMode.h(18,11): message : or       'yarp::dev::impl' [D:\a\robotology-superbuild\robotology-superbuild\build\src\blocktest-yarp-plugins\blocktestyarpplugins.vcxproj]
2023-06-15T04:07:49.0233949Z   C:/Miniconda3/envs/test/Library/include\boost/asio/detail/executor_function.hpp(87): message : see reference to function template instantiation 'void boost::asio::detail::executor_function::complete<Function,Alloc>(boost::asio::detail::executor_function::impl_base *,bool)' being compiled [D:\a\robotology-superbuild\robotology-superbuild\build\src\blocktest-yarp-plugins\blocktestyarpplugins.vcxproj]
~~~

Successful build: https://github.com/robotology/robotology-superbuild/actions/runs/5274079205/jobs/9538173073
Failing build: https://github.com/robotology/robotology-superbuild/actions/runs/5285525654/jobs/9564086180

traversaro, 2023-06-16:
The successful build is using MSVC 19.35.32217.1, while the failure is using MSVC 19.36.32532.0 . However, I am not sure if the problem is solved when using latest boost-asio.

traversaro, 2023-06-16:
@Nicogene is somebody actually using `blocktest-yarp-plugins` ?

pattacini, 2023-06-17:
> @Nicogene is somebody actually using `blocktest-yarp-plugins` ?

It's used in the automatic calibration of the FT sensors.

Nicogene, 2023-06-19:
I checked and `blocktest-yarp-plugins` does not uses asio/boos:

- https://github.com/search?q=repo%3Arobotology%2Fblocktest-yarp-plugins%20boost&type=code
- https://github.com/search?q=repo%3Arobotology%2Fblocktest-yarp-plugins+asio&type=code

On the other hand `blocktest` it does:

- https://github.com/search?q=repo%3Arobotology%2Fblocktest+asio&type=code

I am wondering why it is not failing when compiling blocktest

traversaro, 2023-06-19:
Possibly the use of YARP is forcing the use of a newer C++ standard?

Nicogene, 2023-06-19:
I made some cleanup in blocktest, maybe this fix the problem: https://github.com/robotology/blocktest/pull/63

Nicogene, 2023-06-19:
It should be fixed by:

- https://github.com/robotology/blocktest-yarp-plugins/pull/14

Nicogene, 2023-06-19:
Released a fixed version of byp https://github.com/robotology/blocktest-yarp-plugins/releases/tag/v1.1.4

traversaro, 2023-06-27:
Thanks @Nicogene !

closed 2023-06-27 by traversaro