robotology/robotology-superbuild #1445: Casadi build failed when option "WITH_PROXQP:BOOL" is ON

EhsanRanjbari, 2023-07-18:
I tried to build Casadi. it failed with an error: 

<details><summary>Build log</summary>
<p>

```cmake
Build started...
1>------ Build started: Project: casadi_sundials, Configuration: Release x64 ------
2>------ Build started: Project: casadi_csparse, Configuration: Release x64 ------
3>------ Build started: Project: casadi, Configuration: Release x64 ------
3>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/core/CMakeLists.txt
1>casadi_sundials.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_sundials.lib
2>casadi_csparse.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_csparse.lib
3>casadi.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi.dll
4>------ Build started: Project: casadi_xmlfile_tinyxml, Configuration: Release x64 ------
5>------ Build started: Project: casadi_sundials_common, Configuration: Release x64 ------
6>------ Build started: Project: casadi_rootfinder_nlpsol, Configuration: Release x64 ------
7>------ Build started: Project: casadi_rootfinder_newton, Configuration: Release x64 ------
8>------ Build started: Project: casadi_rootfinder_kinsol, Configuration: Release x64 ------
9>------ Build started: Project: casadi_rootfinder_fast_newton, Configuration: Release x64 ------
10>------ Build started: Project: casadi_nlpsol_sqpmethod, Configuration: Release x64 ------
11>------ Build started: Project: casadi_nlpsol_scpgen, Configuration: Release x64 ------
12>------ Build started: Project: casadi_nlpsol_qrsqp, Configuration: Release x64 ------
13>------ Build started: Project: casadi_nlpsol_ipopt, Configuration: Release x64 ------
14>------ Build started: Project: casadi_nlpsol_feasiblesqpmethod, Configuration: Release x64 ------
15>------ Build started: Project: casadi_linsol_tridiag, Configuration: Release x64 ------
16>------ Build started: Project: casadi_linsol_symbolicqr, Configuration: Release x64 ------
17>------ Build started: Project: casadi_linsol_qr, Configuration: Release x64 ------
18>------ Build started: Project: casadi_linsol_lsqr, Configuration: Release x64 ------
19>------ Build started: Project: casadi_linsol_ldl, Configuration: Release x64 ------
9>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
5>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/interfaces/sundials/CMakeLists.txt
6>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
10>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
12>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
8>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/interfaces/sundials/CMakeLists.txt
11>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
4>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/interfaces/tinyxml/CMakeLists.txt
13>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/interfaces/ipopt/CMakeLists.txt
15>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
16>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
14>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
18>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
17>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
19>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
6>casadi_rootfinder_nlpsol.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_rootfinder_nlpsol.dll
18>casadi_linsol_lsqr.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_linsol_lsqr.dll
20>------ Build started: Project: casadi_linsol_csparsecholesky, Configuration: Release x64 ------
8>casadi_rootfinder_kinsol.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_rootfinder_kinsol.dll
4>casadi_xmlfile_tinyxml.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_xmlfile_tinyxml.dll
10>casadi_nlpsol_sqpmethod.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_nlpsol_sqpmethod.dll
12>casadi_nlpsol_qrsqp.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_nlpsol_qrsqp.dll
5>casadi_sundials_common.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_sundials_common.dll
11>casadi_nlpsol_scpgen.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_nlpsol_scpgen.dll
13>casadi_nlpsol_ipopt.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_nlpsol_ipopt.dll
16>casadi_linsol_symbolicqr.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_linsol_symbolicqr.dll
15>casadi_linsol_tridiag.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_linsol_tridiag.dll
7>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/solvers/CMakeLists.txt
14>casadi_nlpsol_feasiblesqpmethod.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_nlpsol_feasiblesqpmethod.dll
19>casadi_linsol_ldl.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_linsol_ldl.dll
17>casadi_linsol_qr.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_linsol_qr.dll
21>------ Build started: Project: casadi_integrator_idas, Configuration: Release x64 ------
7>casadi_rootfinder_newton.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_rootfinder_newton.dll
22>------ Build started: Project: casadi_integrator_cvodes, Configuration: Release x64 ------
23>------ Build started: Project: casadi_linsol_csparse, Configuration: Release x64 ------
24>------ Build started: Project: casadi_interpolant_linear, Configuration: Release x64 ------
25>------ Build started: Project: casadi_interpolant_bspline, Configuration: Release x64 ------
26>------ Build started: Project: casadi_integrator_rk, Configuration: Release x64 ------
27>------ Build started: Project: casadi_integrator_collocation, Configuration: Release x64 ------
9>casadi_rootfinder_fast_newton.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_rootfinder_fast_newton.dll
[... 36 lines cut ...]
21>casadi_integrator_idas.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi_integrator_idas.dll
34>casadi-cli.vcxproj -> C:\Users\eranjbari\robotology-superbuild\build\src\casadi\Release\casadi-cli.exe
30>C:/Users/eranjbari/robotology-superbuild/build/install/include\proxsuite/linalg/veg/tuple.hpp(357,1): error C2059: syntax error: '...'
30>C:/Users/eranjbari/robotology-superbuild/build/install/include\proxsuite/linalg/veg/tuple.hpp(431): message : see reference to class template instantiation 'proxsuite::linalg::veg::tuple::IndexedTuple<proxsuite::linalg::veg::_detail::_meta::integer_sequence<proxsuite::linalg::veg::usize,Is...>*,Ts...>' being compiled
30>proxqp_interface_meta.cpp
30>C:/Users/eranjbari/robotology-superbuild/build/install/include\proxsuite/linalg/veg/tuple.hpp(357,1): error C2059: syntax error: '...'
30>C:/Users/eranjbari/robotology-superbuild/build/install/include\proxsuite/linalg/veg/tuple.hpp(431): message : see reference to class template instantiation 'proxsuite::linalg::veg::tuple::IndexedTuple<proxsuite::linalg::veg::_detail::_meta::integer_sequence<proxsuite::linalg::veg::usize,Is...>*,Ts...>' being compiled
30>Generating Code...
30>Done building project "casadi_conic_proxqp.vcxproj" -- FAILED.
35>------ Build started: Project: libs, Configuration: Release x64 ------
35>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/casadi/CMakeLists.txt
36>------ Build started: Project: ALL_BUILD, Configuration: Release x64 ------
36>Building Custom Rule C:/Users/eranjbari/robotology-superbuild/src/casadi/CMakeLists.txt
========== Build: 35 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========
========== Build started at 1:01 PM and took 07.574 seconds ==========
``` 

</p>
</details> 

it was built and installed only after I changed the option "WITH_PROXQP:BOOL" from `ON `to `OFF`. 

I am using Microsoft Visual Studio 2022 and MSVC  v143. 

cc @S-Dafarra

traversaro, 2023-07-21:
Thanks for the report! If you have a workaround working for you, I will look into this in August.

traversaro, 2023-08-03:
> Thanks for the report! If you have a workaround working for you, I will look into this in August.

Mid august. : )

traversaro, 2023-09-11:
@EhsanRanjbari I updated to the latest version of proxsuite in https://github.com/robotology/robotology-superbuild/pull/1476 . Can you check if you still have this problem with the latest master version of the robotology-superbuild ?

EhsanRanjbari, 2023-09-13:
> @EhsanRanjbari I updated to the latest version of proxsuite in #1476 . Can you check if you still have this problem with the latest master version of the robotology-superbuild ?

It seems fixed now. I just built and installed with the option `WITH_PROXQP:BOOL` set to `ON` and there were no errors.

closed 2023-09-13 by traversaro